geoffdavis/esphome-mitsubishiheatpump #1: Compile failure

nickrout, 2020-03-12:
Firstly, thanks for getting this into esphome. 

I get the following using esphome 1.14.3. My setup is per the README.md with my own names substituted for denheatpump. If there is any more info you need, let me know.

```
INFO Reading configuration /config/bedroom_heat_pump.yaml...
INFO Detected timezone 'NZST' with UTC offset 12 and daylight savings time from 27 September 02:00:00 to 04 April 03:00:00
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/bedroomheatpump
Processing bedroomheatpump (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757 @ 1.2.2
ESPAsyncTCP-esphome @ 1.2.2 has been successfully installed!
LibraryManager: Installing HeatPump
git version 2.17.1
Cloning into '/config/bedroomheatpump/.piolibdeps/bedroomheatpump/_tmp_installing-2q2rushb-package'...
HeatPump @ 15467bf has been successfully installed!
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.6

Downloading  [#####-------------------------------]   16%
Downloading  [######------------------------------]   16%
Downloading  [######------------------------------]   17%
Downloading  [######------------------------------]   18%
Downloading  [######------------------------------]   19%
Downloading  [#######-----------------------------]   19%
Downloading  [#######-----------------------------]   20%
Downloading  [#######-----------------------------]   21%
Downloading  [#######-----------------------------]   22%
Downloading  [########----------------------------]   22%
Downloading  [########----------------------------]   23%
Downloading  [########----------------------------]   24%
Downloading  [#########---------------------------]   25%
Downloading  [#########---------------------------]   26%
Downloading  [#########---------------------------]   27%
Downloading  [##########--------------------------]   27%
Downloading  [##########--------------------------]   28%
Downloading  [##########--------------------------]   29%
Downloading  [##########--------------------------]   30%
Downloading  [###########-------------------------]   30%
Downloading  [###########-------------------------]   31%
Downloading  [###########-------------------------]   32%
Downloading  [############------------------------]   33%
Downloading  [############------------------------]   34%
Downloading  [############------------------------]   35%
Downloading  [############------------------------]   36%
Downloading  [#############-----------------------]   36%
Downloading  [#############-----------------------]   37%
Downloading  [#############-----------------------]   38%
Downloading  [##############----------------------]   39%
Downloading  [##############----------------------]   40%
Downloading  [##############----------------------]   41%
Downloading  [###############---------------------]   41%
Downloading  [###############---------------------]   42%
Downloading  [###############---------------------]   43%
Downloading  [###############---------------------]   44%
Downloading  [################--------------------]   44%
Downloading  [################--------------------]   45%
Downloading  [################--------------------]   46%
Downloading  [#################-------------------]   47%
Downloading  [#################-------------------]   48%
Downloading  [#################-------------------]   49%
Downloading  [##################------------------]   50%
Downloading  [##################------------------]   51%
Downloading  [##################------------------]   52%
Downloading  [###################-----------------]   53%
[... 259 lines cut ...]
                            ^
src/esphome-mitsubishiheatpump/espmhp.cpp:285:15: error: 'class MitsubishiHeatPump' has no member named 'swing_mode'
         this->swing_mode = climate::CLIMATE_SWING_OFF;
               ^
src/esphome-mitsubishiheatpump/espmhp.cpp:285:28: error: 'CLIMATE_SWING_OFF' is not a member of 'esphome::climate'
         this->swing_mode = climate::CLIMATE_SWING_OFF;
                            ^
In file included from src/esphome/components/api/api_server.h:6:0,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome.h:2,
                 from src/esphome-mitsubishiheatpump/espmhp.h:18,
                 from src/esphome-mitsubishiheatpump/espmhp.cpp:15:
src/esphome-mitsubishiheatpump/espmhp.cpp:287:46: error: 'class MitsubishiHeatPump' has no member named 'swing_mode'
     ESP_LOGI(TAG, "Swing mode is: %i", this->swing_mode);
                                              ^
src/esphome/core/log.h:101:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^
src/esphome-mitsubishiheatpump/espmhp.cpp:287:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "Swing mode is: %i", this->swing_mode);
     ^
Compiling .pioenvs/bedroomheatpump/src/esphome/components/api/list_entities.cpp.o
*** [.pioenvs/bedroomheatpump/src/esphome-mitsubishiheatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 21.05 seconds =========================
```

geoffdavis, 2020-03-12:
Hi Nick,

Looks like this is due to the fact that I’m using several features of the dev branch of ESPHome. Can you try running that and see if that helps?

nickrout, 2020-03-13:
Thanks, it compiled ok under dev version.

closed 2020-03-14 by geoffdavis