geoffdavis/esphome-mitsubishiheatpump #2: Some questions about responsivness and fan and wane settings

jascdk, 2020-03-17:
Hi Geoff.

I now managed to get the heatpump up and running with ESPHOME - thanks alot for putting your effort in to getting this to work:D 

I experience some kind of long responsivness when i change temperature, modes. I see you write instant responsivness in your readme. How fast does your system response? 

The next thing i came to think of is if it possible to get a feedback on the state of the heatpump in the UI. It always says it is off, when the heatpump is actually running / heating. 

Before mowing to ESPHOME i used the code made by gysmo38. It is also a very nice code, but i had some issues with it. In his code it is possible to set the fan to other states - like QUIET and so on. I cant see this in this integration - it goes from Auto, low medium middle and high.

The same thing is with the VANE. In gysmos code there was different modes and also a SWING mode.

Is it something that needs to be put in the code or does ESPHOME not support it?

All the best

Jacob from denmark

jascdk, 2020-03-17:
This is taken from the code at Gysmos repo:

Here is some examples of the states of the fan and vane:

https://pastebin.com/rNznHWEB

if (strcmp(settings.fan, "AUTO") == 0) { 
700     controlPage.replace("_FAN_A_", "selected"); 
701   } 
702   else if (strcmp(settings.fan, "QUIET") == 0) { 
703     controlPage.replace("_FAN_Q_", "selected"); 
704   } 
705   else if (strcmp(settings.fan, "1") == 0) { 
706     controlPage.replace("_FAN_1_", "selected"); 
707   } 
708   else if (strcmp(settings.fan, "2") == 0) { 
709     controlPage.replace("_FAN_2_", "selected"); 
710   } 
711   else if (strcmp(settings.fan, "3") == 0) { 
712     controlPage.replace("_FAN_3_", "selected"); 
713   } 
714   else if (strcmp(settings.fan, "4") == 0) { 
715     controlPage.replace("_FAN_4_", "selected"); 
716   } 
717 
 
718   controlPage.replace("_VANE_V_", settings.vane); 
719   if (strcmp(settings.vane, "AUTO") == 0) { 
720     controlPage.replace("_VANE_A_", "selected"); 
721   } 
722   else if (strcmp(settings.vane, "1") == 0) { 
723     controlPage.replace("_VANE_1_", "selected"); 
724   } 
725   else if (strcmp(settings.vane, "2") == 0) { 
726     controlPage.replace("_VANE_2_", "selected"); 
727   } 
728   else if (strcmp(settings.vane, "3") == 0) { 
729     controlPage.replace("_VANE_3_", "selected"); 
730   } 
731   else if (strcmp(settings.vane, "4") == 0) { 
732     controlPage.replace("_VANE_4_", "selected"); 
733   } 
734   else if (strcmp(settings.vane, "5") == 0) { 
735     controlPage.replace("_VANE_5_", "selected"); 
736   } 
737   else if (strcmp(settings.vane, "SWING") == 0) { 
738     controlPage.replace("_VANE_S_", "selected"); 
739   } 
740 
 
741   controlPage.replace("_WIDEVANE_V_", settings.wideVane); 
742   if (strcmp(settings.wideVane, "<<") == 0) { 
743     controlPage.replace("_WVANE_1_", "selected"); 
744   } 
745   else if (strcmp(settings.wideVane, "<") == 0) { 
746     controlPage.replace("_WVANE_2_", "selected"); 
747   } 
748   else if (strcmp(settings.wideVane, "|") == 0) { 
749     controlPage.replace("_WVANE_3_", "selected"); 
750   } 
751   else if (strcmp(settings.wideVane, ">") == 0) { 
752     controlPage.replace("_WVANE_4_", "selected"); 
753   } 
754   else if (strcmp(settings.wideVane, ">>") == 0) { 
755     controlPage.replace("_WVANE_5_", "selected"); 
756   } 
757   else if (strcmp(settings.wideVane, "SWING") == 0) { 
758     controlPage.replace("_WVANE_S_", "selected"); 
759   }

DeltaAngle, 2020-03-17:
The fan and swing modes in Geoff's code are based on the supported functions in the ESPHome Dev branch. You can see these documented here, https://next.esphome.io/components/climate/index.html. ESPHome would need to be updated to support the numerical fan speed, vane swing, and wide vane swing settings.

jascdk, 2020-03-17:
@DeltaAngle Ahhh thanks - that makes sense;) 

Any thoughts on the responsiveness?

geoffdavis, 2020-03-17:
The lack of responsiveness is likely due to the polling loop only running once every two seconds. You might get it to be more responsive by fiddling with the value for _poll_interval_ - decrease that from the default of 2000 milliseconds to 1000 and see if it works any better. In your lambda, declare it like `auto my_heat_pump = new HeatPump(&Serial, 1000);` in your lambda.

([Code that implements is here](https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/79afe8f7348b4768b09b64548dc33b6e495961c7/espmhp.h#L54))

A possibly better solution would be to convert to callbacks but I was unable to get those to work during my development, and just wanted to get this thing out the door.

geoffdavis, 2020-03-17:
As for the different fan modes, @DeltaAngle nailed it on the head.

I'd love to get the custom modes like gysmo's implementation, but I haven't figured out how to override the ESPHome definitions. There seems to be a lot of work that would have to happen in the ESPHome code base in order to provide custom fan levels, as the conversation between ESPHome and HomeAssistant is pretty heavily locked down. There may be a way to make it work akin to how the FastLED component lets you declare "lamba modes", but that's likely a whole lot of development work.

geoffdavis, 2020-03-18:
@jascdk I did some experiments with the refresh interval, and found that:
* 1000ms is an improvement
* 500 ms didn't yield much in the way of responsiveness

Watching the logs, it appears that the heat pump doesn't report a changed state until the fan vanes open. In particular, if the unit is off, and you turn it on, the HeatPump won't report the change over it's serial port until the slats open up. This can take up to 10 seconds.

So that's one mode of slowness.

The other sluggish part is the temperature changes. I have a feeling that getting the callbacks working is going to help that. Right now, the code sends an update to Home Assistant every polling loop. My theory is that the constant stream of state changes is too much for HomeAssistant, either in the core (not likely) or in the web front end components. By converting to callbacks, update messages would only be sent to home assistant IF something changed. I'll see if I can get that functionality working.

geoffdavis, 2020-03-18:
[Release 1.1.0](https://github.com/geoffdavis/esphome-mitsubishiheatpump/tree/1.1.0) addresses the responsiveness issue.

I got callbacks to work (turns out that my initialization fixes had fixed all of the crashes I was seeing with the callbacks), and increased the polling interval from 2 seconds to 500 ms. The web interface is much snappier, and your temperature graphs shouldn't be as large now due to it only including data points when things change rather than every update.

Please try it out and let me know if you see any problems in a new ticket.

Thanks!

closed 2020-03-18 by geoffdavis

jascdk, 2020-03-18:
> [Release 1.1.0](https://github.com/geoffdavis/esphome-mitsubishiheatpump/tree/1.1.0) addresses the responsiveness issue.
>
> I got callbacks to work (turns out that my initialization fixes had fixed [...]

Hi Geoff:)

That did make an hell of a difference instead of the old one:) Nice that you got it working out! If I have other issues I will get back to you. I have put this repo on my watch list, so I can keep up with eventually other updates! Nice work Geoff!!!!