geoffdavis/esphome-mitsubishiheatpump #3: Wemos D1 Mini compile errors

matthew-larner, 2020-05-15:
I'm getting a bunch of compile errors on a Wemos D1 Mini:

`INFO Reading configuration config/aircon.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/aircon
Processing aircon (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757 @ 1.2.2
ESPAsyncTCP-esphome @ 1.2.2 has been successfully installed!
LibraryManager: Installing HeatPump
git version 2.17.1
Cloning into '/home/ubuntu/config/aircon/.piolibdeps/aircon/_tmp_installing-3vo3y5bm-package'...
HeatPump @ abe6343 has been successfully installed!
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.6
ESPAsyncWebServer-esphome @ 1.2.6 has been successfully installed!
LibraryManager: Installing id=6757
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Looking for AsyncMqttClient-esphome library in registry
LibraryManager: Installing id=6759 @ 0.8.4
AsyncMqttClient-esphome @ 0.8.4 has been successfully installed!
LibraryManager: Installing id=6757 @ ^1.2.2
LibraryManager: Installing id=6798 @ ^1.1.1
Looking for ArduinoJson-esphomelib library in registry
LibraryManager: Installing id=3837 @ 5.13.3
ArduinoJson-esphomelib @ 5.13.3 has been successfully installed!
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <HeatPump> 1.0.0 #abe6343
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/aircon/src/esphome-mitsubishiheatpump/espmhp.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/aircon/src/esphome/components/logger/logger.cpp.o
src/esphome-mitsubishiheatpump/espmhp.cpp: In member function 'virtual esphome::climate::ClimateTraits MitsubishiHeatPump::traits()':
src/esphome-mitsubishiheatpump/espmhp.cpp:72:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_dry_mode'
     traits.set_supports_dry_mode(true);
            ^
src/esphome-mitsubishiheatpump/espmhp.cpp:73:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_only_mode'
     traits.set_supports_fan_only_mode(true);
            ^
src/esphome-mitsubishiheatpump/espmhp.cpp:79:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_on'
     traits.set_supports_fan_mode_on(false);
            ^
src/esphome-mitsubishiheatpump/espmhp.cpp:80:12: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_fan_mode_off'
     traits.set_supports_fan_mode_off(false);
            ^
[... 140 lines cut ...]
               ^
src/esphome-mitsubishiheatpump/espmhp.cpp:282:28: error: 'CLIMATE_SWING_VERTICAL' is not a member of 'esphome::climate'
         this->swing_mode = climate::CLIMATE_SWING_VERTICAL;
                            ^
src/esphome-mitsubishiheatpump/espmhp.cpp:285:15: error: 'class MitsubishiHeatPump' has no member named 'swing_mode'
         this->swing_mode = climate::CLIMATE_SWING_OFF;
               ^
src/esphome-mitsubishiheatpump/espmhp.cpp:285:28: error: 'CLIMATE_SWING_OFF' is not a member of 'esphome::climate'
         this->swing_mode = climate::CLIMATE_SWING_OFF;
                            ^
In file included from src/esphome/components/logger/logger.h:4:0,
                 from src/esphome.h:9,
                 from src/esphome-mitsubishiheatpump/espmhp.h:17,
                 from src/esphome-mitsubishiheatpump/espmhp.cpp:15:
src/esphome-mitsubishiheatpump/espmhp.cpp:287:46: error: 'class MitsubishiHeatPump' has no member named 'swing_mode'
     ESP_LOGI(TAG, "Swing mode is: %i", this->swing_mode);
                                              ^
src/esphome/core/log.h:101:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^
src/esphome-mitsubishiheatpump/espmhp.cpp:287:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "Swing mode is: %i", this->swing_mode);
     ^
*** [.pioenvs/aircon/src/esphome-mitsubishiheatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 24.53 seconds =========================`

DeltaAngle, 2020-05-15:
Your compile error is due to using the wrong version of ESPHome. You'll need to use the Dev version of ESPHome and compile again.

You can find instructions for getting the latest Dev version here, https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version

jamescadd, 2020-05-23:
Any indication which release of ESPHome this will be included in?  Would rather not install dev if I can avoid it.

mihalski, 2020-05-26:
FYI you can have both the normal and dev versions of the add-on installed side by side and use the dev one exclusively for this.

jamescadd, 2020-05-26:
@mihalski it would be ideal to have both versions installed, any info on how to use the dev version just for the heat pump?  There was a mention about port conflicts in the docs somewhere that made me hesitate to try this.

mihalski, 2020-05-27:
@jamescadd The way I have done it is simply to install both adding and go to the sidebar tab of the dev one only when managing devices I want on the dev branch.

geoffdavis, 2020-05-27:
I haven't seen anyone post a dev timeline on the Esphome project, and the Discord channel was pretty cagey about timing.

I had to pin ESPHome to an [older commit|https://github.com/esphome/esphome/commit/8b92456ded97c236846e959fb3797f51b9dda4fc] due to a recent breaking change, and need to revisit that.

geoffdavis, 2020-06-03:
Looks like my need to pin a commit is no longer an issue. I'm back on mainline "Dev" again.

closed 2020-06-03 by geoffdavis