geoffdavis/esphome-mitsubishiheatpump #4: hvac_action: 'off'

yabuts, 2020-05-24:
hi i use this with great success (on esp01), works great, only isue that hvac_action is always 'off'

miGii, 2020-05-28:
Same problem here with MSZ-FH25VE and NodeMCU. Otherwise it works well. Awesome project, thank you.

jamescadd, 2020-06-01:
Same issue here with MSZ-GL09NA and ESP8266MOD chip.  I notice that the State of the climate entity looks like it's reporting the value that I expect to see in hvac_action, but hvac_action is always 'off'.

pgenera, 2020-06-01:
The heat pump doesn't report its current action in the serial protocol (at least not that I could find). I added a hack in #7 to make a guess which pretties up my UI.

yabuts, 2020-06-03:
> The heat pump doesn't report its current action in the serial protocol (at least not that I could find). I added a hack in #7 to make a guess which pretties up my UI.

Well , your hack make it maybe pretty but is not accurate,  as it does not take into account the dofriantul setting that the heat pump is coded with..

Also I have noticed in my units,  that when having it set at a set point on cool it will somehow use 2 degrees below at the mid point, (it might  be something  wrong)let's say I set at 68, system will cool untill 66, so you hack will not even line up

closed 2020-06-03 by geoffdavis

geoffdavis, 2020-06-03:
@yabuts the conversion from F to C to F again leaves a lot to be desired, but it's all being handled at the Home Assistant layer. I've noted that the remote control does a lot of rounding in order to make the temperature iteration appear to be smooth.

geoffdavis, 2020-06-03:
Also, what is `dofriantul`? Is that defrost, or dehumidify? I'm not sure that the serial protocol will actually tell us about it's defrosting. There is a mode in the dev version for "drying" aka dehumidifying so I'll see if I can add that in.

yabuts, 2020-06-03:
No, sorry I had few typographical  error.
 I.meant temperature differential , sometime  referred in HA as tolerances
or cycles

geoffdavis, 2020-06-03:
@yabuts do you happen to know what the Mitsubishi's are set for in terms of temperature differentials (aka "hysteresis")? I'll take a wild guess at +/- 1.5 C, but I don't know.

yabuts, 2020-06-03:
I don't  know,  but since  there is definitely some hysteresis built-in , I
do t think it's ok just to assume that at cool < set is off , and >set is
on.

yabuts, 2020-06-03:
I just tried the new commits,  the UI feels so smooth,  thanks.

geoffdavis, 2020-06-03:
I'll push a new commit with the "idle" mode. It's going to take a lot more work to guess the hysteresis since we have to determine whether the temperature is rising or falling at any given point, and it'll take more time to debug than I currently have.