geoffdavis/esphome-mitsubishiheatpump #5: can u include your project inside esphome ?

blademckain, 2020-05-28:
Can u include your project inside official esphome?
By creating a custom component that can be used directly with:

```yaml
climate:
  - platform: mitsubishiheatpump
.....
```

yabuts, 2020-05-28:
There is already a esphome Mitsubishi  version,  I use it
https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/master/README.md

blademckain, 2020-05-29:
yeah! but u have to manually insert a custom component.
I asked if it is possible to pull this repository directly into the official esphome repository.
In this way it could be maintained by the esphome community also in the next updates / changes

geoffdavis, 2020-06-03:
Hi @blademckain 

Getting this included into mainline ESPhome was always my goal with this project. I wanted to do some testing and get a few people using it before going through the merge process with the main project.

blademckain, 2020-06-03:
ok  @geoffdavis 
thank you very much for your additions to esphome.
in this period I'm buying mitsubishi air conditioners.
I will try your component as soon as it is included.

Phillipzx3, 2020-06-15:
Hello @geoffdavis 

Total github noob here (still learning the terms), so please let me know if I'm going about asking this in an offensive way.  I've been trying to incorporate my 2 Mr. Slim units into HA via MQTT.  I got as far as seeing all the data (using the "SwiCago/Heatpump" library) in the Temperature card in Lovelace but I hadn't got to where I could change anything.  Then I saw your project and decided that might be a better way of going.  :-) 

Ok, here's where I get stumped. You say these files (repository?)  needs to be in my "esphome configuration" directory.  Can I assume that is the same esphome directory I see via using my "File Editor" add-on?  I see the directories are there and I can create a "src" directory I made (inside the esphome folder/directory) there as well. 

Ok...once that's done, I'm lost. I have no clue how to "git clone" into that directory. Each (guessed at) attempt places the files in the esphome folder and not the src folder.  Unfortunatly there are no Youtube videos explaining this process that I could find.

I'm running HA on a Oracle VirtualBox if it matters.  I also have the ESPHome dev version installed. 

If I'm out of line, just ignore this message and I'll wait until you have it ready for "the masses."  :-)

Thanks
Phil
K7JBL

loongyh, 2020-07-02:
> Hi @blademckain
> 
> Getting this included into mainline ESPhome was always my goal with this project. I wanted to do some testing and get a few people using it before going through the merge process with the main project.

@geoffdavis excellent work and many thanks for this integration, it's working very well currently for my units. Since I live in a hot and humid tropical region there isn't a heat mode for my unit, as such I have to modify the traits to hide it. If you're intending to create an official ESPHome component for this it'll be great if the modes can be selected/hidden in the .yaml config.

geoffdavis, 2020-07-06:
@Phillipzx3 You'll want to check out the repository under the HomeAssistant `config` directory, in `config/esphome/src/esphome-mitsubishiheatpump`.

Assuming that you're using HassOS, this could either be done with Visual Studio Code running on your HomeAssistant node, or with the command line git client using the SMB Share plugin from HomeAssistant. I've done both.

Here's what the tree should look like, as viewed in VS Code. Basically, under `config/esphome`, create a subdirectory called `src` and then check out this repository under that `src` directory.

[image]

geoffdavis, 2020-07-06:
@loongyh thanks for your other pull request. In terms of adding flags, presumably the best way to do that would be by adding optional arguments to the constructor for the `MitsubishiHeatPump` class. As it currently stands, the class itself does no parsing of the YAML configuration, so any arguments need to be passed within the `lambda` block.

Using my current configuration as a starting point:

```yaml
climate:
  - platform: custom
    lambda: |-
      auto my_heatpump = new MitsubishiHeatPump(&Serial); /* Add arguments here */
      App.register_component(my_heatpump);
      return {my_heatpump};

    climates:
      - name: ${friendly_name}
```

Phillipzx3, 2020-07-07:
Geoffdavis...Fortunately, I managed to stumble my way into the Visual Studio Code and Samba share add-on.  Once that happened, I saw the light! :-)   My directors are now where they belong. :-)  

I can control fan speed, all the modes ( à la modes? :-) ) and the vertical swing. Still haven't figured out how to do the "split vane" mode, "horizontal swing" or "vane positioning" (I'm using a MSZ-FH09NA) but that's no biggie. The fact that I can do what I can is enough. 

Thank you for providing this code. I'm not ready to dive into MQTT. :-)

kalosbg, 2021-12-01:
Hello @geoffdavis,
Today I switched from SwiCago's project to yours. It worked without any issues.
Thank you!

geoffdavis, 2021-12-03:
> I have contributed ESPHome integrations before, do let me know if you need
> assistance or if you'd like me to do it for you.

Thanks Barry. I’ve made minor contributions to the main ESPHOME repo
before, but I haven’t done a full component like this before.

Right now it’s just a lack of dev cycles to shepherd it through the
process. I’ve been out in the field deploying instruments on towers, so not
much time is available for coding, or hanging Christmas lights for that
matter. :)

loongyh, 2021-12-03:
No worries in #opensource we help each other 😉.

First, we'll probably need to reimplement the mitsubishi protocol used in the swicago library in the component itself, as the maintainers have an aversion to external/lib dependencies and prefer self-contained components. We'll then need to refactor the serial to use `UARTComponent`, should be straightforward to do.

I'll try to find some time to do up the scaffolding, will submit a PR then.

davralin, 2022-02-01:
Let me know if you need any testers for it, I can't submit any code - but I'd be more than happy to test something.

echavet, 2024-01-13:
> No worries in #opensource we help each other 😉.
>
> First, we'll probably need to reimplement the mitsubishi protocol used in the swicago library in the component itself, as the maintainers have an [...]
I did it here:
https://github.com/echavet/MitsubishiCN105ESPHome

closed 2025-03-31 by geoffdavis