geoffdavis/esphome-mitsubishiheatpump #6: UI slow to update?

jamescadd, 2020-05-31:
Hello, not sure if this is an actual issue, but when I change mode from off to cool, the machine reacts immediately and the LED turns on showing that the command was successful.  However the thermostat UI (I'm using simple thermostat) can take ~15 seconds to reflect that change.  Is there a polling interval that can be adjustment to make it more responsive, or could this be an issue with the UI component?

pgenera, 2020-05-31:
the code publishes updates when it receives them from the minisplit; it
takes about 30 seconds for the minisplit to send an update after it gets a
new configuration. I've got a patch that optimistically sends an update
immediately; I'll send a PR once my corporate overlords approve it.

>

pgenera, 2020-05-31:
aforementioned PR: #7

geoffdavis, 2020-06-03:
Merged PR #7 should take care of this. Note that it's blindly issuing the command now and then updating state, but eventually it will sync it's state with the unit if the command didn't actually take effect.

closed 2020-06-03 by geoffdavis