geoffdavis/esphome-mitsubishiheatpump #16: Is ESPHome v1.15.0-dev still required and if so where is it?

zkniebel, 2020-07-17:
ESPHome `v1.15.0-dev` is mentioned in the docs as required for this project to function, but after reviewing both the [releases](https://github.com/esphome/esphome/releases) and [tags](https://github.com/esphome/esphome/tags) of the [ESPHome repository](https://github.com/esphome/esphome) I can't seem to find this pre-release. I also reviewed [all branches](https://github.com/esphome/esphome/branches), including the [dev](https://github.com/esphome/esphome) branch for any indication of `v1.15.0-dev` and couldn't find anything. 

While the links above point at the ESPHome repository, itself, I also checked the [ESPHome Hassio Add-On repository](https://github.com/esphome/hassio) for the same, and came up empty.

It looks like you added the requirement for `v1.15.0-dev` back in March, and since then ESPHome has released [v1.14.4](https://github.com/esphome/esphome/releases/tag/v1.14.4) (5 June 2020) and [v1.14.5](https://github.com/esphome/esphome/releases/tag/v1.14.5) (23 June 2020). 

Do either of these releases contain the unreleased features that you mention this library depends on, in the README?

DeltaAngle, 2020-07-17:
In the ESPHome documentation, it is referred to as the 'bleeding edge' version. You can find the instructions about setting up this version [here](https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version). 

As noted in the linked instructions, there is a separate add-on in the Hassio repository named "ESPHome (dev)". You'll need to install and use that add-on.

zkniebel, 2020-07-17:
Thanks, much appreciated.

closed 2020-07-17 by zkniebel