geoffdavis/esphome-mitsubishiheatpump #17: setRemoteTemperature

Glazzanz, 2020-07-22:
Hi there

Do you support the setRemoteTemperature option from the SwiCago libraries?

Thanks
David

geoffdavis, 2020-09-17:
Sorry I just saw this issue. No, I don't at the moment but I'm interested in adding it, as my units are on exterior walls and tucked behind furniture and the room temperature varies wildly from one side of the room to the other.

Glazzanz, 2020-09-17:
Thanks Geoff - that would be great.  I have the same reasons as you for
wanting this; I have more accurate temperature sources available in HA,
using https://esphome.io/components/sensor/xiaomi_ble.html

IsaacBreuer, 2021-05-16:
any plan on development of this? I am using multiple if this, and also have additional esphome devices collecting temperature from xiaomi vle and different devices. whlukd be nice to have this option either by passing it through html or someway in Homeassistant (similar to genetic thermostat in ha)

geoffdavis, 2021-05-28:
@sijk took care of this today, and I've merged it into release 2.1.0.

closed 2021-05-28 by geoffdavis