geoffdavis/esphome-mitsubishiheatpump #18: Vane position not present

jamescadd, 2020-08-02:
Vane swing mode works on MS-GL09NA, but I don't see an option for vane position in Home Assistant in the details card.  Is it possible to add this or is it an ESPHome limitation?

jamescadd, 2020-08-28:
@geoffdavis any guidance on whether this could be possible, or maybe is it missing from the underlying library?

geoffdavis, 2020-08-29:
I believe it's an ESPHOME limitation - there isn't currently any provision for custom vane positions, though home assistant does support it.

closed 2020-09-17 by geoffdavis

ipetrovits, 2021-01-28:
Hi, Will this issue resolved by https://github.com/home-assistant/core/pull/44495 in HA 2021.2.1 ?

pgenera, 2021-01-28:
> Hi, Will this issue be resolved by home-assistant/core#44495
> <https://github.com/home-assistant/core/pull/44495> in HA 2021.2.1 ?

I believe there also has to be a matching esphome release.

jamescadd, 2021-01-28:
Sadly the comment referencing a new PR was for fan direction, either forward or reverse, not vane position.