geoffdavis/esphome-mitsubishiheatpump #20: Precision

IsaacBreuer, 2020-08-26:
how can i change the precision to get decimal of temperature sensor?

geoffdavis, 2020-08-28:
The short answer is that there's no point.
The Mitsubishi protocol itself only reports whole degrees centigrade, between 16 and 31. Same with the set command, you can only send whole degrees centigrade to the unit.

If you want more accuracy, you can use an external temperature sensor such as one of the Dallas one-wire or the Bosch BMP-280 sensors and hook it into your Esphome config instance. A Wemos D1 Mini should have enough pins to speak serial to the AC and one-wire or I2C to your external sensor, and still fit inside the AC enclosure.

See https://github.com/SwiCago/HeatPump#controlling-the-heat-pump and https://nicegear.nz/blog/hacking-a-mitsubishi-heat-pump-air-conditioner/ for details of the underlying protocol.

closed 2020-08-28 by geoffdavis

IsaacBreuer, 2020-08-31:
Thanks for tbe clarification,  just a unrelated question,  I see all my
Mitsubishis have a offset for temperature setting,  when I set it to 70 it
will actually run as ot set on 68 (this has nothing to do with the esp, it
looks like a setting within the unit itself,  but by having the esp i can
document this behavior with a graph.

I have checked with other people having Mitsubishis they report the same, I
wonder is there is a way around it.

What I mean is, when I set it to 70 (and cool, it will run until it gets 66
and turn off when it gets 68 , )

I know I can always set it with that in mind but its annoying as I set my
whole home to a preset away temperature from hass
(schedy, schedule_prepend,) and have to set different for the Mitsubishis.

jamescadd, 2020-08-31:
I wasn't aware of this, how are you checking the temp on the unit? Mine don't have a display except on the remote but that's a 1 way IR so not accurate.

IsaacBreuer, 2020-08-31:
I have a wired mitibushi thermostat connected to the unit. This one Mitsubishi
PAR-31MAA

geoffdavis, 2020-09-03:
@IsaacBreuer I'm not sure how the PAR-31MAA connects to your unit, but it likely has it's own built-in thermistor. This is separate from the thermistors built into the indoor unit itself. I frequently notice a deviation from the thermistors inside the indoor unit and the thermistors on a separate nodemcu box I have on the opposite side of the room.

Point being - the thermistor inside the unit will likely read something different from an external temperature sensor.

There is a way in the Mitsubishi protocol to tell the indoor unit to override it's version of temperature with that of some external box, but I haven't done that yet in this C++ code.

IsaacBreuer, 2020-09-03:
The temperature reported on the PAR-31MAA always matches the temperature
reported in esphome,

IsaacBreuer, 2020-09-08:
See in this screenshot,  this is reported in HA and matches what is shown
on the physical controll but as you see the blue line (the set temperature
is 70, and the actual temperature fluctuate between 69 and 67 , as if it
whould have been set to 68

IsaacBreuer, 2020-09-08:
[image]

IsaacBreuer, 2020-09-09:
> @IsaacBreuer I'm not sure how the PAR-31MAA connects to your unit, but it likely has it's own built-in thermistor. This is separate from the thermistors built into the indoor unit itself. I [...]

I dont use the sensor in the PAR-31MAA itself,  I have a dedicated PAC-USSEN001-FM-1 Flush Mount Remote Temperature Sensor WALL connected directly to the system.

geoffdavis, 2020-09-17:
@IsaacBreuer i am unable to replicate that behavior on my equipment (listed in the project read me.) You may have discovered either a problem when using CN105 and your external thermostat, or just a weird bug in the logic on your indoor unit itself. If it's a consistent 2 degree F offset, I can add an offset option to the library. Otherwise, as I stated before, I'm only reading the values as reported by the unit itself, including set point.

davidmerrique, 2022-05-22:
@IsaacBreuer have you ever figured out a solution for the temperature offset? I’m also experiencing this with one of my units. 

@geoffdavis an offset option would definitely be helpful!

ripcityhandyman, 2025-01-01:
@davidmerrique @IsaacBreuer @geoffdavis

I'm a bit late to the party, but my MSZ-GL18NA has this same issue. Set at 70deg F, it will heat approximately from 71deg F to 73deg F. I was already planning on using an external sensor via Home Assistant, so my solution has been to simply modify the sensor reading when it is passed along to ESPHOME.

```
# HomeAssistant sensor (must convert from HA units (F) to C)
# offset by 2.0f to account for error in Mitsubishi unit
- platform: homeassistant
  name: "Temperature Sensor From Home Assistant"
  entity_id: sensor.weewx_inside_temperature_office
  on_value:
    then:
      - lambda: 'id(hp).set_remote_temperature((x-32.0+2.0) * 5.0/9.0);' # note the '+2.0' here
  filters:
    - heartbeat: 10s
```

With this modification, you can see that the heat pump is "tricked" into keeping the temperature within approximately +/- 1deg F of the set point.

[image]

Once summer comes around, I will have to confirm the offset is the same direction when cooling. If not, there should be a way to throw together a HA sensor with some templating that will generate the proper offset depending on whether the unit is commanding heating or cooling.

I'll just note that I'm not impressed with the consistency of the heating range, but it looks like this can be attributed to the Mitsubishi itself, which only recognizes temperature in steps of 0.5 deg C 🤷‍♂️