geoffdavis/esphome-mitsubishiheatpump #24: No serial comms between mitsi and d1 mini

Trev-W, 2021-01-01:
Hi

I just implemented this today and the HA <-> D1 side of things is fine. But no comms between the d1 and the mitsi, is there something that needs to be enabled on the mitsi itself to enable the serial comms? I couldn't see any jumpers or dipswitches on the pcb, but just checking.

I couldn't find a suitable connector so had to solder to the back of the cn105 connector and I have continuity on each conductor and no shorts between conductors. I tried to hook up a logic analyser to the rx/tx pins but couldn't see anything but this is also first time using it so I 'm probably doing it completely wrong, and it's just one of those cheap kingst units.

If there is nothing that needs to be enabled on the mitsi itself, my next step will be to try and locate the correct connector, and then use the pullups and a logic level shifter, but others have reported that that's not necessary with the d1 mini, and I'm using a genuine lolin wemos d1 mini.

Any help would be greatly appreciated.  Cheers

`

lounge_ac Uptime | 3176 s |  
-- | -- | --
lounge_ac WiFi Signal | -60 dB |  
lounge_ac ESPHome Version | 1.15.3 Jan 1 2021, 10:40:08 |  
lounge_ac IP | 10.1.0.51 |  
lounge_ac SSID | Tasmota_WAP |  
lounge_ac BSSID | 64:66:B3:A2:DF:72

`
`
[D][MitsubishiHeatPump:231]: control - Was HeatPump updated? YES
[D][climate:262]: 'Lounge AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: AUTO
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: nan°C
[D][climate:282]:   Target Temperature: 24.50°C
[D][sensor:092]: 'lounge_ac WiFi Signal': Sending state -61.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac Uptime': Sending state 175.96899 s with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac WiFi Signal': Sending state -61.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac Uptime': Sending state 235.96899 s with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac WiFi Signal': Sending state -50.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac Uptime': Sending state 295.96201 s with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac WiFi Signal': Sending state -55.00000 dB with 0 decimals of accuracy
[D][sensor:092]: 'lounge_ac Uptime': Sending state 355.96201 s with 0 decimals of accuracy`

geoffdavis, 2021-01-03:
Hi @Trev-W,

I’m using a third party library to communicate with the Mitsubishi units, so I would check there to see what units are supported. There’s a pretty comprehensive thread on GitHub here:

https://github.com/SwiCago/HeatPump/issues/13

The protocol itself is binary, so it’s hard to just throw commands at the UART just to check if the thing is alive. You might play with the baud rate - it’s usually 4800 7E1 but some units run 9600 7E1

Otherwise, most of the genuine Mitsubishi units have a connector in place at CN105.

The connector you want is a JST-XH 5 pin female. Electronics vendors will sell you pre-terminated connectors with bare leads on the other side. In the States, I got mine from Digi-Key. I now have a cheap chinesium knock-off JST crimper, but it’s a pain to use because the pins are so small - just buy the pre-made harness stubs unless you’re doing a bunch of weird ends.

geoffdavis, 2021-01-03:
[Premade cables](https://www.digikey.com/product-detail/en/APAPA22K305/455-3086-ND/6009462)

Order 3. Both ends are terminated so you can cut it in half and get two pigtails

[Connector body](https://www.digikey.com/product-detail/en/PAP-05V-S/455-1489-ND/759977)

miGii, 2021-01-08:
I think D1 Mini v3.0.0 cause for your problems. CH340G on the D1 pulls the line low or high permanently when the module is powered with 5V. There is no resistor in series. 

HP's TX is open-collector, which means that without some form of pull-up the esp cannot receive what the unit is sending.

Use ESP-01 or nodemcu.

bulletmark, 2021-06-12:
I'll leave a note here in case somebody else comes looking ....

I am using the [ESP32 version of the D1 mini](https://github.com/r0oland/ESP32_mini_KiCad_Library) and also could not get comms to work like here so I assumed it had a similar problem. However, I stumbled across [this comment](http://www.weigu.lu/microcontroller/tips_tricks/esp32_tips_tricks/index.html) which mentions Rx and Tx are reversed on that board so I swapped them over and surprisingly that worked. So the pin marked Rx on the ESP32 D1 mini goes to the Rx pin 5 on the CN105, Tx goes to Tx.

awkaplan, 2021-07-01:
> I'll leave a note here in case somebody else comes looking ....
>
> I am using the [ESP32 version of the D1 mini](https://github.com/r0oland/ESP32_mini_KiCad_Library) and also could not get comms to [...]

Confirming @bulletmark 's findings.  I purchased an ACEIRMC Wemos D1 Mini clone on Amazon, and the TX/RX are reversed.  I can also confirm that this project works on the following units _without_ specifying the baudrate as 9600:

- MLZ-KP09NA
- MLZ-KP12NA

lambu76, 2021-07-04:
Hi guys, I have similar problem with my Mitsubishi splitter serie MSZ-BT.

I tried with the official MelCLoud adapter and it works fine as expected.

Currently I'm using ESP8266 Wemos D1 mini v.3 [https://www.amazon.it/gp/product/B08BTYHJM1/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url) w/o pull up resistor so directly connected to port CN105 by using Pigtail cable.

I tried to compile by using both board options: : esp01_1m or d1_mini.

Entity works fine in Hassio, but there is no comunication with the splitter.

I tried to switch TX and RX pin w/o success.

For logger section I set baud_rate equal to 0 and on climate section I set hardware_uart equal to UART0.

I gonna try by using ESP32 board instead [https://www.amazon.it/gp/product/B071P98VTG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url)

Any help will be highly appreciated.

Below my full code.

====================
esphome:
  name: wemos_d1_mini_condizionatore_luigi
  platform: ESP8266
  #board: esp01_1m
  board: d1_mini

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

wifi:
  ssid: "xxxxx"
  password: "xxxx"

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

climate:
  - platform: mitsubishi_heatpump
    name: Condizionatore Luigi
    id: condizionatore
    # Optional
    hardware_uart: UART0
   
    # Optional
    update_interval: 500ms
==================

lambu76, 2021-07-05:
Hi guys,
I tried with and ESP32 board NodeMCU [https://www.amazon.it/gp/product/B071P98VTG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url).
I was able to make it works only configuring as suggested for ES8266 board, so basically disabling logging on UART0 and using for comunicationwith heatpump the UART0.
I tried to use UART2 instead (UART1 seems not mentioned in official documentation for the board) and it was not working.

Below my code:

=======================
esphome:
  name: test_con_luigi
  platform: ESP32
  board: nodemcu-32s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

wifi:
  ssid: "xxxx"
  password: "xxxx"
 

captive_portal:

logger:
  baud_rate: 0

api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time

climate:
  - platform: mitsubishi_heatpump
    name: Test condizionatore luigi

    hardware_uart: UART0

bulletmark, 2021-07-09:
@lambu76 I suspect UART2 doesn't work for anybody with this component. My ESP32 D1 mini works with UART0 (subject to me swapping Tx/Rx as I [mention above](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/24#issuecomment-859968594)). As a test after your comment, I changed the connections and config to UART2 but it did not work with this component. However, I put my own software on the same ESP32 board, and that software can communicate correctly to the unit via UART2 with the same connections, so we know it is not a hardware issue.

lambu76, 2021-07-09:
ok thx.

btw I will continue to use connected on UART0, it's not a problem.

thx for your tests

DeltaAngle, 2021-07-09:
@bulletmark I'm using the same ESP32 D1 Mini and have UART2 working. Coincidentally, I originally had it wired to UART0 but was unaware of the swapped Rx/Tx. When it inevitably failed, I reconfigured to UART2 with no issues. So, I guess YMMV. 🤷

bulletmark, 2021-07-10:
@DeltaAngle that's odd compared to me. So you changed only `hardware_uart: UART0` to `hardware_uart: UART2` in your yaml file? Are you using the latest version of this component, and of esphome?

With reference to [this drawing](https://i1.wp.com/www.esp32learning.com/wp-content/uploads/2018/12/MH-ET_LIVE_D1_mini_ESP32_pinout.png), which exact pins on your ESP32 D1 did you wire to the heatpump CN105 pin 5 Rx, and to CN105 pin 4 Tx?

DeltaAngle, 2021-07-12:
@bulletmark That's correct, `hardware_uart: UART2` is set in my YAML. I'm running ESPHome v1.19.4 with v2.2.0 of this component.

My ESP32 is wired to IO16 (u2RX) and IO17 (u2TX) which I identified based on this [map](https://i0.wp.com/www.esp32learning.com/wp-content/uploads/2017/12/esp32minikit.jpg).

geoffdavis, 2022-09-21:
I believe this is answered. I'm going to close it out.

closed 2022-09-21 by geoffdavis