geoffdavis/esphome-mitsubishiheatpump #30: Flashing on ESP_01S board ends up in compilation error

ipetrovits, 2021-05-28:
Hi, first of all, great update!!

I was excited to try out the new 2.1 release, and here's what I got (seems like a conflict with esphome's native code for esp8266 chips?!):

```yaml
In file included from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:26:0,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:30,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecure.h:41,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41,
                 from src/esphome/components/wifi/wifi_component.h:18,
                 from src/esphome.h:58,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h: In static member function 'static time_t fs::FS::_defaultTimeCB()':
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:243:49: error: reference to 'time' is ambiguous
     static time_t _defaultTimeCB(void) { return time(NULL); }
                                                 ^
In file included from /root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/wchar.h:4:0,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/cwchar:44,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/postypes.h:40,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/char_traits.h:40,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/string:40,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:2,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/../include/time.h:53:18: note: candidates are: time_t time(time_t*)
 time_t    _EXFUN(time,     (time_t *_timer));
                  ^
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4:0,
                 from src/esphome.h:21,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/time/real_time_clock.h:11:16: note:                 namespace esphome::time { }
 namespace time {
                ^
```

ipetrovits, 2021-05-28:
> I'm running 3 ESP_01S boards at home and I'm not seeing this issue. I'm guessing it's old cached data from previous compilations.
> 
> I had to nuke _everything_ in my `esphome` source directory that I didn't put there (i.e. it wasn't my YAML files or include files) in order to get this to build, because ESPHome and PlatformIO seem to cache things in all manner of random locations.
> 
> In particular, I had to nuke:
> 
> * under `/config/esphome/` on my HASS.IO box
>   ** `.esphome/` - note the dot
>   ** `nodeFoo` - where `nodeFoo` is the ESPHome build artifacts directory corresponding to `nodeFoo.yaml`
> * I also had to run the "Clean Build Artifacts" task for the individual node to ensure the `.platformio` stuff got cleaned out
> 
> Also, make sure you're using ESPHome 1.18.0.
> 
> Note that I'm running ESPHome from a Hass.IO install on a Raspberry Pi 4 using the [pre-Nabu Casa takeover add-on from GitHub](https://github.com/esphome/hassio). It _should_ be identical to the version in the Home Assistant Community Add-ons repository.

ipetrovits, 2021-05-28:
Sorry i clicked the wrong button.. 

I was able to compile only if i removed **time:** entry from the configuration.

Fortunately i won't use actual time component in this device, so not a big deal..
But in fact even after nuking the whole esphome setup i was unable to compile the example from the **README.md** if **time:** was used.
btw I'm using 1.18.0 in hass.io addon.

geoffdavis, 2021-05-28:
I'm not quite sure what's going on. I hate to be that guy that says "works on my machine, I'm going to lunch", but it works on my machine and I'm going to lunch.

That's with the slight caveat that I have to comment out the unused "libraries" and "includes" in the demo yaml config, which should be corrected on github shortly.

Here's my test YAML file:
```
esphome:
  name: hptest
  platform: ESP8266
  board: esp01_1m
  # Boards tested: ESP-01S (ESP8266), Wemos D1 Mini (ESP8266); ESP32 Wifi-DevKit2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "HPTest Fallback Hotspot"
    password: !secret fallback_password

# Note: if upgrading from 1.x releases of esphome-mitsubishiheatpump, be sure
# to remove any old entries from the `libraries` and `includes` section.
#libraries:
  # Remove reference to SwiCago/HeatPump

#includes:
  # Remove reference to src/esphome-mitsubishiheatpump

captive_portal:

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

# Enable Web server.
web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: hptest ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: hptest IP
    ssid:
      name: hptest SSID
    bssid:
      name: hptest BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: hptest Uptime
[... 210 lines cut ...]
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs/spiffs_cache.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/spiffs_api.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/sqrt32.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/time.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/uart.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling /data/hptest/.pioenvs/hptest/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving /data/hptest/.pioenvs/hptest/libFrameworkArduino.a
Linking /data/hptest/.pioenvs/hptest/firmware.elf
Retrieving maximum program size /data/hptest/.pioenvs/hptest/firmware.elf
Checking size /data/hptest/.pioenvs/hptest/firmware.elf

RAM:   [=====     ]  48.3% (used 39560 bytes from 81920 bytes)
Flash: [====      ]  44.5% (used 455632 bytes from 1023984 bytes)
Creating BIN file "/data/hptest/.pioenvs/hptest/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/hptest/.pioenvs/hptest/firmware.elf"
======================== [SUCCESS] Took 190.52 seconds ========================
INFO Successfully compiled program.
```

ipetrovits, 2021-06-18:
Still unknown, however removing time did the trick.. Won't bother me anymore :)

closed 2021-06-18 by ipetrovits