geoffdavis/esphome-mitsubishiheatpump #31: external sensor readme need update

IsaacBreuer, 2021-05-28:
o followed the new instructions to use external temperature sensor,  works great,  but o believe it extremely important to add to instructions that if your HA is set to use Fahrenheit as mine, you have to convert the ha value before passing it to HeatPump 
as         
  - lambda: 'id(hp).set_remote_temperature((x-32) * 5/9);'
instead of 
          - lambda: 'id(hp).set_remote_temperature(x);'

IsaacBreuer, 2021-05-28:
but still after converting.  I am not getting the exact temperature,  as I don't know the rounding this set remote temperature does. 
is it set to .5 C increments?

IsaacBreuer, 2021-05-28:
see here my log from esphome that current temperature get randomly set to different what the set_remote_temperature  did 

['sensor.kitbt_temperature': Got state 66.70
[D][sensor:099]: 'kitbt from homeassistant': Sending state 66.70000  with 1 decimals of accuracy
[D][MitsubishiHeatPump:410]: Setting remote temp: 19.3
[D][sensor:099]: 'kit ac Uptime': Sending state 6765.68115 s with 0 decimals of accuracy
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[D][homeassistant.sensor:019]: 'sensor.kitbt_temperature': Got state 66.70
[D][sensor:099]: 'kitbt from homeassistant': Sending state 66.70000  with 1 decimals of accuracy
[D][MitsubishiHeatPump:410]: Setting remote temp: 19.3
[D][sensor:099]: 'kit ac Uptime': Sending state 6825.68311 s with 0 decimals of accuracy
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[D][homeassistant.sensor:019]: 'sensor.kitbt_temperature': Got state 66.90
[D][sensor:099]: 'kitbt from homeassistant': Sending state 66.90000  with 1 decimals of accuracy
[D][MitsubishiHeatPump:410]: Setting remote temp: 19.4
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 20.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][sensor:099]: 'kit ac Uptime': Sending state 6885.68311 s with 0 decimals of accuracy
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 20.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 19.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 19.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[D][sensor:099]: 'kit ac Uptime': Sending state 6945.68701 s with 0 decimals of accuracy
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 20.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][homeassistant.sensor:019]: 'sensor.kitbt_temperature': Got state 67.10
[D][sensor:099]: 'kitbt from homeassistant': Sending state 67.10000  with 1 decimals of accuracy
[D][MitsubishiHeatPump:410]: Setting remote temp: 19.5
[D][sensor:099]: 'kit ac Uptime': Sending state 7005.68213 s with 0 decimals of accuracy
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 19.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][sensor:099]: 'kit ac Uptime': Sending state 7065.68701 s with 0 decimals of accuracy
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: IDLE
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 20.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][climate:262]: 'Kit AC' - Sending state:
[D][climate:265]:   Mode: COOL
[D][climate:267]:   Action: COOLING
[D][climate:270]:   Fan Mode: HIGH
[D][climate:273]:   Swing Mode: OFF
[D][climate:276]:   Current Temperature: 20.50°C
[D][climate:282]:   Target Temperature: 21.00°C
[D][sensor:099]: 'kit ac WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy
[D][homeassistant.sensor:019]: 'sensor.kitbt_temperature': Got state 67.10
[D][sensor:099]: 'kitbt from homeassistant': Sending state 67.10000  with 1 decimals of accuracy
[D][MitsubishiHeatPump:410]: Setting remote temp: 19.5

geoffdavis, 2021-05-28:
Try forcing the math to be floats all the way through the conversion process by using decimal numbers instead of integers:

```- lambda: 'id(hp).set_remote_temperature((x-32.0) * 5.0/9.0);'```

I have to admit I got lazy and just switched my US-based install to "_degrees of **C**ommunism_" instead of "_degrees of **F**reedom_" to avoid a bunch of the artifacting issues with temperature conversions. Wife acceptance factor was diminished, but she got used to it eventually.

IsaacBreuer, 2021-05-28:
unfortunately  forcing the decimal point in the conversion did not resolve
the issue, and for me to change my house to Communists setting as a no no
for WAF.

IsaacBreuer, 2021-05-28:
Changing the system to C did not solve it for me either.
i have also tried the following , exposing a HA sensor that is rounded to
.5 C just to rule out rounding  when my sensor actually show 20.70 , and
was not resolved

sijk, 2021-05-28:
This could possibly be something to do with your heat pump model. From upstream issues (e.g. SwiCago/HeatPump#69, SwiCago/HeatPump#144) it sounds like there are two methods used by different models to set the remote temperature, and who knows if that covers everything? Either way, the relevant code is at https://github.com/SwiCago/HeatPump/blob/master/src/HeatPump.cpp#L248 if that helps.

IsaacBreuer, 2021-05-30:
The function gets through successfully,  but then after like 30 seconds
reverts back , so I added a automation in HA to update a attribute for that
particular sensor ever 25 seconds,  so far this work,  but is just a
workaround

javilopezalarcon, 2021-12-05:
> The function gets through successfully, but then after like 30 seconds reverts back , so I added a automation in HA to update a attribute for that particular sensor every 25 seconds, so far this work, but is just a workaround
> […](#)

How do you do to update the temperature of the external sensor every 25 seconds from ha? I'm with the same problem, I send him the average temperature at home but after a bit he takes the temperature of his sensor (the one on the remote control or the one on the air return, I tried both)

akifbayram, 2022-08-06:
> The function gets through successfully, but then after like 30 seconds reverts back , so I added a automation in HA to update a attribute for that particular sensor every 25 seconds, so far this work, but is just a workaround

Can you share the automation you used to update the sensor? My attempts aren't updating the whole sensor. I'm having the same issue with my MVZ air handler. Reverts to onboard temp sensor after ~3 minutes.

closed 2025-03-31 by geoffdavis