geoffdavis/esphome-mitsubishiheatpump #33: Missing vertical (and horizontal in general) swing mode steps

staxDB, 2021-06-10:
Hey there, 

thank you so much for the great integration in Home Assistant!
Currently I‘m missing the option to set the fan swing modes of my heatpump.
Mine currently supports a vertical and a horizontal direction in different steps. As I found in the original [source code](https://github.com/SwiCago/HeatPump), there are options to customize these kind of settings:

```json
heatpumpSettings settings = {
    "ON",  /* ON/OFF */
    "FAN", /* HEAT/COOL/FAN/DRY/AUTO */
    26,    /* Between 16 and 31 */
    "4",   /* Fan speed: 1-4, AUTO, or QUIET */
    "3",   /* Air direction (vertical): 1-5, SWING, or AUTO —> how to set these in HA? */
    "|"    /* Air direction (horizontal): <<, <, |, >, >>, <>, or SWING  —> how to set these in HA? */
}; 
```

As far as I understand, I‘m currently only able to set this setting: `swing_mode: [OFF, VERTICAL]`

Is there a way to integrate these settings via HA?

geoffdavis, 2021-06-11:
You'll want to fiddle around with the [advanced options] that were recently added in the 2.x release.

In particular:
* supports (Optional): Supported features for the device.
  * mode (Optional, list): Supported climate modes for the HeatPump. Default: ['AUTO', 'COOL', 'HEAT', 'DRY', 'FAN_ONLY'] 
  * fan_mode (Optional, list): Supported fan speeds for the HeatPump. Default: ['AUTO', 'DIFFUSE', 'LOW', 'MEDIUM', 'MIDDLE', 'HIGH'] 
  * swing_mode (Optional, list): Supported fan swing modes. Most Mitsubishi units only support the default. Default: ['OFF', 'VERTICAL']

[advanced options]: https://github.com/geoffdavis/esphome-mitsubishiheatpump#advanced-configuration

geoffdavis, 2021-06-11:
Further research shows that the esphome protocol doesn't support sending custom climate modes to home assistant either, per https://github.com/esphome/esphome/blob/dev/esphome/components/api/api.proto

Any fine-grained swing mode control will require multiple changes in at least the ESPHome code base in order to support either custom swing modes or separate horizontal and vertical swing modes.

wavswavs, 2021-07-06:
> After digging around in the code for this component, and ESPHome's [climate component](https://github.com/esphome/esphome/blob/beta/esphome/components/climate/climate_mode.h) code, there's no way in Esphome to get granular on the horizontal vane position. There is a way however to say you want either horizontal swing, vertical swing, or both swing, but the HORIZONTAL and BOTH modes [aren't currently supported] (https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/master/components/mitsubishi_heatpump/espmhp.cpp#L222) in this custom component.
> 
> I'm happy to try and add support for HORIZONTAL and BOTH, but I have no way to test it as my equipment doesn't support it.

Hi, 
I'm using your integration in four devices, two of them (MSZ-AP35VG wall unit) supports HORIZONTAL, VERTICAL and BOTH modes. 
If you add HORIZONTAL and BOTH support I could test it.

I also have an MFZ-KT25VG floor unit with a differet vane behaviour that could be test.

Do you know how ask esphome dev team to add granular vane position? Could be possible to add some positions in a separate switch?

Great job!

F43RY, 2021-07-07:
Hi,
I tried to patch core climate esphome in order to add horizontal and vertical position controls. I can't test the code. If someone is intrested and want check can have a look at this issue in esphome github project
[esphome#1968](https://github.com/esphome/esphome/pull/1968)

WeekendWarrior1, 2021-07-13:
If [https://github.com/esphome/esphome/pull/1998](https://github.com/esphome/esphome/pull/1998) gets merged then there will be an easy template for your codebase to do this, with [https://github.com/esphome/esphome/pull/1998/commits/531f5f7b7066a913e638913118eb72938889ef1b](https://github.com/esphome/esphome/pull/1998/commits/531f5f7b7066a913e638913118eb72938889ef1b), climate_ir.h will have a function to set supported swing modes.
The Pansonic climate currently has them as yaml options eg:
```yaml
climate:
  - platform: panasonic
    name: "Lounge AC"
    supports_horizontal_swing: true
```

wavswavs, 2021-07-13:
> Hi,
> I tried to patch core climate esphome in order to add horizontal and vertical position controls. I can't test the code. If someone is intrested and want check can have a look at this issue in esphome github project
> [esphome#1968](https://github.com/esphome/esphome/pull/1968)

I don't know how to test with a development version of esphome. Also, I think that we need the code in this library too in order to can test hvac with esphome, right?

jhuizingh, 2021-11-09:
I have equipment that would work with the vane positions and I am interested in getting this feature implemented. It's been a while since this discussion happened. What's the current state of this? I might be able to help.

kalosbg, 2021-12-01:
My ac also supports horizontal and vertical swing if, you need to test something feel free to reach me.

jhuizingh, 2021-12-02:
@kalosbg I asked about this on the developers channel on the ESPHome discord. It seems that the way that the mitsubishi units work doesn't fit in with the model of how Home Assistant (and by extension ESPHome) think about `climate` devices. So it's unlikely that full support will be added for our devices any time soon. I intend to make a custom controller that can be used by ESPHome that doesn't concern itself with implementing a `climate` device and instead is completely custom. I'll share it and post back here if I'm able to make that happen in my spare time.

kalosbg, 2021-12-02:
Thank you for the feedback back!

earwin, 2021-12-02:
I did not dig into HA internals yet, but Mitsubishi cloud integration somehow manages to expose horizontal/vertical vanes, see: https://www.home-assistant.io/integrations/melcloud/#climate

It does it via ... attributes? Which are apparently custom attributes (e.g. not defined by climate device class). And on a second look, climate defines very few attributes, and instead uses properties + service setters. Which is different in HA and mightily confusing! (offtopic: anybody has a link that explains the difference and motivations?)

But anyway, we can do the same and expose attributes for everything that climate misses?

geoffdavis, 2021-12-03:
A pre-work thought on this - I can’t think of a way in ESPHOME to set
additional attributes on a particular device, but it’s been a while since
I’ve dug into the internals. We can however provide the additional settings
as a second or third device and register it as a “cover” in Home Assistant.

If someone with more time and available brain cells wants to take a stab at
this, or knows of a better solution for adding attributes to a device, let
me know.

earwin, 2021-12-03:
Okay, I had a quick look @ ’api.proto’, and there is no attribute support whatsoever.

If we want to implement this given existing model, my vote is on spawning a secondary ’fan’ entity for horizontally-endowed units. It has these nicely fitting oscillation and position fields, and irrelevant ones can be disabled.

I may take a stab at it, but don't hold your breath, I have an attention span of a toothpick.

cg089, 2021-12-22:
> After digging around in the code for this component, and ESPHome's [climate component](https://github.com/esphome/esphome/blob/beta/esphome/components/climate/climate_mode.h) code, there's no way in Esphome to get granular on the horizontal vane position

Hi, I also came across this issue. However, I only have vertical position.
What would be the code to use only those different vertical positions?

cg089, 2021-12-23:
Is it because in espmhp.cpp in line 221 is commented out?
This is basically just what's missing!

 //const char* VANE_MAP[7]        = {"AUTO", "1", "2", "3", "4", "5", "SWING"};

jhuizingh, 2021-12-23:
@cg089 the issue as I understand it is that the climate component in esphome can't even represent the possible states of the Mitsubishi units. [See Here. ](https://esphome.io/components/climate/index.html)

The code to do it exists in the library that interacts with  the heat pump units, so that part isn't particularly hard. It's just representing it within esphome in a way that isn't completely counterintuitive that is the challenge.

cg089, 2021-12-24:
> @cg089 the issue as I understand it is that the climate component in esphome can't even represent the possible states of the Mitsubishi units. [See Here. ](https://esphome.io/components/climate/index.html)
> 

No, I don't think so. I am experimenting with two Mitsubishi AC. Both are integrated in Home Assistant climate, one with this esphome here and the other with mqtt (via https://github.com/gysmo38/mitsubishi2MQTT ).

While the mqtt climate is able to set all vertical positions:
[image]

the esphome component only shows "Off" and "Vertical".
[image]

The dropdown field however allows for more options to show in Home Assistant.

jhuizingh, 2021-12-24:
Right, that's exactly what I'm saying. It's entirely possible to control the devices fully with the c++ library that is being used under the covers of this extension. The issue is that esphome has a generic device category called "climate" which is what this package is using. 

The climate device type has generic options which they hoped would cover all possible climate devices that people would want to integrate into esphome. Clearly the "off" and "vertical" swing mode options don't cover the 1-5 and auto and swing plus some horizontal choices too that Mitsubishi units need to fully cover the bases. They are unlikely to add additional options into the climate device type because then they'd have to add more for every single other device that also doesn't fit, and it would start to look crazy.

The choices for us are either to be limited to the options within the climate device type, or represent the options that don't fit in a different way.

jamescadd, 2021-12-24:
Would a standalone sensor/entity at least expose the functionality? I'd be fine with some jank in the UI just to have the setting available, like a separate drop-down to choose the value.

cg089, 2021-12-29:
> Would a standalone sensor/entity at least expose the functionality? I'd be fine with some jank in the UI just to have the setting available, like a separate drop-down to choose the value.

Yes, I think that would be the only "workaround", which would be sufficent and better than nothing.

scootec, 2022-01-11:
We could use the newish select entity in [Home Assistant](https://developers.home-assistant.io/docs/core/entity/select/) and [ESPHome](https://esphome.io/components/select/index.html) to make those options available.

geoffdavis, 2022-02-01:
> We could use the newish select entity in [Home Assistant](https://developers.home-assistant.io/docs/core/entity/select/) and [ESPHome](https://esphome.io/components/select/index.html) to make those options available.

Probably worth a shot. It seems a bit cleaner than using the method described in the [stepper motor documentation for esphome](https://esphome.io/components/stepper/index.html#stepper-ha-config).

jcam, 2022-07-27:
As an interim solution, I used swing "vertical" for position 5, "horizontal" for position 2, and "both" for vertical swing.

This doesn't solve the problem of controlling the horizontal position, but at least gives me the two positions i typically use.
https://github.com/jcam/esphome-mitsubishiheatpump/commit/8e4ef6d21b17338fcbd27a5621f4f64463f195c1

In that same branch I also overrode the fan speeds included for passing up to the climate object in home assistant, since there's no way to correctly set the order of the speeds and i didn't want 1, 2, 4, 3, quiet.

I then used the simple thermostat HACS custom component to rename them

[image]

jamescadd, 2022-07-27:
This is awesome, I can now hack together what I want to get around HA limitations. Thanks for writing this up!

jcam, 2022-07-27:
If you or someone else wants to leverage my work, feel free!

Just update your external component in esphome like so:
`external_components:
  - source: github://jcam/esphome-mitsubishiheatpump@overload-vane-names`

And, set this for your lovelace card:
`type: custom:simple-thermostat
entity: climate.family_room_minisplit
control:
  hvac:
    heat_cool: false
    heat: false
  fan:
    _name: Fan
    _hide_when_off: true
    low:
      name: Quiet
      icon: mdi:fan-minus
    medium:
      name: 1
      icon: mdi:fan-speed-1
    high:
      name: 2
      icon: mdi:fan-speed-2
    focus:
      name: 3
      icon: mdi:fan-speed-3
    diffuse:
      name: 4
      icon: mdi:car-turbocharger
  swing:
    _name: Angle
    _hide_when_off: true
    'off':
      name: Auto
      icon: mdi:autorenew
    vertical:
      name: Vertical
      icon: mdi:dots-vertical
    horizontal:
      name: Horizontal
      icon: mdi:dots-horizontal
    both:
      name: Swing
`

mattwolkoff, 2022-08-23:
@jcam I just wanted to say thanks! this worked great for me. It's may not be ideal for flexibility, but fits my use case well(set to one position, and leave it for the season).

jcam, 2022-08-23:
Yay! Yeah with the esphome/HA climate entity we get 4 "choices", so I decided to have auto, swing, near vertical and near horizontal, as 95% of the time that's enough. 
I might tweak things once we start the heating season, and might also look to add a second entity for horizontal, though i only have horizontal on one of the wall units, so not missing too much there...

aleksanderlvr, 2023-03-04:
Thanks to both authors. Jesse, is there a way to change the code so that the horizontal position at 1 is vertical at 2?

aleksanderlvr, 2023-03-05:
> As an interim solution, I used swing "vertical" for position 5, "horizontal" for position 2, and "both" for vertical swing.
> [image]

Thanks to both authors. Jesse, is there a way to change the code so that the horizontal position is at 1 and vertical at 2?

seime, 2023-06-19:
I've created a PR with this missing functionality if someone would like to test; https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/103

Smandurlo, 2023-07-16:
With tasmota and mqtt I can use any combination of hvac mode/fan/swing orizzontal and vertical I want. What if we use the mqtt way? Will esphome be able to use those settings from mqtt and setup the ac?

jhuang0, 2023-08-04:
> If you or someone else wants to leverage my work, feel free!
>
> Just update your external component in esphome like so:
>
> ```
> external_components:
>   - source: [...]

@jcam - I'm literally only hours into my ESP journey so forgive me if I'm not understanding some of the intricacies here. What are your thoughts on setting the thermostat to the midpoint (3 or 4) every time the climate control is turned on? For the average user, this would in effect give an additional state in addition to the ones you've provided. The end user would then be able to override that selection (or employ an automation to select away from it.)

SGXander, 2023-11-20:
Jumping on the bottom of this thread so sorry if I've missed an important bit but is there a list of supported options for "swing_mode:" somewhere? and did anyone manage to get vertical/horizontal vane step control working at all?

supermarkert, 2023-11-20:
Yeah, I kind of have the same question. I only need Vertical, but it's unclear what the individual step modes that are supported (if any). My mini-split supports Auto, 1-5, and Swing, but there doesn't seem to be any notes about what can be configured except simply ('OFF', 'VERTICAL'). What will the unit interpret that as? AUTO?

SGXander, 2023-11-20:
I might try setting the ESPHome config swing_modes to the below and see if it works. I'd love to set the fan modes to something custom too as medium and middle are confusing my wife already :D 

Output from melcloud integration for one of my splits:
```
hvac_modes: off, heat, dry, cool, fan_only, heat_cool
min_temp: 10
max_temp: 31
target_temp_step: 0.5
fan_modes: auto, 1, 2, 3, 4, 5
swing_modes: auto, 1_up, 2, 3, 4, 5_down, swing
vane_horizontal_positions: auto, 1_left, 2, 3, 4, 5_right, split, swing
vane_vertical_positions: auto, 1_up, 2, 3, 4, 5_down, swing
supported_features: 41
```

rudizl, 2023-11-20:
> I'd love to set the fan modes to something custom too as medium and middle are confusing my wife already :D
> 
Your wife is not alone in confusion🤣 Not to mention that the gouges are the same🫤
[image]
And even worse - in my language someone has decided to translate both the same way 🤣
[image]

jcam, 2023-11-20:
My mod is still working :)

[image]

jcam, 2023-11-21:
Because ESPHome + homeassistant together only allow specific things to come through, and individual controls of the vanes aren't part of that, I overloaded its use so that i can choose auto + swing modes as well as stepped vane positions 2 and 5, all on the vertical.
I also changed the names of the fan speed settings so that they are easy to understand for me and my family (quiet and 1-4... no "middle" speed)

echavet, 2024-01-10:
I've just updated my version of the esphome-mitsubishiheatpump component which is so different that I've changed the project name:
https://github.com/echavet/MitsubishiCN105ESPHome
It is not tested as the original project, but I'm currently using it with esp8266 w1 module.
It includes mitsubishi van orientations. 
I've had an extraComponents which adds a select component and a compressor_freq one

closed 2025-03-31 by geoffdavis