geoffdavis/esphome-mitsubishiheatpump #40: Entities diseappered after ESPHOME upgrade to 1.19.4

lambu76, 2021-06-25:
Hi,
I performed the official upgrade of ESPHome core library to last version in subject and immediately after the entities are not anymore available for the device integrated in HomeAssitant.

What is strange is that during the compiling a lot fo WARNING for deprecated methods are risen in addition to the CHANGE LOG related to this new version or ESPHome some changes on HeatPump management have been included.

Can you please check it ?

thx

nao-pon, 2021-07-02:
I have a similar problem. HomeAssistant does not recognize all entities, including Climate, in the following environments:

HomeAssistant: core-2021.6.6
ESP Home: 1.19.4 (Same on 1.19.1, 1.19.0)
esphome-mitsubishiheatpump: 2.2.0

ESP Home 1.18.0 + esphome-mitsubishiheatpump: 2.1.0 works fine.

The Home Assistant log contains the following:

```
2021-07-02 13:25:43 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 214, in on_login
    entity_infos, services = await cli.list_entities_services()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
    entities.append(cls(**kwargs))
  File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in __init__
    self.supported_modes = __attr_converter_supported_modes(supported_modes)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/enum.py", line 339, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.8/enum.py", line 662, in __new__
    raise ve_exc
ValueError: 6 is not a valid ClimateMode
```

Note that the compile-time warning is output with ESP Home 1.19.4. However, this may not be relevant to this issue.

```
src/main.cpp: In function 'void setup()':
src/main.cpp:218:66: warning: 'void esphome::climate::ClimateTraits::set_supports_auto_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:54): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_auto_mode(true);
                                                                  ^
src/main.cpp:219:66: warning: 'void esphome::climate::ClimateTraits::set_supports_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:56): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_cool_mode(true);
                                                                  ^
src/main.cpp:220:66: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:58): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_heat_mode(true);
                                                                  ^
src/main.cpp:221:65: warning: 'void esphome::climate::ClimateTraits::set_supports_dry_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:66): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_dry_mode(true);
                                                                 ^
src/main.cpp:222:70: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_only_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:62): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_only_mode(true);
                                                                      ^
src/main.cpp:223:70: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_auto(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:80): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_auto(true);
                                                                      ^
src/main.cpp:224:73: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_diffuse(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:92): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_diffuse(true);
                                                                         ^
src/main.cpp:225:69: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_low(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:82): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_low(true);
                                                                     ^
src/main.cpp:226:72: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_medium(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:84): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_medium(true);
                                                                        ^
[... cut ...]
   mitsubishiheatpump->config_traits().set_supports_swing_mode_off(true);
                                                                       ^
src/main.cpp:230:76: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_vertical(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:135): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_swing_mode_vertical(true);
```

lambu76, 2021-07-02:
exactly the same....

aqualx, 2021-07-02:
Everything is working as expected with latest HA (core-2021.6.6) and ESPHome (1.19.4) after [Fix esphome v1.19 changes](https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/38).
Try clean build folder "/config/esphome/%YOUR_FOLDER%" & rebuild firmware.

nao-pon, 2021-07-03:
@aqualx Yes, After cleaning the contents of "/config/esphome/% YOUR_FOLDER%/" and "/config/esphome/.esphome/external_components/", I did a clean build and it worked fine. Thank you!

However, the following build error will be notified and I think it needs to be fixed for the future.

```
Compiling /data/mitsubishi_ac/.pioenvs/mitsubishi_ac/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
src/esphome/components/climate/climate.cpp: In constructor 'esphome::climate::Climate::Climate(const string&)':
src/esphome/components/climate/climate.cpp:431:58: warning: 'esphome::climate::Climate::away' is deprecated (declared at src/esphome/components/climate/climate.h:196): away is deprecated, use preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : Nameable(name) {}
                                                          ^

Compiling /data/mitsubishi_ac/.pioenvs/mitsubishi_ac/src/esphome/components/sensor/automation.cpp.o
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In constructor 'MitsubishiHeatPump::MitsubishiHeatPump(HardwareSerial*, uint32_t)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:40:42: warning: 'void esphome::climate::ClimateTraits::set_supports_away(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:119): This method is deprecated, use set_supported_presets() instead [-Wdeprecated-declarations]
     this->traits_.set_supports_away(false);
                                          ^

Compiling /data/mitsubishi_ac/.pioenvs/mitsubishi_ac/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:216:71: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:60): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_heat_cool_mode(true);
                                                                       ^
src/main.cpp:217:66: warning: 'void esphome::climate::ClimateTraits::set_supports_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:56): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_cool_mode(true);
                                                                  ^
src/main.cpp:218:66: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:58): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_heat_mode(true);
                                                                  ^
src/main.cpp:219:65: warning: 'void esphome::climate::ClimateTraits::set_supports_dry_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:66): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_dry_mode(true);
                                                                 ^
src/main.cpp:220:70: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_only_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:62): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_only_mode(true);
                                                                      ^
src/main.cpp:221:70: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_auto(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:80): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_auto(true);
                                                                      ^
src/main.cpp:222:73: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_diffuse(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:92): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_diffuse(true);
                                                                         ^
src/main.cpp:223:69: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_low(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:82): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_low(true);
                                                                     ^
src/main.cpp:224:72: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_medium(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:84): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_fan_mode_medium(true);
                                                                        ^
[... cut ...]
                                                                       ^
src/main.cpp:228:76: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_vertical(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:135): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
   mitsubishiheatpump->config_traits().set_supports_swing_mode_vertical(true);
                                                                            ^
```

jamescadd, 2021-07-07:
I've tried cleaning and manually removing the directory with the .yaml file name, but still get the "reference to 'time' is ambiguous' error.  I did the directory removal in Visual Studio Code.  I also can't find the directory from the build output - /root/.platformio does not exist.  Is there somewhere else I should look to remove the old library files?  I feel like at one point I may have manually copied them to the HA install but I can't remember where.

`Compiling /data/mitsubishi_dining_room/.pioenvs/mitsubishi_dining_room/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/mitsubishi_dining_room/.pioenvs/mitsubishi_dining_room/src/esphome/components/json/json_util.cpp.o
Compiling /data/mitsubishi_dining_room/.pioenvs/mitsubishi_dining_room/src/esphome/components/logger/logger.cpp.o
Compiling /data/mitsubishi_dining_room/.pioenvs/mitsubishi_dining_room/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
Compiling /data/mitsubishi_dining_room/.pioenvs/mitsubishi_dining_room/src/esphome/components/ota/ota_component.cpp.o
In file included from /root/.platformio/lib/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:27:0,
                 from src/esphome/components/web_server_base/web_server_base.h:5,
                 from src/esphome/components/web_server/web_server.h:5,
                 from src/esphome.h:35,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h: In static member function 'static time_t fs::FS::_defaultTimeCB()':
/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:243:49: error: reference to 'time' is ambiguous
     static time_t _defaultTimeCB(void) { return time(NULL); }`

geoffdavis, 2022-09-21:
Cleaning the build directory was noted in the README. Closing this out.

closed 2022-09-21 by geoffdavis