geoffdavis/esphome-mitsubishiheatpump #43: MQTT topics to set config

barcar, 2021-09-19:
Does anybody have a sample config showing how to use MQTT to set config (target temp, mode, etc)? I’m assuming esphome climate does the heavy lifting and I can subscribe to topics with no issues but I just cannot figure out the topics needed to set values.

any pointers gratefully received.

closed 2021-09-20 by barcar

Marc--, 2021-10-19:
@barcar Hello, i have exactly the same question and feel stupid , did you find a solution ?

Marc--, 2021-10-20:
Thanks for your very quick reply !

I'm a real noob, but i manage to make it work ! The last thing to not forget is... MQTT !!

Thank you for your help with "set" command and thanks to the developpers of this project !