geoffdavis/esphome-mitsubishiheatpump #47: Double declaration of delay() function

F43RY, 2021-10-24:
Hi,
I tried to upgrade esphome to latest release in my esp01 in my heatpump by I got issues at compile time.
I deleted old folder with old files but the issue came up again. Here below the full log of esphome dashboard.
No issues with past release of esphome.
I think the issue is a conflict due to a double declaration of function delay() in two different librearies.

`INFO Reading configuration ./heatpump_camera.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing heatpump-camera (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3

Downloading  [------------------------------------]    0%
Downloading  [#######-----------------------------]   20%
Downloading  [##############----------------------]   40%
Downloading  [#####################---------------]   60%
Downloading  [############################--------]   80%
Downloading  [####################################]  100%

Unpacking  [------------------------------------]    0%
Unpacking  [#-----------------------------------]    3%
Unpacking  [##----------------------------------]    6%
Unpacking  [###---------------------------------]    9%
Unpacking  [####--------------------------------]   12%
Unpacking  [#####-------------------------------]   15%
Unpacking  [######------------------------------]   18%
Unpacking  [#######-----------------------------]   21%
Unpacking  [#########---------------------------]   25%
Unpacking  [##########--------------------------]   28%
Unpacking  [###########-------------------------]   31%
Unpacking  [############------------------------]   34%
Unpacking  [#############-----------------------]   37%
Unpacking  [##############----------------------]   40%
Unpacking  [###############---------------------]   43%
Unpacking  [################--------------------]   46%
Unpacking  [##################------------------]   50%
Unpacking  [###################-----------------]   53%
Unpacking  [####################----------------]   56%
Unpacking  [#####################---------------]   59%
Unpacking  [######################--------------]   62%
Unpacking  [#######################-------------]   65%
Unpacking  [########################------------]   68%
Unpacking  [#########################-----------]   71%
Unpacking  [###########################---------]   75%
Unpacking  [############################--------]   78%
Unpacking  [#############################-------]   81%
Unpacking  [##############################------]   84%
Unpacking  [###############################-----]   87%
Unpacking  [################################----]   90%
Unpacking  [#################################---]   93%
Unpacking  [##################################--]   96%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.0.0

Unpacking  [------------------------------------]    0%
Unpacking  [#-----------------------------------]    2%
Unpacking  [##----------------------------------]    5%
Unpacking  [###---------------------------------]    8%
Unpacking  [####--------------------------------]   11%
Unpacking  [#####-------------------------------]   14%
Unpacking  [######------------------------------]   17%
Unpacking  [#######-----------------------------]   20%
Unpacking  [########----------------------------]   22%
Unpacking  [#########---------------------------]   25%
Unpacking  [##########--------------------------]   28%
Unpacking  [###########-------------------------]   31%
Unpacking  [############------------------------]   34%
[... 284 lines cut ...]
      ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:472:39: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
     cool_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 1);
                                       ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:472:62: error: expected primary-expression before '>' token
     cool_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 1);
                                                              ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:473:39: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
     heat_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 2);
                                       ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:473:62: error: expected primary-expression before '>' token
     heat_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 2);
                                                              ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:474:39: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
     auto_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 3);
                                       ^
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:474:62: error: expected primary-expression before '>' token
     auto_storage = global_preferences.make_preference<uint8_t>(this->get_object_id_hash() + 3);
                                                              ^
Compiling .pioenvs/heatpump-camera/src/esphome/components/mqtt/mqtt_component.cpp.o
*** [.pioenvs/heatpump-camera/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 13.80 seconds =========================`

Thx for your time

davidmerrique, 2021-10-24:
@geoffdavis Looks like it's fixed in the develop branch, but not in the master branch.

geoffdavis, 2021-10-24:
That's what I get for trying to do this from an iPad in an airport lounge. Let me cut a new release.

geoffdavis, 2021-10-24:
OK should be there in 2.3.2

closed 2021-10-24 by geoffdavis

davidmerrique, 2021-10-24:
@geoffdavis Thanks so much