geoffdavis/esphome-mitsubishiheatpump #49: Losing Mitsu connection after firmware update

earwin, 2021-12-02:
I'm running ESPHome on a random D1 Mini Pro.
Every time I update it, rebuild the firmware, and upload to the controller, it loses the ability to communicate with Mitsu.
A manual reset at this point always fixes things.

I bodged up a switch to do it, like
```yaml
switch:
  - platform: restart
    name: Mitsubishi AC Controller Reset
```
so it's no longer a problem. And I can maybe later add some hook to firmware update, so I don't have to remember.

But raising this here in case anyone has ideas or wants to share their experience.

geoffdavis, 2022-09-21:
@earwin You likely either have a flakey piece of hardware, or you need to implement a voltage divider or level shifter for the logic. My three bottom-of-the-barrel ESP8266 boards have been rock solid for several years now with just a simple voltage divider circuit on the Rx line.

closed 2022-09-21 by geoffdavis

earwin, 2022-09-21:
@geoffdavis I have a divider, and hardware is rock solid, except for this one specific issue.
E.g. never ever loses a connection outside of firmware upload, always does when uploading.

geoffdavis, 2022-09-21:
I would try another D1 Mini Pro if you haven't already. Also take a look at #56 as there are several folks with flakey comms issues that have since resolved them in that thread.

earwin, 2022-09-21:
Aha, thanks. The thing that bugs me is that there is no "flakiness" whatsoever.
Memory problems from your linked issue (related to web_server) could be a good candidate for what I'm seeing — as they might have this repeatability, but no, doesn't seem so.

Anyway, as I said, just put the issue here in case someone else hits this and does a search.