geoffdavis/esphome-mitsubishiheatpump #50: Conflict with Homeassistant time and this component in espome 2021.12 prevents succesful build

nmeachen, 2021-12-18:
Since the release of esphome 2021.12 the homeassistant time platform has to be removed in order to get heatpump firmware to build. It would appear something needs to be namespace qualified.

<pre>Compiling .pioenvs/hptest/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/ota/ota_component.cpp.o
In file included from /piolibs/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:27:0,
                 from src/esphome/components/web_server_base/web_server_base.h:9,
                 from src/esphome/components/web_server/web_server.h:7,
                 from src/esphome.h:52,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/config/.esphome/platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h: In static member function 'static time_t fs::FS::_defaultTimeCB()':
/config/.esphome/platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:243:49: error: reference to 'time' is ambiguous
     static time_t _defaultTimeCB(void) { return time(NULL); }
                                                 ^
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/wchar.h:4:0,
                 from /config/.esphome/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/cwchar:44,
                 from /config/.esphome/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/postypes.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/char_traits.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/string:40,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:2,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/config/.esphome/platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/../include/time.h:53:18: note: candidates are: time_t time(time_t*)
 time_t    _EXFUN(time,     (time_t *_timer));
                  ^
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4:0,
                 from src/esphome.h:24,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/time/real_time_clock.h:11:16: note:                 namespace esphome::time { }
 namespace time {
                ^
Compiling .pioenvs/hptest/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/hptest/src/esphome/components/sensor/sensor.cpp.o
*** [.pioenvs/hptest/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 23.74 seconds =========================</pre>

nmeachen, 2021-12-19:
Turns out this is an esphome issue related to the sequence of the inclusion of options in the YAML file. I re-sequenced and the problem went away.

closed 2021-12-19 by nmeachen

apnar, 2022-10-11:
@nmeachen I'm running into this same issue.  I've tried a number of yaml orderings without success.  Builds fine if I remove homeassistant time block from yaml.  Which ordering did you use to get it to work?

nmeachen, 2022-10-11:
@apnar The sequence of my working YAML is

esphome:
logger:
api:
ota:
wifi:
captive_portal:
web_server:
time:
switch:
sensor:
binary_sensor:
external_components:
climate: