geoffdavis/esphome-mitsubishiheatpump #51: Energy/Power measurement?

MarkoPaasila, 2022-01-07:
Melcloud reports the energy usage of my heat pump in 0.1 kWh increments. I expect the actual granularity to be much better locally. Could this be read by an ESP? Do you know if anyone has looked into this? Probably only the newer units have this functionality.

jamescadd, 2022-01-30:
I wonder if this is just a relatively inaccurate calculation from Mitsubishi based on power/mode settings?  Either way from a quick search of the underlying library I didn't see anything that mentions power or energy:

https://github.com/SwiCago/HeatPump/blob/master/src/HeatPump.cpp

geoffdavis, 2022-02-12:
Yeah I don't know of any true way of getting the real power draw out of the Mitsubishi equipment, other than a wild guess based on the compressor's efficiency and the operating duty cycle. The problem there is that the "operating" status doesn't tell you how hard it's operating - just barely running the exchanger, or pumping full blast, or something in between. As @jamescadd pointed out, it's not in the underlying protocol for the units - at least not in the stuff that's been decoded by the community.

I was going to install a CT clamp on the electrical shutoff j-box for my unit at some point to get total system draw (3 indoor units on one ODU.)