geoffdavis/esphome-mitsubishiheatpump #52: External updates from IR remote reverted

jsn-0, 2022-01-11:
According to the [readme](https://github.com/SwiCago/HeatPump/blob/60f0f18f8c73c0f30a53f5075d86cacbf8ec3dd2/README.md#getting-updates-from-the-heat-pump) in SwiCago/HeatPump, settings made externally to the heat pump settings via the IR remote will be automatically reverted each time `sync()` is called unless `enableExternalUpdate()` is called to allow it.

Is that the case with this implementation or am I missing something? Thanks so much for your work on this!

jamescadd, 2022-01-30:
I'd also like to be able to allow the IR remote to be used alongside Home Assistant, nice catch!  It looks easy enough to add the call during setup and see if it works, though I did wonder if there was something larger going on with the flow of settings from hp to ESPHome based on this comment that could cause an issue:

https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/ac207f52d925f568dc6d7ce6357b0be2ae50d525/components/mitsubishi_heatpump/espmhp.cpp#L257

geoffdavis, 2022-02-12:
As currently implemented, if I change things with my IR remote, ESPHome sees the changes and updates Home Assistant. No overriding of the IR remote calls occurs unless explicitly requested.

closed 2022-02-12 by geoffdavis