geoffdavis/esphome-mitsubishiheatpump #56: Continuous Reboot on ESP32

jdgarcia99, 2022-02-16:
Hi

I'm trying to use this component with an ESP32 like this https://www.az-delivery.de/es/collections/mikrocontroller/products/esp32-d1-mini

Compilation and upload through ESPHome-Flasher is ok, but once uploaded the module is constantly rebooting

Here is an example of logs:

howing logs:
[21:10:43]E (10294) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:10:43]E (10294) task_wdt:  - loopTask (CPU 1)
[21:10:43]E (10294) task_wdt: Tasks currently running:
[21:10:43]E (10294) task_wdt: CPU 0: IDLE0
[21:10:43]E (10294) task_wdt: CPU 1: IDLE1
[21:10:43]E (10294) task_wdt: Aborting.
[21:10:43]abort() was called at PC 0x4015fbd8 on core 0
[21:10:43]
[21:10:43]ELF file SHA256: 0000000000000000
[21:10:43]
[21:10:43]Backtrace: 0x40088ac8:0x3ffbf820 0x40088d45:0x3ffbf840 0x4015fbd8:0x3ffbf860 0x40087205:0x3ffbf880 0x40177b77:0x3ffbc160 0x401614e3:0x3ffbc180 0x4008b505:0x3ffbc1a0 0x40089d56:0x3ffbc1c0
[21:10:43]
[21:10:43]Rebooting...
[21:10:43]ets Jun  8 2016 00:22:57
[21:10:43]
[21:10:43]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:10:43]configsip: 0, SPIWP:0xee
[21:10:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:10:43]mode:DIO, clock div:2
[21:10:43]load:0x3fff0018,len:4
[21:10:43]load:0x3fff001c,len:1044
[21:10:43]load:0x40078000,len:8896
[21:10:43]load:0x40080400,len:5828
[21:10:43]entry 0x400806ac
[21:10:44][I][logger:214]: Log initialized
[21:10:44][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[21:10:44][D][esp32.preferences:114]: Saving preferences to flash...
[21:10:44][I][app:029]: Running through setup()...
[21:10:44][D][text_sensor:067]: 'heatpump-test ESPHome Version': Sending state '2022.2.0 Feb 16 2022, 21:05:11'
[21:10:44][I][MitsubishiHeatPump:063]: ESPHome MitsubishiHeatPump version 2.3.2
[21:10:44][C][MitsubishiHeatPump:416]: Setting up UART...
[21:10:44][C][MitsubishiHeatPump:428]: Intializing new HeatPump object.
[21:10:44][C][MitsubishiHeatPump:455]: hw_serial(0x3ffc1adc) is &Serial(0x3ffc1af8)? NO
[21:10:44][C][MitsubishiHeatPump:457]: Calling hp->connect(0x3ffc1adc)
[21:10:49]E (10294) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:10:49]E (10294) task_wdt:  - loopTask (CPU 1)
[21:10:49]E (10294) task_wdt: Tasks currently running:
[21:10:49]E (10294) task_wdt: CPU 0: IDLE0
[21:10:49]E (10294) task_wdt: CPU 1: IDLE1
[21:10:49]E (10294) task_wdt: Aborting.
[21:10:49]abort() was called at PC 0x4015fbd8 on core 0
[21:10:49]
[21:10:49]ELF file SHA256: 0000000000000000
[21:10:49]
[21:10:49]Backtrace: 0x40088ac8:0x3ffbf820 0x40088d45:0x3ffbf840 0x4015fbd8:0x3ffbf860 0x40087205:0x3ffbf880 0x40177b77:0x3ffbc160 0x401614e3:0x3ffbc180 0x4008b505:0x3ffbc1a0 0x40089d56:0x3ffbc1c0
[21:10:49]
[21:10:49]Rebooting...
[21:10:49]ets Jun  8 2016 00:22:57
[21:10:49]
[21:10:49]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:10:49]configsip: 0, SPIWP:0xee
[21:10:49]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:10:49]mode:DIO, clock div:2
[21:10:49]load:0x3fff0018,len:4
[21:10:49]load:0x3fff001c,len:1044
[21:10:49]load:0x40078000,len:8896
[21:10:49]load:0x40080400,len:5828
[21:10:49]entry 0x400806ac
[21:10:49][I][logger:214]: Log initialized
[21:10:49][C][ota:461]: There have been 2 suspected unsuccessful boot attempts.
[21:10:49][D][esp32.preferences:114]: Saving preferences to flash...
[21:10:49][I][app:029]: Running through setup()...
[21:10:49][D][text_sensor:067]: 'heatpump-test ESPHome Version': Sending state '2022.2.0 Feb 16 2022, 21:05:11'
[21:10:49][I][MitsubishiHeatPump:063]: ESPHome MitsubishiHeatPump version 2.3.2
[21:10:49][C][MitsubishiHeatPump:416]: Setting up UART...
[21:10:49][C][MitsubishiHeatPump:428]: Intializing new HeatPump object.
[21:10:49][C][MitsubishiHeatPump:455]: hw_serial(0x3ffc1adc) is &Serial(0x3ffc1af8)? NO
[21:10:49][C][MitsubishiHeatPump:457]: Calling hp->connect(0x3ffc1adc)
[21:10:54]E (10294) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:10:54]E (10294) task_wdt:  - loopTask (CPU 1)
[21:10:54]E (10294) task_wdt: Tasks currently running:
[21:10:54]E (10294) task_wdt: CPU 0: IDLE0
[21:10:54]E (10294) task_wdt: CPU 1: IDLE1
[21:10:54]E (10294) task_wdt: Aborting.
[21:10:54]abort() was called at PC 0x4015fbd8 on core 0
[21:10:54]
[21:10:54]ELF file SHA256: 0000000000000000
[21:10:54]
[21:10:54]Backtrace: 0x40088ac8:0x3ffbf820 0x40088d45:0x3ffbf840 0x4015fbd8:0x3ffbf860 0x40087205:0x3ffbf880 0x40177b77:0x3ffbc160 0x401614e3:0x3ffbc180 0x4008b505:0x3ffbc1a0 0x40089d56:0x3ffbc1c0
[21:10:54]
[21:10:54]Rebooting...
[21:10:55]ets Jun  8 2016 00:22:57
[21:10:55]
[21:10:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:10:55]configsip: 0, SPIWP:0xee
[21:10:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:10:55]mode:DIO, clock div:2
[21:10:55]load:0x3fff0018,len:4
[21:10:55]load:0x3fff001c,len:1044
[21:10:55]load:0x40078000,len:8896
[21:10:55]load:0x40080400,len:5828
[21:10:55]entry 0x400806ac
[21:10:55][I][logger:214]: Log initialized
[21:10:55][C][ota:461]: There have been 3 suspected unsuccessful boot attempts.

This is my ESPHome configuration file for this device:

substitutions:
  name: heatpump-test
  friendly_name: Test Heatpump

esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback Hotspot"
    password: ap_password

captive_portal:

# Enable Web server.
web_server:
  port: 80
  
# MQTT
mqtt:
  broker: 192.168.10.211
  port: 1883
  topic_prefix: ${name}
  
# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time
    
# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID
      
# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s
    
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART1

Any Idea?

Thanks in advance!

undies420, 2022-02-20:
I am having the same issue.

sitexw, 2022-02-21:
I have exactly the same problem...

mattatcha, 2022-03-08:
FYI, I ran into the same issue when rx,tx weren't connected to the configured uart port.

sitexw, 2022-03-08:
I must admit that I had doubts about my RX and TX connections when I did the tests. My connection wasn't exactly the right one, so I had to scrape it off with a cutter. Visually, it fits well, but I'm not 100% sure that the connections are made...

I will try to provide me with the correct side pon connector and get back to you.

MarkoPaasila, 2022-03-12:
I can confirm that connecting to the heatpump solves this problem for me.

Stendec1, 2022-04-11:
Just got it up-and-running on the AZ-Delivery D1 Mini ESP32 using UART0.
You have to swap the RX & TX connection!!

Following is my ESPHome configuration file for this device:

```
esphome:
  name: ${name}

esp32:
  board: lolin_d32
  variant: ESP32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${name}
    password: !secret wifi_password

# Note: if upgrading from 1.x releases of esphome-mitsubishiheatpump, be sure
# to remove any old entries from the `libraries` and `includes` section.
#libraries:
  # Remove reference to SwiCago/HeatPump

#includes:
  # Remove reference to src/esphome-mitsubishiheatpump

captive_portal:

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

# Enable Home Assistant API
api:
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(hp).set_remote_temperature(temperature);'

    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'

ota:
  password: !secret 'esphome_api_password'
  
# Enable Web server.
web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

  - platform: homeassistant
    name: Temperature Sensor From Home Assistant
    entity_id: sensor.ble_temperature_a4c13818705a
    on_value:
    - then:
      - lambda: !lambda |-
          id(hp).set_remote_temperature(x);
    disabled_by_default: false
    force_update: false
    accuracy_decimals: 1
    state_class: ''
      
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    id: hp
    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART0
    baud_rate: 2400
    disabled_by_default: false
    visual: {}
    update_interval: 500ms

substitutions:
  name: mitsubishi-hvac-livingroom
  friendly_name: livingroom_HVAC  
```

Stendec1, 2022-04-15:
Did some further digging and found several issues that I have tackled, resulting in a flawless operation for 2 days on 2 units.

1) Swap RX & TX lines.
2)  revert back to 'nodemcu-32s' board and use level converters for RX & TX lines.
3) Disable web_server ( very important as it eats away necessary memory!! )

Current YAML:

```
esphome:
  name: ${name}

esp32:
  board: nodemcu-32s
  variant: ESP32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${name}
    password: !secret wifi_password

# Note: if upgrading from 1.x releases of esphome-mitsubishiheatpump, be sure
# to remove any old entries from the `libraries` and `includes` section.
#libraries:
  # Remove reference to SwiCago/HeatPump

#includes:
  # Remove reference to src/esphome-mitsubishiheatpump

captive_portal:

# Enable logging
logger:
  level: NONE
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0
  #hardware_uart: UART1

# Enable Home Assistant API
api:

ota:
  password: !secret 'esphome_api_password'
  
# Enable Web server.
#web_server:
#  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

  # You could use a Bluetooth temperature sensor
  - platform: xiaomi_lywsdcgq
    mac_address: "58:2D:34:3A:5B:F8"
    humidity:
      name: "LYWSDCGQ Humidity"
    battery_level:
      name: "LYWSDCGQ Battery Level"  
    temperature:
      name: "LYWSDCGQ Temperature"
      on_value:
        then:
          - lambda: !lambda |-
              id(hp).set_remote_temperature(x);
  
esp32_ble_tracker:
  
      
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    id: hp
    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART0
    baud_rate: 2400
    disabled_by_default: false
    visual: {}
    update_interval: 500ms

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 2400
  rx_buffer_size: 512

button:
- platform: restart
  name: Restart
  entity_category: config
  disabled_by_default: false
  device_class: restart
- platform: shutdown
  name: Shutdown
  entity_category: config
  disabled_by_default: false
  icon: mdi:power
  
switch:
  - platform: safe_mode
    name: "Living Room Restart (Safe Mode)"
    

substitutions:
  name: mitsubishi-hvac-livingroom
  friendly_name: livingroom_HVAC      

```

Andreastap, 2022-07-18:
Hi Stendec1, I was inspired by your configuration but once sent these errors occur, how come you can help me ???

INFO Successfully compiled program.
INFO Resolving IP address of mitsubishi-hvac-livingroom.local
ERROR Error resolving IP address of mitsubishi-hvac-livingroom.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

Stendec1, 2022-07-18:
Hi @Andreastap,  using DHCP with .local addresses gives problems sometimes, with me as well.
I therefor have reverted to specify a fixed address in the yaml wifi section:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${name}
    password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.XXX.XXX  
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.XXX.XXX
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
    dns1: 192.168.XXX.XXX
    dns2: 192.168.XXX.XXX

```

Give it a try.

MarkSmithAU, 2022-07-26:
Has anyone ever got this to work with UART2 (ideally) or UART1 (soldering required on a nodemcu32)?
It seems like it might be electrical (level shifter required) or there is a bug in the code that's not actually selecting the correct UART specified in the configuration?

geoffdavis, 2022-09-21:
@MarkSmithAU The readme notes that you have to use a level shifter or a voltage divider. The mitsubishi board uses 5V TTL logic, while the ESP8266 and ESP32 use 3.3V TTL logic.

MarkSmithAU, 2022-09-22:
> The readme notes that you have to use a level shifter or a voltage divider. The mitsubishi board uses 5V TTL logic, while the ESP8266 and ESP32 use 3.3V TTL logic.

Thanks for Replying @geoffdavis .  I did further research and found that UART2 is likely used by the SPI to access the flash and UART1 is broken - multiple threads on multiple forums.  I ended up getting more appropriately sized esp8266 modules and they have worked flawlessly.  It was not a level shift/voltage divider issue; I am using pullup resistors and removed an other inputs from the usb-serial chip on the board so they don't clash with the aircon's output which seems to be an open collector configuration.

geoffdavis, 2022-10-07:
@MarkSmithAU wow, interesting.

I'm using the little cheesy ESP01S boards with a perf-board breakout, rather than any of the D1 mini or nodemcu style boards.

closed 2022-10-07 by geoffdavis