geoffdavis/esphome-mitsubishiheatpump #57: Error due to Removed bool HeatPump::connect(HardwareSerial *serial, int bitrate) function

F43RY, 2022-02-18:
Hi,
yesteraday was made a commit (https://github.com/SwiCago/HeatPump/commit/b5ebb6113f8bb75574eccad021cadbde7ceb24d3) that removed the following function:

`bool HeatPump::connect(HardwareSerial *serial, int bitrate)`

Now, trying to install we got the following errors:
`src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:459:56: error: no matching function for call to 'HeatPump::connect(HardwareSerial*, int&)'
  459 |     if (hp->connect(this->get_hw_serial_(), this->baud_)) {
      |                                                        ^
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:23,
                 from src/esphome.h:18,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:237:10: note: candidate: 'bool HeatPump::connect(HardwareSerial*)'
  237 |     bool connect(HardwareSerial *serial);
      |          ^~~~~~~
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:237:10: note:   candidate expects 1 argument, 2 provided
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:238:10: note: candidate: 'bool HeatPump::connect(HardwareSerial*, int, int)'
  238 |     bool connect(HardwareSerial *serial, int rx, int tx);
      |          ^~~~~~~
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:238:10: note:   candidate expects 3 arguments, 2 provided
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:239:10: note: candidate: 'bool HeatPump::connect(HardwareSerial*, int, int, int)'
  239 |     bool connect(HardwareSerial *serial, int bitrate, int rx, int tx);
      |          ^~~~~~~
.piolibdeps/heatpump-camera/HeatPump/src/HeatPump.h:239:10: note:   candidate expects 4 arguments, 2 provided
Compiling .pioenvs/heatpump-camera/src/esphome/components/uptime/uptime_sensor.cpp.o
*** [.pioenvs/heatpump-camera/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1`

Thx

toaday, 2022-02-19:
From goosfand, who committed the change in SwiCago\HeatPump:
@goosfand:matrix.org
[[m]](https://matrix.to/#/@goosfand:matrix.org?utm_source=gitter)
[13:29](https://gitter.im/Mitsubishi-Heat-Pump/Lobby?at=62116145d1b64840db2d18e1)
Looks like my commit broke that. The function isn't removed, but two new parameters are added to it. In espmhp.cpp, find if (hp->connect(this->get_hw_serial_(), this->baud_)) { and instead try if (hp->connect(this->get_hw_serial_(), this->baud_, -1, -1)) {, hopefully that addresses the issue.

basettoni89, 2022-02-20:
I'm having the same issue mentioned by @F43RY during installation on ESPHome.
There is any workaround to make the code compile again?

Thx

SwiCago, 2022-02-20:
This was brought to my attention in my project chat. I am not responsible for third party usage and their upkeep when my library changes, but as a courtesy I have added back backwards compatible support. It may go away in the future, especially as more refactoring is done.

geoffdavis, 2022-02-22:
Thanks. I'll update this library accordingly to reflect the changed function signature.

javilopezalarcon, 2022-03-16:
good afternoon, there is no solution yet for this right? I'm not in a hurry, it's just in case there was and it would have happened to me, I'm Spanish and I use a translator to understand it, I'm sorry

geoffdavis, 2022-04-05:
@SwiCago any chance you could start cutting versioned releases of your HeatPump library? I'm trying to replicate this issue on my install, but since the dependency I have in the project's "climate.py" is just limited to the project URL, I can't force ESPHome to update.

I'm going to pin the dependency to the current git hash for now, but that's got bit-rot built in.

geoffdavis, 2022-04-05:
Fixed in GH-60

closed 2022-04-05 by geoffdavis