geoffdavis/esphome-mitsubishiheatpump #59: ESP crash at fan speed change for MSZ-LN35VGW

MarkoPaasila, 2022-03-12:
Everything works until I change fan_mode through esphome. Then it abruptly disconnects from wifi. Here's the end of the log with very_verbose loglevel:
```
[15:20:51][VV][api.service:519]: on_climate_command_request: ClimateCommandRequest {
  key: 680327010
  has_mode: NO
  mode: CLIMATE_MODE_OFF
  has_target_temperature: NO
  target_temperature: 0
  has_target_temperature_low: NO
  target_temperature_low: 0
  has_target_temperature_high: NO
  target_temperature_high: 0
  has_legacy_away: NO
  legacy_away: NO
  has_fan_mode: YES
  fan_mode: CLIMATE_FAN_MEDIUM
  has_swing_mode: NO
  swing_mode: CLIMATE_SWING_OFF
  has_custom_fan_mode: NO
  custom_fan_mode: ''
  h
[15:20:51][D][climate:010]: 'esphome-mitsubishi' - Setting
[15:20:51][D][climate:023]:   Fan: MEDIUM
INFO 10.0.0.221: Ping timed out!
INFO Disconnected from ESPHome API for 10.0.0.221
WARNING Disconnected from API
```
Checking from Melcloud, the fan_mode doesn't change. The same happens with all speeds.
I am using a nodemcu-v2 / devkitC board without pullup resistors. Tried both with and without a level shifter in between.
I observed the available fan_modes (in esphome log) by stepping through them all (controlling through Melcloud). 
All values can be read, and are updated when controlling with IR or Melcloud. Setting temperature and climate mode works with esphome.
Here's my config:
```
substitutions:
  friendly_name: esphome-mitsubishi
  platform: ESP32
  board: nodemcu-32s
  powersavemode: none
  output_power: 10dB
  static_ip: 192.168.1.221
  logger_baud_rate: '115200'
  logger_uart: UART1
  loglevel: VERY_VERBOSE
esphome:
  name: esphome-mitsubishi
  build_path: .esphome/build/esphome-mitsubishi
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
esp32:
  board: nodemcu-32s
  framework:
    version: 1.0.6
    source: ~3.10006.0
    platform_version: platformio/espressif32 @ 3.5.0
    type: arduino
  variant: ESP32
external_components:
- source:
    url: https://github.com/geoffdavis/esphome-mitsubishiheatpump.git
    type: git
  refresh: 1d
  components: all
climate:
- platform: mitsubishi_heatpump
  name: esphome-mitsubishi
  id: hp
  hardware_uart: UART0
  baud_rate: 9600
  supports:
    fan_mode:
    - AUTO
    - HIGH
    - MIDDLE
    - MEDIUM
    - LOW
    - DIFFUSE
    mode:
    - HEAT_COOL
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    swing_mode:
    - 'OFF'
    - VERTICAL
  disabled_by_default: false
  visual: {}
  update_interval: 500ms
sensor:
- platform: homeassistant
  name: Temperature Sensor From Home Assistant
  entity_id: sensor.temperature_to_heatpump
  on_value:
  - then:
    - lambda: !lambda |-
        id(hp).set_remote_temperature(x);
  disabled_by_default: false
  force_update: false
  accuracy_decimals: 1
  state_class: ''
- platform: wifi_signal
  name: esphome-mitsubishi RSSI
  device_class: signal_strength
  state_class: measurement
  update_interval: 300s
  disabled_by_default: true
  entity_category: diagnostic
  force_update: false
  unit_of_measurement: dBm
  accuracy_decimals: 0
wifi:
  domain: .lan
  reboot_timeout: 15min
  output_power: 10.0
  power_save_mode: NONE
  manual_ip:
    static_ip: 192.168.1.221
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 0.0.0.0
  ap:
    ssid: esphome-mitsubishi AP
    password: !secret 'webserver_password'
    ap_timeout: 1min
  fast_connect: false
  networks:
  - ssid: !secret 'webserver_username'
    password: !secret 'webserver_password'
    priority: 0.0
  use_address: 192.168.1.221
captive_portal: {}
logger:
  baud_rate: 115200
  hardware_uart: UART1
  level: VERY_VERBOSE
  tx_buffer_size: 512
  deassert_rts_dtr: false
  logs: {}
api:
  password: !secret 'esphome_api_password'
  port: 6053
  reboot_timeout: 15min
ota:
  password: !secret 'esphome_api_password'
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
web_server:
  port: 80
  auth:
    username: !secret 'webserver_username'
    password: !secret 'webserver_password'
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
  include_internal: false
  ota: true
time:
- platform: homeassistant
  id: homeassistant_time
  timezone: EET-2EEST,M3.5.0/3,M10.5.0/4
  update_interval: 15min
mdns:
  disabled: false
button:
- platform: restart
  name: Restart
  entity_category: config
  disabled_by_default: false
  device_class: restart
- platform: shutdown
  name: Shutdown
  entity_category: config
  disabled_by_default: false
  icon: mdi:power
```

MarkoPaasila, 2022-04-06:
Now it looks like I have no communication between the unit and the esp32. I'll close this issue and open a separate one to discuss how to troubleshoot the unit.

closed 2022-04-06 by MarkoPaasila

markalston, 2023-11-03:
Did you solve this issue? Having the exact same crash when changing fan modes.

MarkoPaasila, 2023-11-03:
At least I set the baud rate at 2400. I put a capacitor vetwren 3v and gnd. Also tried another esp. I can’t remember what fixed it.