geoffdavis/esphome-mitsubishiheatpump #61: Not grabbing updates/packets from heatpump unless "update_interval: 500ms" explicitly defined.

elmoret, 2022-04-13:
Model SVZ-KP24NA, ESP-01S (also tried Wemos D1), Comms to the heatpump (setting desired temperature) worked fine, once baud rate changed to 2400bps (8E1). Current temperature was always reporting "nan". Used logic analyzer, packets were only being sent to the heatpump when a setting changed in the web interface - no polling was occurring.

Explicitly defined the "optional" update_interval parameter, problem solved.

Thanks for the code, geoffdavis. Happy to buy you a beverage of your choice is you post a donate link?

biotinker, 2022-04-29:
Same model SVZ-KP24NA. I'm interestingly having the exact same issue- I can set modes and temperatures, but I can't seem to get info back from the heat pump and current temperature is perpetually NAN.

I forked my own version of the project and added some logging statements at the top of `MitsubishiHeatPump::hpStatusChanged()` and disabled callbacks so it should be called by `void MitsubishiHeatPump::update() {
    // This will be called every "update_interval" milliseconds.` on the interval, but it is not- logging statements are not showing up.

I'll continue to troubleshoot and see if I can dig down to a root cause. Will update in this issue as I find things.

biotinker, 2022-04-29:
I booted the ESP32 while also logging the boot sequence over UART, and found this:
```
[C][MitsubishiHeatPump:421]: Setting up UART...
[C][MitsubishiHeatPump:433]: Intializing new HeatPump object.
[C][MitsubishiHeatPump:460]: hw_serial(0x3ffc1a10) is &Serial(0x3ffc1a48)? NO
[C][MitsubishiHeatPump:462]: Calling hp->connect(0x3ffc1a10)
[C][MitsubishiHeatPump:472]: Connection to HeatPump failed. Marking MitsubishiHeatPump component as failed.
[E][component:112]: Component esphome.coroutine was marked as failed.
```
The heat pump controls however came up fine, and I'm still able to turn heat/AC/fans on and off and control speeds etc. But my state in the logs is a perpetual `Current Temperature: nan°C` and `Action: IDLE`

biotinker, 2022-04-29:
Edited to call a `hp->sync()` regardless of whether `hp->connect()` worked. It crashed, but I managed to get a backtrace out, not that I can read it:
```
[C][MitsubishiHeatPump:462]: Calling hp->connect(0x3ffc1a10)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f08b6  PS      : 0x00060b30  A0      : 0x800f051e  A1      : 0x3ffb1b90
A2      : 0x00000000  A3      : 0x00000960  A4      : 0x0800001e  A5      : 0xffffffff
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x00000960  A9      : 0x3ffb1b70
A10     : 0x0003ec94  A11     : 0x00000000  A12     : 0x0003ec94  A13     : 0x3f403063
A14     : 0x00000008  A15     : 0x00000010  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400f08b6:0x3ffb1b90 0x400f051b:0x3ffb1bd0 0x400f05ad:0x3ffb1c10 0x400f05c2:0x3ffb1c30 0x400f066a:0x3ffb1c50 0x400db4e2:0x3ffb1c90 0x400e49ce:0x3ffb1cc0 0x4016f159:0x3ffb1d10 0x400e4721:0x3ffb1d30 0x400e6d42:0x3ffb1d80 0x400f2996:0x3ffb1fb0 0x40089aba:0x3ffb1fd0
```

biotinker, 2022-04-29:
I checked the return value of `hp->update()` at the end of `control()`, and found it's returning false but my settings are still managing to get through somehow.

My suspicion is that the issue is not at all in ESPHome, and is instead somewhere in SwiCago's HeatPump repo. I'll start poking around there next.

biotinker, 2022-04-29:
Update for anyone else with a similar problem to me:
Turns out I was a dummy and didn't have the pull-up resistors properly connected. Once I plugged those in, everything started working as normal.

Except for "Current Temperature", that's still NAN. But prior to this install I had actually used my old thermostat to change the heat pump setting such that it only considered thermostat remote temperatures rather than the internal temp of air from the intake, as my air intake is located in a different area of the house than the heat pump warms/cools. So that might explain that.

elmoret, 2022-04-29:
Makes sense - the ESP-01S is setting on a level shifter board in my case. The NAN issue went away when I added the update interval parameter mentioned in this issue subject.

My understanding is that if you're not providing remote temperatures, the heatpump's controller falls back to the internal sensor automatically. But wouldn't hurt to change that setting and retry! What I ended up doing is relocating the internal heat pump thermistor inside the house, as I didn't want to rely on HA for the heat pump to be able to grab indoor temperatures.

biotinker, 2022-04-30:
Oh, I almost forgot to add- once I got everything working, I tested removing the `update_interval` parameter and sure enough it did not update if that was not set, as you mentioned. Odd.

You're correct that the internal thermistor will be used as a fallback- for me, I see it being used until it's able to pull the remote temp from HA.

One thing to note- this library assumes the temperature coming from HA is in Celsius. If your sensor reports farenheit, you need to convert to celsius prior to feeding into this app.

geoffdavis, 2022-05-02:
It does sound like the lack of a level shifter was the fault here. It sounds like the Heat pump could hear your esp32 but not the other way around.

I've been running the resistor networks on my units for several years now and it's worked fine.

geoffdavis, 2022-05-02:
In terms of temperature, the underlying hardware works entirely in centigrade. Even the remote protocol sends centigrade when it's displaying Fahrenheit.

I haven't seen a way for ESPHOME to determine the units that Home Assistant is using, so until that happens, just assume that this component expects centigrade for input.

geoffdavis, 2022-09-21:
Looks like this has been answered.

closed 2022-09-21 by geoffdavis