geoffdavis/esphome-mitsubishiheatpump #62: Unable to build after update to ESPhome 2022.3.2

JanJoh, 2022-04-18:
Previously I was on 2022.2.0 where it worked as expected

Updating /config/heatpump-etage.yaml
------------------------------------------------------------

INFO Reading configuration /config/heatpump-etage.yaml...
INFO Updating https://github.com/geoffdavis/esphome-mitsubishiheatpump.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing heatpump-etage (board: huzzah; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <ArduinoJson> 6.18.5
|-- <HeatPump> 1.0.0+sha.4c3e51f
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/homeassistant/sensor/homeassistant_sensor.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/text_sensor/filter.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/text_sensor/text_sensor.cpp.o
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:459:64: error: no matching function for call to 'HeatPump::connect(HardwareSerial*, int&, int, int)'
  459 |     if (hp->connect(this->get_hw_serial_(), this->baud_, -1, -1)) {
      |                                                                ^
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:23,
                 from src/esphome.h:30,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
.piolibdeps/heatpump-etage/HeatPump/src/HeatPump.h:191:10: note: candidate: 'bool HeatPump::connect(HardwareSerial*)'
  191 |     bool connect(HardwareSerial *serial);
      |          ^~~~~~~
.piolibdeps/heatpump-etage/HeatPump/src/HeatPump.h:191:10: note:   candidate expects 1 argument, 4 provided
.piolibdeps/heatpump-etage/HeatPump/src/HeatPump.h:192:10: note: candidate: 'bool HeatPump::connect(HardwareSerial*, int)'
  192 |     bool connect(HardwareSerial *serial, int bitrate);
      |          ^~~~~~~
.piolibdeps/heatpump-etage/HeatPump/src/HeatPump.h:192:10: note:   candidate expects 2 arguments, 4 provided
Compiling .pioenvs/heatpump-etage/src/esphome/components/time/automation.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/time/real_time_clock.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/uptime/uptime_sensor.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/version/version_text_sensor.cpp.o
Compiling .pioenvs/heatpump-etage/src/esphome/components/web_server/list_entities.cpp.o
*** [.pioenvs/heatpump-etage/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 12.86 seconds =========================
=========== [ERROR] /config/heatpump-etage.yaml ===========

takthetank, 2022-04-27:
Same error occurs in version 2022.4.0.

JanJoh, 2022-04-27:
Works in 2022.4.6 after clean of files

closed 2022-04-27 by JanJoh

takthetank, 2022-04-27:
> Works in 2022.4.6 after clean of files

How do we clean files?

JanJoh, 2022-04-27:
> 

You really haven't looked very closely of the options you are presented when hitting the "..." icon in ESPHome, have you? ;)

takthetank, 2022-04-27:
Wasn't sure what it does exactly, thanks!