geoffdavis/esphome-mitsubishiheatpump #65: With an ESP32, how do I specify UART and pins to use?

lmamakos, 2022-07-10:
I see when configuring the component, I can specify UART0, UART1 or UART2 in the YAML.  So I can do something like:

```
climate:
    #  TX GPIO17
    #  RX GPIO16
 - platform: mitsubishi_heatpump
   name: attic
   hardware_uart: UART2
   update_interval: 500ms
```
And I'm hoping it will use GPIO pins 16 and 17 since that seems to be the default mapping, at least based on the diagram of the module pin-outs and other trivia I've seen on the internets.

But is there a way to explicitly specify the pins?  It seems like you can do this on the ESPHome `uart` component... but I didn't see any means there to  indicate which UART the configuration was intended for?  I'm trying to ensure that I get at least this part correct before I open up the mini-split system and start connecting things.  Debugging is going to be difficult since serial I/O not working just causes watchdog timer reboots.

And its seems these days that the ESPHome serial stuff knows how to do even parity, so wouldn't it be great if the ESPHome UART component could be used, rather than calling the Arduino drivers inside the library.  If the serial port isn't connected, I get endless watchdog errors and reboots 
```
[11:42:18][I][MitsubishiHeatPump:063]: ESPHome MitsubishiHeatPump version 2.4.0
[11:42:18][C][MitsubishiHeatPump:416]: Setting up UART...
[11:42:18][C][MitsubishiHeatPump:428]: Intializing new HeatPump object.
[11:42:18][C][MitsubishiHeatPump:455]: hw_serial(0x3ffc1c40) is &Serial(0x3ffc1c78)? NO
[11:42:18][C][MitsubishiHeatPump:457]: Calling hp->connect(0x3ffc1c40)
[11:42:23]E (10418) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:42:23]E (10418) task_wdt:  - loopTask (CPU 1)
[11:42:23]E (10418) task_wdt: Tasks currently running:
[11:42:23]E (10418) task_wdt: CPU 0: IDLE0
[11:42:23]E (10418) task_wdt: CPU 1: IDLE1
[11:42:23]E (10418) task_wdt: Aborting.
```
which is going to make debugging difficult..  I suppose "fixing" this means re-factoring the library that's used to abstract out the serial I/O code somehow..

juan11perez, 2022-09-18:
This post my help

https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/56

geoffdavis, 2022-09-21:
I'm going to close this as a duplicate of #56

closed 2022-09-21 by geoffdavis