geoffdavis/esphome-mitsubishiheatpump #66: ESP32 online but can't connect after mitsubishiheatpump config install

Iddref, 2022-07-18:
Hi,

A few months ago I used the "esphome-mitsubishiheatpump" example configuration shown in the wiki page to set up a controller for my mitsubishi AC. Everything worked perfectly and still does. Since I was so happy with the result I had another Mitsubishi heatpump installed and wanted to reproduce the setup and make a second ESP32 controller. I have copied the exact same configuration (with the climat component) from the working ESP32 to a new ESP32, but after upload the ESP32 can't connect (Error 111 :WARNING Can't connect to ESPHome API for living-room-ac.local). 
The strange thing is that the ESP32 is still online and I can still do OTA updates.  To rule out that it is an issue with my ESP32 I installed a config with the Midea Climat component for testing and this worked fine, i.e. after upload the ESP32 connects and I can see the mideal climat entity in Home Assistant.

Does anyone have ideas of what the issue could be and how to resolve it? I have updated esphome from 2022.4.2 to 2022.6.2 since I installed the first ESP32 which is still working (with the mitsubishi_heatpump component), however since I have encountered this issue, I have reverted back to 2022.4.2 but same issue persist. Again the first ESP32 set up under 2022.4.2 is still working.

Config below:

`esphome:
  name: esphome-web-7c1bc0

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "********"
    password: "*********"

captive_portal:

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART2
    `

MarkSmithAU, 2022-07-26:
I believe this is the same issue as you have: https://community.home-assistant.io/t/mitsubishi-ac-with-wemos-d1-mini-pro/107007/208
If you disconnect the unit from the AC and connect it to a serial terminal (i.e. TeraTerm) do you get a similar output?

geoffdavis, 2022-09-21:
I'd try to do the flash of the controller with the ESP32 hooked up to a serial port for debugging. Otherwise, it's anyone's guess as to why the device is offline.

Note that ESPhome will go into a recovery mode after some number of failed boot attempts, which is likely why you are able to reflash over the air.

geoffdavis, 2022-09-21:
I'd also recommend looking at #56, as this seems like it's most likely a duplicate.

closed 2022-09-21 by geoffdavis