geoffdavis/esphome-mitsubishiheatpump #68: Locked to a specific version of SwiCago/HeatPump

dwildstr, 2022-08-20:
Line 93 of climate.py indicates a specific version of the underlying SwiCago/HeatPump library to use. Is there a good reason for this? The library has had a few communication-improving commits recently in the last two months which, due to this version-locking, aren't reflected in esphome-based heat-pump controllers.

juan11perez, 2022-09-16:
@dwildstr to address this is it a matter of just removing the version?

`    cg.add_library(
        name="HeatPump",
        repository="https://github.com/SwiCago/HeatPump",)
      `

juan11perez, 2022-09-18:
@dwildstr found the answer. to pull the last swicago librabry all needed is this:

`    cg.add_library("SwiCago/HeatPump", "1.0.0")`

However, as pointed out the library has changed and when you try to compile in esphome the component breaks with the following:

`src/esphome/components/mitsubishi_heatpump/espmhp.cpp:440:5: error: no matching function for call to 'HeatPump::setSettingsChangedCallback(MitsubishiHeatPump::setup()::<lambda()>)'
     );
     ^
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:23:0,
                 from src/esphome.h:34,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
.piolibdeps/testac/HeatPump/src/HeatPump.h:232:10: note: candidate: void HeatPump::setSettingsChangedCallback(void (*)())
     void setSettingsChangedCallback(SETTINGS_CHANGED_CALLBACK_SIGNATURE);
          ^
.piolibdeps/testac/HeatPump/src/HeatPump.h:232:10: note:   no known conversion for argument 1 from 'MitsubishiHeatPump::setup()::<lambda()>' to 'void (*)()'
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:446:5: error: no matching function for call to 'HeatPump::setStatusChangedCallback(MitsubishiHeatPump::setup()::<lambda(heatpumpStatus)>)'
     );
     ^
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:23:0,
                 from src/esphome.h:34,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
.piolibdeps/testac/HeatPump/src/HeatPump.h:233:10: note: candidate: void HeatPump::setStatusChangedCallback(void (*)(heatpumpStatus))
     void setStatusChangedCallback(STATUS_CHANGED_CALLBACK_SIGNATURE);
          ^
.piolibdeps/testac/HeatPump/src/HeatPump.h:233:10: note:   no known conversion for argument 1 from 'MitsubishiHeatPump::setup()::<lambda(heatpumpStatus)>' to 'void (*)(heatpumpStatus)'
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:459:64: error: no matching function for call to 'HeatPump::connect(HardwareSerial*, int&, int, int)'
     if (hp->connect(this->get_hw_serial_(), this->baud_, -1, -1)) {
                                                                ^
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:23:0,
                 from src/esphome.h:34,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
.piolibdeps/testac/HeatPump/src/HeatPump.h:192:10: note: candidate: bool HeatPump::connect(HardwareSerial*)
     bool connect(HardwareSerial *serial);
          ^
.piolibdeps/testac/HeatPump/src/HeatPump.h:192:10: note:   candidate expects 1 argument, 4 provided
.piolibdeps/testac/HeatPump/src/HeatPump.h:193:10: note: candidate: bool HeatPump::connect(HardwareSerial*, bool)
     bool connect(HardwareSerial *serial, bool retry);
          ^
.piolibdeps/testac/HeatPump/src/HeatPump.h:193:10: note:   candidate expects 2 arguments, 4 provided`

I have no programming skills. So i cant help further.

@geoffdavis perhaps it's not a complete rewrite?

Thank you.

geoffdavis, 2022-09-21:
Pull requests gratefully accepted. $DAYJOB has me flat out for the moment.

geoffdavis, 2022-09-21:
Since the library has changed significantly, we need to lock to this particular version of the SwiCago library. I think a new ticket for supporting version 1.0.0 of SwiCago's library is the correct way to handle this.

closed 2022-09-21 by geoffdavis

grischard, 2022-10-27:
Duplicate of #70