geoffdavis/esphome-mitsubishiheatpump #71: Current Temperature: nan°C

kikonkike, 2022-10-04:
Hello

First of all thanks for your work :)

I have managed to make it work in a duct machine, everything works except that the temperature of the room comes out as "Current Temperature: nan°C"

[image]

Do you know if there is any way to solve this and take the value of the machine???

In Home assistant I don't get the temperature of the room either.

I've tried with the service in home assistant, but I can't get anything, I'm sure I'm doing something wrong....

api:
   services:
     - service: use_internal_temperature
       then:
         - lambda: 'id(hp).set_remote_temperature(0);'

This is the climate I use:

climate:
   - platform: mitsubishi_heatpump
     id: 'hp'
     name: $upper_devicename
     baud_rate: 2400
     update_interval: 500ms
     supports:
       swing_mode: []
       fan_mode: ['AUTO', 'LOW', 'MEDIUM', 'MIDDLE', 'HIGH']
     visual:
       min_temperature: 17
       max_temperature: 28
       temperature_step: 0.5 #temperature scaling steps

Thank you very much :)

geoffdavis, 2022-10-07:
Wild guess is bidirectional comms are not actually working. I originally had the library not update until the heat pump acknowledged the change, but now it optimistically sets the current state to the requested state...

All that's to say - you should verify that you actually have bidirectional comms working. Level shifters or voltage divider circuit.

kikonkike, 2022-10-07:
Ok, I'll try again, I had a Level shifters set, but I removed it because at first it didn't work for me and it turns out that it was because my machine needs 2400 baud to communicate :man_facepalming:.

Right now it's directly connected.

I'll try to put the Level shifters and I'll tell you.

Thanks for the reply :)

kikonkike, 2022-10-09:
Well, no luck, I've tried a voltage divider, but it still doesn't work, the rest of the values if I get them right, in fact I turn the machine on and off, I can set the different operating modes, I can set the temperature I want , that is if I can only put integers, that is, I cannot put the temperature at 25.5º, it leaves it at 25º....

It could be because it has two temperature probes, the machine is controlled from a PAR-31 thermostat and I think it has a temperature probe apart from the one on the machine.

[image]

I will continue investigating to see if I find why, it does not return the temperature of the room correctly,

Thank you very much

hornyak2, 2022-10-22:
Hello. First of all, I also wanted to say thanks! I have the same issue, so no temperature or mode feedback from the A/C. It means, if I switch on by the remote controller, HA says, that the machine is still off. I use Wemos D1 mini, with 10k pullup resistors. Control from HA works, but no feedback. 
I thing it uses older protocoll, because I also needed to set 2400 as bitrate. I tried to increase the update interval (thinking on slow communication) with no luck.
Type: MSZ-EF25VE2B (year of maufacture - 2015)

Has anyone some tips to where to start to find the problem?
Thank you!

DavidLBCN, 2022-12-08:
Hello,

I'm running the same issue than @kikonkike 

Does anyone have a solution or any additional information?

Thanks for all your work.

slackpj, 2023-07-04:
Hi
I've got the same issue.. I believe my issue is with the bidirectional communication too, as I can send commands (from home assistant TO Aircon), but if I use the IR remote control, theres no feedback of the changes (From Aircon TO Home Assistant) 

I'm using a  d1 mini v3.0.0...(https://www.wemos.cc/en/latest/_static/files/sch_d1_mini_v3.0.0.pdf)
I had to change my baud_rate to 2400 for anything to work
(tried 4800 and 9600 with and without the pull up resistors)

Originally had no pull up resistors, just used direct cable, and I had the same issue, so I've tired the pull up resistors, and I've still got the same problem

Any other ideas?

AC unit a new unit MSZ-AP25VGD (Australia) with no external temperature probes

echavet, 2023-07-23:
Same for me. With esphome. But it works well in both direction with [mitsubishi2MQTT](https://github.com/gysmo38/mitsubishi2MQTT).

nixmaster2000, 2023-10-10:
I have the same problem where I can send commands to the AC but can't get status back.

I have a ducted PEA-M100 indoor unit (Australia) with a PAR-41MAA wall controller. I'm controlling with a WT32-ETH01 connected via a level shifter (but also got the same results using 10k pull-ups).

The WT32-ETH01 has on-board LEDs tied to the TX and RX pins of UART2 so you can see when comms are happening. When sending a command from the ESP the TX LED gives a short flash followed by a short flash on the RX LED and the AC starts operating. If any commands are sent from the wall controller there are no LED flashes on the ESP, which suggests to me that this particular AC unit isn't offering updates via the CN105 connector.

I've been hunting around to see if there's a service setting in the AC to enable this but so far haven't found anything.

KazWolfe, 2023-10-13:
@nixmaster2000 I spent far too many hours debugging this today, and what I finally found is that the WT32-ETH01's UART ports apparently are... weird. I had to manually patch this library to define the pins to use for TX/RX.

This was done by editing this line:

https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/master/components/mitsubishi_heatpump/espmhp.cpp#L459

and making the following change:

```diff
-     if (hp->connect(this->get_hw_serial_(), this->baud_, -1, -1)) {
+     if (hp->connect(this->get_hw_serial_(), this->baud_, 5, 17)) {
```

This seems like a definitions issue *somewhere*, but I'm honestly not sure where to report and everything at least appears to be working now.

geoffdavis, 2023-10-13:
Good find! It sounds to me like the WT32-ETH01 uses different tx and rx pins than the defaults for that platform. I suppose the Platform definitions might be updated to make note of the different default tx and rx pins, but I have no idea how that works at the Platform.IO layer.

I personally haven't run any of this code on an ESP32 myself, I have it all running on ESP_01S boards. All of the ESP32 is from other contributors. With that said...

Also note that the code doesn't interact too closely with the UART component from ESPHome itself. There may be some options to set pins in a UART component, but I'm not using them because the SwiCago/HeatPump library didn't support them when I first wrote this.

Some breadcrumbs for anyone that wants to dig in further:

ESPHOME supports `tx_pin` and `rx_pin` for the [UART component](https://esphome.io/components/uart.html), but for our purposes it only has an effect on ESP32. (It forces a software serial mode on the 8266 which SwiCago/HeatPump doesn't know how to use.)

The [ESPHome code, at least for the ESP32](https://github.com/esphome/esphome/blob/da3e3903ddb36f5a5c3161bf6e23c8eadfdcfbbe/esphome/components/uart/uart_component_esp32_arduino.cpp#L115) takes those `tx_pin` and `rx_pin` definitions and uses them in a direct call to Serial.connect(), same as what SwiCago/HeatPump is doing.

I imagine that extending the [configuration schema](https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/980271337a87970ff0e3607c61ee12150c16f524/components/mitsubishi_heatpump/climate.py#L39) for this custom component to add `tx_pin` and `rx_pin` definitions (defaulting them to `-1`), and then passing that to the `hp->connect()` call would do what you need it to do.

nixmaster2000, 2023-10-13:
@KazWolfe do you still specify the UART pins in your yaml config after making this patch? I've attempted the change you made but on first go I'm not having any luck.

Here's my relevant yaml:

```
climate:
  - platform: mitsubishi_heatpump
    name: "Living Areas"
    hardware_uart: UART2
    baud_rate: 2400
    update_interval: 500ms

uart:
  tx_pin: 17
  rx_pin: 5
  baud_rate: 2400
  rx_buffer_size: 512
```

KazWolfe, 2023-10-13:
> I imagine that extending the [configuration [...]

[Seems to already be done](https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/91), nice. Didn't know that branch existed, but can give it a shot later.

> do you still specify the UART pins in your yaml config after making this patch? I've attempted the change you made but on first go I'm not having any luck.

No, I just manually patched the `.cpp` file. If the `develop` branch method works though, that'll fix this problem cleanly. To my knowledge, you don't need to define a `uart` stanza at all (and, in fact, doing so might cause issues.)

Just for the sake of experimentation, this *might* work?

```yml
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@develop

climate:
  - platform: mitsubishi_heatpump
    name: "Living Areas"
    hardware_uart: UART2
    tx_pin: 17
    rx_pin: 5
    baud_rate: 2400
    update_interval: 500ms
```
 
I'll give it a shot and report back as well.

Desperado88, 2023-11-05:
Hello,

Thank you for your integration work.

I tried to run my esp-01 module on my Mitsubitshi Ecodan ERST20D-VM2D heat pump, but without success.

Communication doesn't work at all. I have access to my esp but the instructions are not received by the heat pump, no change of instructions. I modified the BAUD, without success.

I followed the instructions available here:
https://github.com/geoffdavis/esphome-mitsubishiheatpump/tree/master

With Home Assistant 2023.11.1 and ESPHome 2023.10.6.

Do you have any ideas ?

Here are my logs and my installation:

```
INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/esp-01s-pac.yaml...
INFO Detected timezone 'Europe/Paris'
INFO Starting log output from esp-01s-pac.local using esphome API
INFO Successfully connected to esp-01s-pac.local
[22:25:02][I][app:102]: ESPHome version 2023.10.6 compiled on Nov  5 2023, 21:55:54
[22:25:02][C][wifi:546]: WiFi:
[22:25:02][C][wifi:382]:   Local MAC: B4:E6:2D:7B:FE:58
[22:25:02][C][wifi:383]:   SSID: [redacted]
[22:25:02][C][wifi:384]:   IP Address: 192.168.1.73
[22:25:02][C][wifi:385]:   BSSID: [redacted]
[22:25:02][C][wifi:387]:   Hostname: 'esp-01s-pac'
[22:25:02][C][wifi:389]:   Signal strength: -81 dB ▂▄▆█
[22:25:02][C][wifi:393]:   Channel: 11
[22:25:02][C][wifi:394]:   Subnet: 255.255.255.0
[22:25:02][C][wifi:395]:   Gateway: 192.168.1.1
[22:25:02][C][wifi:396]:   DNS1: 192.168.1.1
[22:25:02][C][wifi:397]:   DNS2: 0.0.0.0
[22:25:02][C][logger:416]: Logger:
[22:25:02][C][logger:417]:   Level: DEBUG
[22:25:02][C][logger:418]:   Log Baud Rate: 0
[22:25:02][C][logger:420]:   Hardware UART: UART0
[22:25:02][C][uptime.sensor:031]: Uptime Sensor 'hptest Uptime'
[22:25:02][C][uptime.sensor:031]:   Device Class: 'duration'
[22:25:02][C][uptime.sensor:031]:   State Class: 'total_increasing'
[22:25:02][C][uptime.sensor:031]:   Unit of Measurement: 's'
[22:25:02][C][uptime.sensor:031]:   Accuracy Decimals: 0
[22:25:02][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[22:25:02][C][homeassistant.time:010]: Home Assistant Time:
[22:25:02][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[22:25:02][C][version.text_sensor:021]: Version Text Sensor 'hptest ESPHome Version'
[22:25:02][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[22:25:02][I][MitsubishiHeatPump:062]: ESPHome MitsubishiHeatPump version 2.4.1
[22:25:02][I][MitsubishiHeatPump:504]:   Supports HEAT: YES
[22:25:02][I][MitsubishiHeatPump:505]:   Supports COOL: YES
[22:25:02][I][MitsubishiHeatPump:506]:   Supports AWAY mode: NO
[22:25:02][I][MitsubishiHeatPump:507]:   Saved heat: -1.0
[22:25:02][I][MitsubishiHeatPump:508]:   Saved cool: -1.0
[22:25:02][I][MitsubishiHeatPump:509]:   Saved auto: -1.0
[22:25:02][C][captive_portal:088]: Captive Portal:
[22:25:03][C][web_server:161]: Web Server:
[22:25:03][C][web_server:162]:   Address: esp-01s-pac.local:80
[22:25:04][C][mdns:115]: mDNS:
[22:25:04][C][mdns:116]:   Hostname: esp-01s-pac
[22:25:04][C][ota:097]: Over-The-Air Updates:
[22:25:04][C][ota:098]:   Address: esp-01s-pac.local:8266
[22:25:04][C][ota:101]:   Using Password.
[22:25:04][C][api:139]: API Server:
[22:25:04][C][api:140]:   Address: esp-01s-pac.local:6053
[22:25:04][C][api:142]:   Using noise encryption: YES
[22:25:04][C][wifi_info:009]: WifiInfo IPAddress 'hptest IP'
[22:25:04][C][wifi_info:011]: WifiInfo SSID 'hptest SSID'
[... 105 lines cut ...]
climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    id: hp

    # ESP32 uniquement - remplacez UART0 par UART1 ou UART2 et supprimez le
    # logging:baud_rate ci-dessus pour permettre à l'UART0 intégré de fonctionner pendant
    # journalisation.
    hardware_uart: UART0
    # Décomenter les lignes suivantes en fonction des usages, defaut 4800
    baud_rate: 2400 #0, 2400 ou 9600, defaut 4800
    supports:
       mode: [ COOL, HEAT]
    #   fan_mode: [AUTO, LOW, MEDIUM, HIGH]
    #   swing_mode: [OFF, VERTICAL]
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 1.0
```

[image]
[image]

Thank you so much

echavet, 2023-11-06:
Hello,
I had the same problem.
So I changed many little things looking at the code from mistubishi2mqtt,
because this one was working well with my heaters.

https://github.com/gysmo38/mitsubishi2MQTT

Now I've got it working...
You can have a look if you want.

Here is my espconf:
Note that I'm using my GitHub rep.

substitutions:
  name: "hpsejour"
  friendly_name: Climatisation Séjour

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: esp01_1m

mqtt:
  broker: homeassistant.local
  username: heatpumpSejour
  password: !secret mqtt_password
  discovery: true

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG
  logs:
    MitsubishiHeatPump: DEBUG
    HeatPump: WARN
    json: WARN
    sensor: WARN
    #packetSent : DEBUG
    #packetRecv : DEBUG
    #Decoder    : VERBOSE

# Enable Home Assistant API
api:
  services:
    - service: reboot
      then:
        - logger.log: "Redémarrage en cours..."
        - lambda: |-
            ESP.restart();

#  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(hp_sejour).set_remote_temperature(temperature);'

    - service: use_internal_temperature
      then:
        - lambda: 'id(hp_sejour).set_remote_temperature(0);'

  encryption:
    key: "***"

ota:
  password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heatpump-Sejour Fallback Hotspot"
    password: "****"

captive_portal:

web_server:
  port: 80
  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

external_components:
  - source: github://echavet/esphome-mitsubishiheatpump
    refresh: 0s

##external_components:
##  - source:
##      type: local
##      path: components/mitsubishi_heatpump

        # Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    id: hp_sejour
    baud_rate: 0
    # Optional
    hardware_uart: UART0
    # Optional
    update_interval: 500ms
    visual:
      min_temperature: 18
      max_temperature: 28
      temperature_step: 0.5
    action_state_topic: "mitsubishi2mqtt/sejour/state"
    away_command_topic: "mitsubishi2mqtt/sejour/away/set"
    away_state_topic: "mitsubishi2mqtt/sejour/state"
    current_temperature_state_topic: "mitsubishi2mqtt/sejour/state"
    fan_mode_command_topic: "mitsubishi2mqtt/sejour/fan/set"
    fan_mode_state_topic: "mitsubishi2mqtt/sejour/state"
    mode_command_topic: "mitsubishi2mqtt/sejour/mode/set"
    mode_state_topic: "mitsubishi2mqtt/sejour/state"
    preset_command_topic: "mitsubishi2mqtt/sejour/preset/set"
    preset_state_topic: "mitsubishi2mqtt/sejour/state"
    swing_mode_command_topic: "mitsubishi2mqtt/sejour/vane/set"
    swing_mode_state_topic: "mitsubishi2mqtt/sejour/state"
    target_temperature_command_topic: "mitsubishi2mqtt/sejour/temp/set"
    target_temperature_state_topic: "mitsubishi2mqtt/sejour/state"
    target_temperature_high_command_topic:
"mitsubishi2mqtt/sejour/temp_high/set"
    target_temperature_high_state_topic: "mitsubishi2mqtt/sejour/state"
    target_temperature_low_command_topic:
"mitsubishi2mqtt/sejour/temp_low/set"
    target_temperature_low_state_topic: "mitsubishi2mqtt/sejour/state"

Desperado88, 2023-11-06:
Hello,

Thank you for your reply.

I modified my code to match yours. It seems to work better but the commands still do not work, even though the connection seems established (see log).

I tried the MQTT code alone, but the result is worse, it can't even connect to the boiler.

Does anyone understand what Warnings [W] mean ?
I also don't understand what the last lines of your code, "action_state_topic: "mitsubishi2mqtt/sejour/state"", are for files on Git?

Always up for ideas, thank you.

Here are my logs:

```
[20:34:51][D][MitsubishiHeatPump:062]: update() called
[20:34:51][D][MitsubishiHeatPump:067]: hp->isConnected=false ;@( 
[20:34:53][D][MitsubishiHeatPump:373]: Connection à la clim établie
[20:34:53][D][packetSent:399]: FC 5A 01 30 02 CA 01 A8 
[20:34:54][W][HeatPump:639]: did not find any start, returning an error...
[20:34:54][W][component:214]: Component esphome.coroutine took a long time for an operation (3.03 s).
[20:34:54][W][component:215]: Components should block for at most 20-30ms.
[20:34:54][C][logger:416]: Logger:
[20:34:54][C][logger:417]:   Level: DEBUG
[20:34:54][C][logger:418]:   Log Baud Rate: 115200
[20:34:54][C][logger:420]:   Hardware UART: UART1
[20:34:54][C][logger:424]:   Level for 'MitsubishiHeatPump': DEBUG
[20:34:54][C][logger:424]:   Level for 'HeatPump': WARN
[20:34:54][C][logger:424]:   Level for 'json': WARN
[20:34:54][C][logger:424]:   Level for 'sensor': WARN
[...]
[20:35:09][I][MitsubishiHeatPump:064]: ESPHome MitsubishiHeatPump version 2.4.1
[20:35:09][I][MitsubishiHeatPump:617]:   Version de Eric Chavet 1.0
[20:35:09][I][MitsubishiHeatPump:618]:   Supports HEAT: YES
[20:35:09][I][MitsubishiHeatPump:619]:   Supports COOL: YES
[20:35:09][I][MitsubishiHeatPump:620]:   Supports AWAY mode: NO
[20:35:09][I][MitsubishiHeatPump:621]:   Saved heat: -1.0
[20:35:09][I][MitsubishiHeatPump:622]:   Saved cool: -1.0
[20:35:09][I][MitsubishiHeatPump:623]:   Saved auto: -1.0
[20:35:09][I][MitsubishiHeatPump:624]:   ---- 
[20:35:09][I][MitsubishiHeatPump:625]:   hp : 
[20:35:09][I][MitsubishiHeatPump:626]: heatpump connected:
[20:35:09][D][MitsubishiHeatPump:627]: heatpump room temp: 0.000000
[20:35:09][I][MitsubishiHeatPump:628]: heatpump room temp: 0.000000
[20:35:09][D][MitsubishiHeatPump:629]: heatpump operating: 
[20:35:09][I][MitsubishiHeatPump:630]:   ---- 
[20:35:09][I][MitsubishiHeatPump:635]: Using callbacks
[20:35:09][D][MitsubishiHeatPump:062]: update() called
[20:35:09][D][MitsubishiHeatPump:067]: hp->isConnected=false ;@( 
[20:35:11][D][MitsubishiHeatPump:373]: Connection à la clim établie
[20:35:11][D][packetSent:399]: FC 5A 01 30 02 CA 01 A8 
[...]
[20:37:37][D][climate:011]: 'PAC' - Setting
[20:37:37][D][climate:040]:   Target Temperature: 21.00
[20:37:37][D][MitsubishiHeatPump:256]: control - Was HeatPump updated? YES
[20:37:37][D][climate:380]: 'PAC' - Sending state:
[20:37:37][D][climate:383]:   Mode: HEAT
[20:37:37][D][climate:385]:   Action: IDLE
[20:37:37][D][climate:388]:   Fan Mode: OFF
[20:37:37][D][climate:400]:   Swing Mode: OFF
[20:37:37][D][climate:403]:   Current Temperature: 0.00°C
[20:37:37][D][climate:409]:   Target Temperature: 21.00°C
[20:37:37][D][packetSent:399]: FC 41 01 30 10 01 07 00 01 01 0A 00 00 00 00 00 00 00 00 00 00 6A 
[20:37:38][W][HeatPump:864]: AUCUNE donnée disponible!
[20:37:38][D][Decoder:867]: [packet decode is a FAILLURE]
[...]
[20:37:59][I][ota:117]: Boot seems successful, resetting boot loop counter.
```

![Capture d’écran 2023-11-06 à 20 26 53]

Thank you

(https://github.com/geoffdavis/esphome-mitsubishiheatpump/assets/102601357/5d5901b3-3ebf-4c05-98fc-3e52c5539f88)

echavet, 2023-11-07:
Hey

Sorry Geoff I don't remember why I chose to customize my Matt topics. This
is not code, these last lines are just custom topics for the mqtt.
I don't think that the mqtt config is mandatory because you can use the
esphome homeassistant integration to access your heater.
I had to make it work through mqtt.

echavet, 2023-11-07:
I have added some logs in the code. Some logs are not reliable.
It says it is connected to the heater. But it is not.
The heater does not seem to be connected well in yours logs.
Sorry you have to dig deeper !
Perhaps you did not connect the Rx tx correctly. Perhaps your heater does
not work with this project.
In my case the esp couldn't get it's UART to work with the heater until I
unplugged the programming usb port from my raspberry pi.
I had 2 uart in my esphome yaml conf because one was used for the heater
and the other for me logs.
But I don't know why both didn't work together at the same time.

Desperado88, 2023-11-07:
Hello,

Thank you very much,

I tested by changing the rx and tx, but the result is the same.
If I comment on bebug's lines, I can't do anything anymore.

My heat pump is an air/water pump not air/air, that may be the problem.
I connected a green USB UART adapter to the CN105 socket, but I didn't see any frames. It doesn't matter what the BAUD is. It may not be possible on my device to connect a wifi module, you are right.

Thanks for your help anyway.

adepssimius, 2023-11-11:
@KazWolfe @geoffdavis I tried your configuration with a WT32-ETH01 board and when I went to install on the board I got a bunch of compile errors from the develop branch. I don't do C++ so I'm not sure where to go with it. I do see part of the error stating `note: 'std::optional' is only available from C++17 onwards`, but not sure what to do about that.

```Compiling .pioenvs/wt32-eth01-2/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
Compiling .pioenvs/wt32-eth01-2/src/main.cpp.o
In file included from src/esphome.h:28,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:21:
src/esphome/components/mitsubishi_heatpump/espmhp.h:163:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_operating_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:163:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_operating_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:164:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_idle_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:164:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_idle_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:165:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_ping_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:165:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_ping_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:166:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_remote_temperature_sensor_update_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:166:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_remote_temperature_sensor_update_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:167:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:167:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::control(const esphome::climate::ClimateCall&)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:129:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
     if (last_remote_temperature_sensor_update_.has_value()) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:129:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
     if (last_remote_temperature_sensor_update_.has_value()) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::set_remote_temperature(float)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:450:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
         last_remote_temperature_sensor_update_ =
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:450:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
         last_remote_temperature_sensor_update_ =
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:453:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
         last_remote_temperature_sensor_update_.reset();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:453:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
         last_remote_temperature_sensor_update_.reset();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::ping()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:461:5: error: 'last_ping_request_' was not declared in this scope
     last_ping_request_ = std::chrono::steady_clock::now();
     ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::set_remote_operating_timeout_minutes(int)':
[... cut ...]
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:167:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
         ^~~
*** [.pioenvs/wt32-eth01-2/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
*** [.pioenvs/wt32-eth01-2/src/main.cpp.o] Error 1
========================= [FAILED] Took 25.46 seconds =========================```

echavet, 2023-11-11:
Esp32 and Esp8266 does not share the same libs. I don't think this conf
works with esp32.

adepssimius, 2023-11-11:
@echavet The poster who stated that the configuration worked on the develop branch was using the same ESP32 based WT32-ETH01 that I am using.

KazWolfe, 2023-11-12:
@adepssimius See #117. Add this to your YAML:

```yml

esphome:
  platformio_options:
    build_unflags: 
      - "—std=gnu++11"
    build_flags: 
      - "-std=gnu++17"
```

echavet, 2023-11-12:
ah ok sorry

stewartadam, 2023-11-18:
I have also been running into this and determined it to be an issue with recent changes to the SwiCago/HeatPump library. It looks like the version pinning attempted while adding the library in `climate.py` here fails, and it's always pulling from `master` instead.

Reverting to commit SwiCago/HeatPump commit cea90c5ed48d24a904835f8918bd88cbc84cb1be fixes the issue immediately. I'll have PR out shortly.

echavet, 2023-11-20:
Oh sorry I had not read your message well !

Marksonamap, 2023-11-21:
@slackpj I think I might be in the same boat as you, I have new MSZ-AP42VGD and MSZ-AP22VGD models (Australia) and using D1 Minis. I can turn the units on and off, but can't set temperature and don't have any communication from the unit back to HA.

Thanks for sharing your solution. Any chance you have a photo of pins you had to cut off? I'm a little lost with the schematic.

echavet, 2023-11-22:
I have 2 Mitsubishi hp.
I used to have the same pb. I've tried to add some debug instructions in the code. I've directly included the swicago lib in the esphome conf. And one day it began to work with my 2 hp units.
I've been using my testing repository since this day! 
Last week I wanted to install the same wemos d1 mini in a friend of mine's heat pump. He has exactly the same model.
I use exactly the same code. 
I can't manage to get rid of this PB with this last unit. 
Nan temp. Which means communication is broken with the heat pump.
I've changed the cable. The esp. Still doesn't work... still digging. Perhaps a pin is tied inside the male hp connector? I have to check...
I've read all this is about a swicago regression on the head branch. (Because the version in the climate.py is ignored)

Guisch, 2023-11-23:
👋🏼 Had this issue recently. Going back to v2.4.0 fixed it.

```yaml
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@2.4.0
```

danolson1, 2023-12-11:
I had a communication problem and solved it with a pull-down resistor. See detailed description [here](https://github.com/unixko/MitsuCon/issues/15#issuecomment-1849293534).

echavet, 2023-12-11:
Oh thank you for the tip.
I have noticed many checksum errors in the com too. I'was about to test a
pull-down too.
But why are you using a level converter?

danolson1, 2023-12-11:
Mitsubishi TTL is 5v, Wemos D1 wants 3.3v TTL

echavet, 2023-12-12:
thanks.
why did you connect both 5 and 3.3?

my experience shows that:
wemos d1 mini does it with direct connection to the hp but with some data
corruption
adding an ext 3.3 converter and connecting it to the esp 5v in ou 3.3 does
remove 99% of the checksum errors.
with no resistance

Envoyé avec Gmail Mobile

danolson1, 2023-12-12:
I connected the 5v supply from the mitsubishi to the D1 to provide power.

How are you powering yours?

-Dan

echavet, 2023-12-12:
Wow I didn't know you could do that.
I have first connected the 5v hp to the the esp. It was working but I had
many loss.
I'm writing a new version of the EspHome MitsubishiHeatPump with the
SwiCago directly included and with no delays (
https://github.com/echavet/MitsubishiCN105ESPHome). EspHome is not meant to
have some delays in the loop method.
So I added some logs and I noticed a lot of checksum errors.
Now I got everything ok but I still have a little loss. My 2 esp are
connected to my 2 hp the same way, with a regulator (
https://www.amazon.fr/dp/B072FTMS89?ref=ppx_yo2ov_dt_b_product_details&th=1)
but connected to the 5v input.
The 3.3 was not soldered ! So I tried and it worked very well. But I have
to check if it work better with the AMS1117 output connected to the 3.3 esp
input.

Also I was logging on the UART1 (d4 pin) on a USB to TTL (USB CH340C)
directly connected to a Pi3. And I had to connect the grounds of the USB
CH340C to  the hp ground and to the esp ground to all this to work
together. And I noticed I had no loss this way. No chkSum error.

danolson1, 2023-12-12:
I'm curious to see a photo or a wiring diagram of your setup.

The part you described (AMS1117) looks like a voltage regulator. So I'm
assuming you power it with 5v from your heat pump and it outputs 3.3v that
you then use to power your D1. One potential problem with your setup is
that you are connecting the 5v TTL signal from the heat pump to the 3.3v
TTL signal from the D1. The 3.3v signal from your D1 is probably read just
fine by the heat pump, but the 5v signal it sends back could potentially
damage the D1.

-Dan

jemarr, 2023-12-12:
I had this issue as well (nanºC) as Wemos D1 Mini didn't receive any data from AC unit. I can send data but no response.

I thought to put a pulldown resistor on Rx line as Danolson1 suggested, but finally I just cutted the Rx trace just before CH340 chip on Wemos D1. Now everything is ok. Upsate software is with OTA, so no need Rx line anymore.

[image]

echavet, 2023-12-13:
Thanks for these informations.
I didn't even know there was a 3.3 or 5v ttl. I thought it was 5v.
So what you're saying is that resistance is not an option?
ok I was about to add it since you post your analyse... but did not test yet
https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/captures/IMG_20231213_165310.jpg

this one is not a good picture but it wasn't meant to be published. It's
when I was debugging:
https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/captures/PXL_20231206_200948742.jpg

These 2 are not connected to the hp.
https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/captures/PXL_20231211_132305690.jpg

danolson1, 2023-12-13:
You can definitely convert logic levels using discrete components (such as
resistors), but that involves a bit of design work. The whole point of
logic level converters is that they're cheap and do the job well.

I'm curious about your post where you say you "added logs and saw checksum
errors." Can you provide a detailed description of how you did this? I
would like to try this myself, to confirm that my pull-down resistor method
is not causing software errors, but am not sure where to start.

-Dan

echavet, 2023-12-13:
Well, two things:

1st:  With the original geoffdavis esphome-mitsubishiheatpump version the
only way to log was to log over the air (OTA) .

So I began a new branch which have many log instructions added. The way to
use is is to change the line :

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

by

external_components:
  - source: github.com/echavet/esphome-mitsubishiheatpump

in your yaml file.

But this was not enough because the setup method, where the first
connection is made was not present in the traces. This because OTA logs are
too long to get initialized.
So I decided to use the second UART of the ESP for logs (because the 1st
one is used for the hp of course). This one only has one line for sending.
So I connected it to the TTL converter like on this picture:
https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/captures/PXL_20231206_200948742.jpg
The d4 pin of the ESP has to be connected to the TXD pin of the TTL
converter. You also have to connect the ground of the USB TTL converter to
the ESP and to the HP. That's because TTL is relative to ground.
Then I just had to connect the TTL converter to the usb port of a Pi and to
launch an "esphome logs" (over serial) command with my yaml configured with
logger to UART1.

logger:
  hardware_uart: UART1
  level: DEBUG

But actually it wasn't enough. Because I had the damn T° : Not a Number
issue on a friend of mine hp.

2nd thing:
So I decided to dig deeper. And I made another version of the
esphome-mitsubishiheatpump geoffdavis project. A version in which I do the
initializations later.
In this repo I have included and modified the SwiCago librarie. But without
all the delays instructions which are normally forbidden in esphome
architecture.

You can see in this file that there are buttons to launch the setupUART and
to send the first connection packet.

https://github.com/echavet/MitsubishiCN105ESPHome/blob/main/hp-debug.yaml

I find my version much better (because it is). No need to wait 2 seconds
each time a packet is sent because the loop method always reads the UART
response of the HP and never misses a packet. Except when I get checksum
errors.

All you have to do is to change the external component ref to this:

external_components:
  - source: github://github.com/echavet/MitsubishiCN105ESPHome

or to get the code in your esphome directory and to reference it locally
like this:

external_components:
 - source:
     type: local
     path: components

[... 138 lines cut ...]
[18:15:23][W][component:215]: Components should block for at most
20-30ms.
[18:15:23][D][CN105:1144]: sending a request room temp packet (0x03)
[18:15:23][D][CN105:1161]: creating Info packet
[18:15:23][D][CN105:998]: writing packet...
[18:15:23][D][WRITE:1226]: FC 42 01 30 10 03 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 7A
[18:15:23][D][Header:523]: command: (62) data length: [10]<-- header
[18:15:23][D][READ:1226]: FC 62 01 30 10 03 00 00 0A 00 00 A9 00 00 00
00 00 00 00 00 00 A7
[18:15:23][D][Decoder:604]: [0x03 room temperature]
[18:15:23][D][Decoder:614]: [Room °C: 20.500000]
[18:15:23][D][CN105:1147]: sending a request status paquet (0x06)
[18:15:23][D][CN105:1161]: creating Info packet
[18:15:23][D][CN105:998]: writing packet...
[18:15:23][D][WRITE:1226]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 77
[18:15:23][D][Header:523]: command: (62) data length: [10]<-- header
[18:15:23][D][READ:1226]: FC 62 01 30 10 06 00 00 07 01 00 00 00 00 00
00 00 00 00 00 00 4F
[18:15:23][D][Decoder:635]: [0x06 is status]
[18:15:23][D][Decoder:643]: [Operating: 1]
[18:15:23][D][Decoder:644]: [Compressor Freq: 7]

Hope it helps

echavet, 2023-12-13:
About the 3.3 or 5v TTL Logic, here is what I found:

GPIOs

The ESP8266 has 16 GPIOs (General Purpose Input/Output) that can be used to control external devices. The GPIOs are numbered from GPIO0 to GPIO15.

RX and TX

The RX and TX pins are used for serial communication. The RX pin receives data from the serial port and the TX pin sends data to the serial port. **The RX and TX pins are compatible with 5V TTL logic**.

jpollard-cs, 2023-12-16:
I have the same problem with bidirectional comms not working and I’m using ESP32 on MSZ-GL****. I noticed there wasn’t anything in the docs about setting up MQTT in the config. Is this just assumed by experienced ESPHome users? I’m completely new to the HA ecosystem. Happy to contribute to the docs. I also saw the notes in the thread about RX/TX issues, but want to make sure I have everything in order before attempting with the dev branch.

Any interest in a Discord @geoffdavis? I’d be happy to help set it up and administer / moderate it.

jpollard-cs, 2023-12-16:
I should add before I tried the ESPHome integration I compiled and uploaded the relevant sketch in Arduino studio and it was communicating over MQTT just fine. Now I’m not seeing anything in the logs for this client / device so I assume that’s part of the problem if not the whole problem. The web server is running and I can reach the device fine.

jpollard-cs, 2023-12-19:
Managed to get this working with the develop branch as others had mentioned. Working example:

```yaml

substitutions:
  name: nunya-biznass-mini-split
  friendly_name: Nunya Biznass Mini Split

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"

preferences:
  flash_write_interval: 10min

esp32:
  board: nodemcu-32s

# Enable logging
logger:
  # level: VERBOSE

api:
  encryption:
    key: "****"
 
ota:
  password: "****"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret fallback_ssid
    password: !secret fallback_password

mqtt:
  broker:!secret mqtt_host
  port: !secret mqtt_port
  username: !secret mqtt_username
  password: !secret mqtt_password

captive_portal:

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@develop

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    hardware_uart: UART2
    tx_pin: 17
    rx_pin: 16
    baud_rate: 2400
    update_interval: 500ms
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step: 0.1
```

danolson1, 2023-12-22:
@echavet Eric, does your logging require a second UART? Or is it possible to use UART0 for communication with the heatpump and send the logs to ESPHome (or MQTT) via wifi?

-Dan

danolson1, 2023-12-22:
@jpollard-cs Jordan, we should find a place to collect working examples. I've got a working example for an ESP32-C3 from Seeed studios.

-Dan

jpollard-cs, 2023-12-23:
> @jpollard-cs Jordan, we should find a place to collect working examples. I've got a working example for an ESP32-C3 from Seeed studios.
> 
> 
> 
> -Dan

I thought this thread was where we were doing that? 😆

It does seem like development has slowed on this project, but out of respect I'd like to defer to the author before suggesting we spin up a fork. I'd also be happy to setup a discord server. It would really be nice to have a general ESP Discord even.

adepssimius, 2023-12-24:
@jpollard-cs There is already an (official, I think) ESPHome discord. https://discord.com/invite/Ve73Ma32ce

echavet, 2023-12-24:
For info,
I have started a new version of the project which has included the swicago
lib directly into the esphome component code. This looks like a regression
because the lib is not a lib anymore. But actually, the swicago lib does
not fit the esphome requirements:

The lib requires a delay each time a packet is written so as the read
process only starts when it is sure the response packet is complete, and it
can read it in a single loop.

This is not the right way to do.

Like it is done in the Firmata project the main loop must always read
packets from the input UART and write when needed.
This way, we don't loose packet, and the component matches the esphome
requirements (no delay). Also there is many logs in the code.

It is working well on a esp8266 D1 mini and with my 2 Mitsubishi hp. But
the project needs to be tested and improved...

Here is the code.

https://github.com/echavet/MitsubishiCN105ESPHome

We could start a new project with the best of both projects.

geoffdavis, 2023-12-24:
Development has indeed slowed on this project, and I'm unsure of the best way to proceed. I keep hoping that I'll have time to work on it, but that's clearly not happening as I'm spending less time in front of a desk and more time out in the field.

I'm happy to allow other admins to vet commits if that makes sense, or transfer ownership if that makes things easier.

However, I'd like to see compatibility maintained with ESP8266 units, as that's what I'm using and presumably others are using it as well.

danolson1, 2023-12-25:
Here's a working example for controlling a Mitsubishi heat pump with a Seeed XIAO ESP32C3 board.

[image]

```
# Notes: need to attach wifi antenna for good signal reception
# https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html
# Use pin D6 for serial Tx, use pin D7 for serial Rx. 
# Connect serial pins to level converter to interface 5v TTL signal from heat pump to 3.3v TTL signal of ESP32C3

substitutions:
  name: hp-2
  friendly_name: HP-2

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
    board_build.extra_flags:
      # Override, defaults to '-DARDUINO_USB_CDC_ON_BOOT=1'     
      # This is necessary to allow mitsubishi heatpump to work correctly
      # More info here https://wiki.seeedstudio.com/XIAO_ESP32C3_Pin_Multiplexing/
      - "-DARDUINO_USB_CDC_ON_BOOT=0"  

esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.4.0 # 6.4.0 and 5.4.0 worked

# Enable logging
logger:
 #hardware_uart: UART0
 baud_rate: 0 # disable the logger

# Enable Home Assistant API
api:
  encryption:
    key: "*****"

ota:
  password: "*****"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "*****"
    password: "*****"

web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    
    # ESP32C3, must be UART0
    hardware_uart: UART0
```

echavet, 2023-12-25:
Yes you can use UART0 for the heat pump and UART1 for logging.
Logging through WiFi or mqtt remains possible of course as itis the esphome
logging feature.
I have just added log instructions in the code, and tested on UART1 of
esp8266.

Protoncek, 2024-01-03:
Hi guys!
A couple of days ago i built and installed my esp module, too. I experienced a strange phenomenon, though: each time i make OTA upgrade of FW it seems that esp doesn’t communicate with A/C after finished upgrade (i get no temp readings)l BUT…if i manually reboot esp again (i have reboot button programmed) all starts to work and it works perfectly afterwards. Did anyone had such a problem? What could be the cause?

Reagarding UART communication: esp modules with built-in UART-to-USB chip can be nasty… some of these chips tend to hold tx or rx level up or down when not in use, so no communication via uart is possible with other devices. So, for this purpose it’s best to use a module without usb, say, esp01. It’s cheap, ultrasmall and it works. Or, as someone above did: cut traces from esp to uart chip.

Finally, regarding level converter: i didn’t use it, but it works. Original schematic doesn’t use it also. So, i wonder…is it needed or not?  Esp module’s data say pins are 5v tolerant, so we can’t destroy esp, while sending 3.3V is already enough for AC to see this voltage as active 1… any thoughts?

Protoncek, 2024-01-03:
I agree, but, on the other hand…isn’t a reboot made immediately after succesfull OTA?  It smells as that reboot isn’t exactly as clean as it should be… it makes me think if anything similar as “forced second reboot” after ota update is possible to program into yaml…

elreydetodo, 2024-01-03:
@Protoncek there are some ESP32 pins that change the boot mode if they are pulled high or low during boot. I wonder if there might be a problem with that, or possibly the receiving of data during the reboot is putting it into an odd state. Does it happen predictably after _all_ updates, or only sometimes?

Protoncek, 2024-01-03:
That’s …possible, but not likely… board is ESP01 (so it’s 8266), and after each OTA update sensors like IP, MAC, wifi signal are always there, it’s just AC part that’s offline until i do a second reboot.  So, esp is fully booted and functional. I have pull-up resistors on tx and rx signals, like on original schematic.

Protoncek, 2024-01-04:
Interesting… today i updated my HA (to 2024.1) and after that it happened same thing: temperature unavailable. After esp restart all is ok again… so it’s something between HA and module communication… ? But only for climate…? (As said, other sensors always work and did work today, too).

echavet, 2024-01-05:
For information: one of the 3 heat pumps on which I installed ESP8266.
One of them couldn't get connected whereas it was just the same settings of the two others. 
I've changed the cables, swapped the chips... No way to make the 3rd one work.
So I tried to use a 3.3v regulator. The reason is a long story.
The result is that it works now.
Here is the module I'm using:
AZDelivery 3 x AMS1117 Module... https://www.amazon.fr/dp/B07CPXVDDN?ref=ppx_pop_mob_ap_share

echavet, 2024-01-06:
One thing is sure: adding a regulator between the heat pump 5v and the chip
vin makes the connection far more reliable.

I tested it on a heat pump that has been refusing to communicate on the
UART until I installed one.

On 2 other heat pumps on which all seemed to work well, I've noticed some
packet loss. No more loss with the regulator.

Check this one's:
AZDelivery 3 x AMS1117 Module d'alimentation 3.3V Compatible avec Raspberry
Pi incluant Un EBook! https://amzn.eu/d/0AJG871

Protoncek, 2024-01-06:
Hm… which esp module did you use? I mean - does esp even survives 5V power supply? I know i destroyed esp module before just because i connected to 5V power supply, so i thought it’s mandatory (and i do have it).

danolson1, 2024-01-06:
Yeah, most of the modules have a built-in voltage regulator that takes 5v
in and converts it to 3.3v for the chip. So Eric, if you're using a
separate voltage regulator to supply 3.3v directly to the chip, it seems
like you might just be bypassing the built-in voltage regulator. Maybe the
built-in voltage regulator on your board is damaged?

This setup should work fine, but I think it's unnecessary for most people.

-Dan

[5 more replies]

PR #125 (2024-06-23): Fix version pinning of SwiCago/HeatPump library
Pins the library to SwiCago/HeatPump@cea90c5ed48d24a904835f8918bd88cbc84cb1be which is the last stable commit I was able to test with my MSZ-GL06NA. More recent commits cause serial communication to silently fail (fixes #71 for me).

This PR also moves the version pin into the repository parameter as the `version` property of `cg.add_library()` appears to be entirely ignored in favor of version provided by the metadata in the repository instead. Since SwiCago doesn't version its library, we should version by Git commit instead.

Prior to this cha [... cut]
closed 2025-03-31 by geoffdavis