geoffdavis/esphome-mitsubishiheatpump #73: Unable to compile for WEMOS Lolin S2 Mini

gunnaraas, 2022-10-25:
When attempting to compile this for use with an S2 mini, I get a compilation error that seems to be related to the USB-C port on the board. 

esphome yaml:
```yaml
esphome:
  name: s2test

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.0.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S2Test Fallback Hotspot"
    password: !secret fallback_password

captive_portal:
    
web_server:
  port: 80

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "My Heat Pump"

    # Optional
    hardware_uart: UART0

    # Optional
    update_interval: 500ms
```

Terminal Output: 
``` bash
INFO Reading configuration /config/esphome/s2test.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/s2test/.pioenvs
INFO Compiling app...
Processing s2test (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32 @ 5.0.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
INFO Installing ottowinter/ESPAsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing git+https://github.com/SwiCago/HeatPump
INFO Installing git+https://github.com/SwiCago/HeatPump
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-fvejyke4'...
Library Manager: HeatPump@1.0.0+sha.d6a2913 has been installed!
INFO HeatPump@1.0.0+sha.d6a2913 has been installed!
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
|-- HeatPump @ 1.0.0+sha.d6a2913
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/api_connection.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/api_server.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/list_entities.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/proto.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/api/user_services.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/climate/climate.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/climate/climate_mode.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/climate/climate_traits.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/esp32/core.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/json/json_util.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/logger/logger.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/md5/md5.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/network/util.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:46: error: comparison between distinct pointer types 'HardwareSerial*' and 'USBCDC*' lacks a cast
             YESNO(this->get_hw_serial_() == &Serial)
                                              ^~~~~~
src/esphome/core/log.h:94:90: note: in definition of macro 'esph_log_config'
   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                          ^~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:449:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(
     ^~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:13: note: in expansion of macro 'YESNO'
             YESNO(this->get_hw_serial_() == &Serial)
             ^~~~~
Compiling /data/s2test/.pioenvs/s2test/src/esphome/components/socket/socket.cpp.o
*** [/data/s2test/.pioenvs/s2test/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 57.89 seconds =========================
```

mafrosis, 2022-12-06:
I just got this one too. I picked up the S2 mini as it's recommended by some of the folks over in https://gitter.im/Mitsubishi-Heat-Pump/Lobby as working without any additional resistors or breakout boards.

mafrosis, 2022-12-07:
I also found that `hardware_uart: UART0` fails to compile:

```
Compiling .pioenvs/f4accc/lib74d/WiFi/WiFiUdp.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:477:54: error: no matching function for call to 'MitsubishiHeatPump::MitsubishiHeatPump(USBCDC*)'
   mitsubishiheatpump = new MitsubishiHeatPump(&Serial);
                                                      ^
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/mitsubishi_heatpump/espmhp.h:55:9: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(HardwareSerial*, uint32_t)'
         MitsubishiHeatPump(
         ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:55:9: note:   no known conversion for argument 1 from 'USBCDC*' to 'HardwareSerial*'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(const MitsubishiHeatPump&)'
 class MitsubishiHeatPump : public PollingComponent, public climate::Climate {
       ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note:   no known conversion for argument 1 from 'USBCDC*' to 'const MitsubishiHeatPump&'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(MitsubishiHeatPump&&)'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note:   no known conversion for argument 1 from 'USBCDC*' to 'MitsubishiHeatPump&&'
Compiling .pioenvs/f4accc/lib590/FS/FS.cpp.o
Archiving .pioenvs/f4accc/libf23/libAsyncTCP-esphome.a
Compiling .pioenvs/f4accc/lib590/FS/vfs_api.cpp.o
*** [.pioenvs/f4accc/src/main.cpp.o] Error 1
Indexing .pioenvs/f4accc/libf23/libAsyncTCP-esphome.a
```

However, `UART1` works. In order for the device to operate you must set the baud rate as in #72

geoffdavis, 2023-01-21:
That check is there to make sure you're configuring the board correctly. It looks like I need to cast the pointers to void before doing the comparison.

The deal with the UART0/UART1 stuff is that you can't use the port that is configured for logging with the heat pump. If you want to use UART0, you have to disabled the serial logging (set the baud rate to zero in the logging section.) But this isn't necessary on ESP32 boards, just the ESP8266 boards.

I think you'll find that configuring the UART correctly this compilation problem will go away. Setting the hardware UART to 

That said, please post your config (with passwords sanitized.)

geoffdavis, 2023-01-21:
See GH-86 for void casting fix

Corruptsector, 2023-03-25:
> I just got this one too. I picked up the S2 mini as it's recommended by some of the folks over in https://gitter.im/Mitsubishi-Heat-Pump/Lobby as working without any additional resistors or breakout boards.

Hey, On a S2 mini how do you know which pins are the UART1?

imtheyoyo, 2023-10-06:
Hi,
Has anyone managed to get the WEMOS Lolin S2 Mini to work?
I would like to have the yaml code please.
Thanks.

gunnaraas, 2023-10-18:
> Hi, Has anyone managed to get the WEMOS Lolin S2 Mini to work? I would like to have the yaml code please. Thanks.

Hey @imtheyoyo , 

I had given up on getting it the S2 mini working, and opted to just set up with a HiLetGo ESP-01 ([amazon link](https://www.amazon.com/HiLetgo-Wireless-Transceiver-Development-Compatible/dp/B010N1ROQS)). Seemed that there was too much variance in quality & that the market was overrun with counterfeits at the time (November 2022). That would be my advise for how to reduce hassle with getting esphome on your heat pump.

jjhuff, 2023-10-26:
BTW, I got this to work w/ ESPHome 2023.10.3
```
esphome:
  name: mitsubishi 
  name_add_mac_suffix: true
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"

preferences:
  flash_write_interval: 10min

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino
    platform_version: 6.4.0

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@develop

# Enable logging
logger:
  #level: VERBOSE

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dashboard Fallback Hotspot"
    password: ""

captive_portal:

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

climate:
  - platform: mitsubishi_heatpump
    name: "Heat Pump"
    hardware_uart: UART1
    rx_pin: 18
    tx_pin: 16
    baud_rate: 2400
    update_interval: 500ms
 ```

MikeDembek, 2023-11-24:
I used the config from @jjhuff to get this going on on S2 mini hooked up to a MSZ-GL09NA using ESPHome - I can send commands and get what seems to be good comms back, I see successful serial response, but just get NaN for temps and no updates if I make a change over IR.  Any idea what I might be doing wrong?  Here's a log excerpt, the packetRecv seems like what we'd want to see but that's all I get from the HP.

`[16:49:47][V][MitsubishiHeatPump:622]: PKT: [packetSent] FC 41 01 30 10 01 0F 00 01 01 09 00 00 00 00 00 00 00 00 00 00 63
[16:49:48][V][MitsubishiHeatPump:622]: PKT: [packetRecv] FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E
[16:49:48][W][component:214]: Component api took a long time for an operation (1.18 s).
[16:49:48][W][component:215]: Components should block for at most 20-30ms.

@stewartadam described a similar-sounding issue in #125, but the commit his PR pins to results in compile issues in combo with the config from @jjhuff.  
`

SoftlyRaining, 2024-05-04:
Thanks for the merge! Casting to void* fixes the same issue I was having on adafruit_feather_esp32s2_tft

closed 2025-03-31 by geoffdavis