geoffdavis/esphome-mitsubishiheatpump #77: Support for dual set-point controls

LubergAlexander, 2022-10-28:
Hi!

The firmware has `set_supports_two_point_target_temperature` set to false. Is it possible to override this behavior and use dual set point thermostat in homeassistant?

I think it could be configured in homeassistant to switch between heat and cool modes if it goes higher or lower from set range, but it would be great to have it implemented as part of firmware.

pgenera, 2022-10-30:
I've been thinking about this for literal years, and really want it :)

The 'algorithm' used in the hp auto mode is documented in the mitsubishi manuals - room temp > setpoint for 15 minutes -> heat, the opposite for cool. The same thing would be simple enough to do on the microcontroller, just with two set points (and some minimum distance between them).

einstine909, 2022-11-03:
This can be done with another virtual thermostat climate that controls the heat pump:

```
globals:
  - id: minispit_ready
    type: bool
    restore_value: no
    initial_value: "false"
    
sensor:
  - platform: template
    id: current_temperature
    lambda: |-
      return id(mitsubishi).current_temperature;
    update_interval: 10s

climate:
  - platform: thermostat
    id: thermostat
    name: Minisplit Thermostat
    sensor: current_temperature
    min_cooling_off_time: 10s
    min_cooling_run_time: 10s
    min_heating_off_time: 10s
    min_heating_run_time: 10s
    min_idle_time: 10s
    cool_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              target_temperature: !lambda 'return id(thermostat).target_temperature_low;'
              mode: COOL
              fan_mode: DIFFUSE
    heat_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              target_temperature: !lambda 'return id(thermostat).target_temperature_high;'
              mode: HEAT
              fan_mode: DIFFUSE
    idle_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              mode: FAN_ONLY
              fan_mode: AUTO
    default_preset: Home
    preset:
      - name: Home
        default_target_temperature_low: 20 °C
        default_target_temperature_high: 22 °C

  - platform: mitsubishi_heatpump
    id: mitsubishi
    hardware_uart: UART0
    on_state:
      - if:
          condition: 
            # Check to see if the component is set up and connect to
            - lambda: 'return !isnan(id(mitsubishi).current_temperature);'
          then:
            - globals.set:
                id: minispit_ready
                value: "true"
            - sensor.template.publish:
                id: current_temperature
                state: !lambda 'return id(mitsubishi).current_temperature;'
```

LubergAlexander, 2022-11-05:
Thank you @einstine909!

Here is the updated code which works as expected
```yaml
climate:
  - platform: mitsubishi_heatpump
    name: "Mitsubishi Heatpump"
    id: mitsubishi
    hardware_uart: UART0
    on_state:
      - if:
          condition: 
            # Check to see if the component is set up and connect to
            - lambda: 'return !isnan(id(mitsubishi).current_temperature);'
          then:
            - globals.set:
                id: minispit_ready
                value: "true"
            - sensor.template.publish:
                id: current_temperature
                state: !lambda 'return id(mitsubishi).current_temperature;'

  - platform: thermostat
    id: heatpump_thermostat
    name: Minisplit Thermostat
    sensor: current_temperature
    min_cooling_off_time: 10s
    min_cooling_run_time: 10s
    min_heating_off_time: 10s
    min_heating_run_time: 10s
    min_idle_time: 10s
    default_preset: Standard
    preset:
      - name: Standard
        default_target_temperature_low: 67 F
        default_target_temperature_high: 75 F
    cool_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              target_temperature: !lambda 'return id(heatpump_thermostat).target_temperature_low;'
              mode: COOL
              fan_mode: DIFFUSE
    heat_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              target_temperature: !lambda 'return id(heatpump_thermostat).target_temperature_high;'
              mode: HEAT
              fan_mode: DIFFUSE
    idle_action:
      if:
        condition: # Make sure minisplit is ready
          - lambda: 'return id(minispit_ready);'
        then:
          - climate.control:
              id: mitsubishi
              mode: FAN_ONLY
              fan_mode: AUTO

sensor:
  - platform: template
    id: current_temperature
    lambda: |-
      return id(mitsubishi).current_temperature;
    update_interval: 10s

```

closed 2022-11-05 by LubergAlexander