geoffdavis/esphome-mitsubishiheatpump #85: RFE: Passthrough support to enable MHK1/MHK2 usage simultaneously

kpfleming, 2023-01-19:
(Note: I'm not actually asking anyone to implement this... I plan to do it myself, but opened this issue to be transparent about it and see if others are interested!)

I'll be using this component on an M5Stick-C (ESP32-PICO-D4 based) connected to a PVA-series air handler. In order to ensure that the existing MHK2 wireless thermostat continues operating, this component will need to implement passthrough of serial data packets between two UARTs. There's an example of doing this (for MHK1, but similar things should work for MHK2) here: https://github.com/akamali/mhk1_mqtt

Unfortunately that method uses a heavily modified version of the SwiCago library, so I don't know how practical it will be to complete this project, but I will be giving it a go :)

If this fails, I'll end up using the HeatPumpEmulator from the same author to implement a 'fake' HeatPump for the MHK2 to talk to, and Home Assistant will handle taking data from it and supplying it to the ESPHome-based interface. Should be fun either way!

2xaronl, 2023-04-17:
@kpfleming Were you able to make progress on this?  I've got a PVA being installed with a MHK2 as I type this and I'd love to have the MHK2 operate simultaneously.

kpfleming, 2023-04-17:
I have not yet, sadly, other projects have gotten in the way. It's getting close to the top of the stack though!

thomashillyer, 2023-04-25:
I just set up this project last night and have this exact same setup and question. I am very interested to see your work in this.

hgabignon, 2023-05-28:
Marking myself as interested to participate in this effort. I have a couple of FS units and a PVA at my home with a MHK2/MIFH2, so once I'm operational, I'd love to help / start this branch.

cfamfl2019, 2023-06-02:
Also interested and happy to help test.

closed 2025-03-31 by geoffdavis