geoffdavis/esphome-mitsubishiheatpump #90: Add extra options via different components

brywithawhy, 2023-02-09:
Just had a MSZ-AP71VGD Indoor Unit installed which has Powerful and Eco Modes, as well as Horizontal vane control and a Night Mode (which I don't know what it does yet).

I understand the climate component in ESPHome is limited, however I've previously used a similar integration for my Panasonic units from https://github.com/DomiStyle/esphome-panasonic-ac which has the ability to use select components for vane position/swing, a switch for the Eco Mode, and uses the "Preset" menu for the climate component to control Powerful/Quiet modes.

Is this something that could be added to this component to get this functionality working for Mitsubishi Units?

Smandurlo, 2023-07-16:
I control my mitsubishi ACs with tasmota and I have the possibility to use any combination of hvac mode/fan speed/swing mode. It can be orizzontal, vertical and so on. No problem to add extra mode like your night, powerful, eco... I have quiet too.
Everything is managed with mqtt. Will it work with esphome and mqtt instead of the api? I am very new to esphome and I have some difficulties to understand how it work.

I am attracted to switch to esphome because it has the possibility to use an external temperature sensor averriding the AC internal one and, afaik, I cannot do it with tasmota.

If you know more about mqtt in esphome we could work on it.

brywithawhy, 2023-07-16:
> I control my mitsubishi ACs with tasmota and I have the possibility to use any combination of hvac mode/fan speed/swing mode. It can be orizzontal, vertical and so on. No problem to add extra mode like your night, powerful, eco... I have quiet too. Everything is managed with mqtt.
> 

Have you got any info on your tasmota setup? I'm going to ditch esphome because there's just too little features and I have strange behaviour on occasion.

Smandurlo, 2023-07-16:
I am interested in the external temperature sensor, it is not possible afaik with tasmota. Other than that, they have zero problem. What is the strange behaviour you have?

For tasmota you need to compile your firmware adding the mitsubishi ac component and flash your esp. Than you make your configuration with mqtt.

Smandurlo, 2023-07-16:
[image]

Just a screenshot of the code. You can add anything you need according with your settings

brywithawhy, 2023-07-20:
Thanks for the info @Smandurlo
Guess I've got no reason to stay with this esphome version now.

Smandurlo, 2023-07-20:
Yes, I tried esphome and there are some very good things, but I still prefer tasmota.

geoffdavis, 2023-07-24:
Interesting, the Tasmota Mitsubishi AC component didn't exist when I wrote this integration years ago. Thanks for the pointer.

The Climate integration in ESPHome is limited to a list of predefined actions in the API, detailed here: https://esphome.io/api/classesphome_1_1climate_1_1_climate_call.html

There's no way through Home Assistant to add anything else without expanding the whole API.

closed 2023-07-24 by geoffdavis

brywithawhy, 2023-07-25:
They could be added as separate switches or input selects along side the climate component.
That's how the Panasonic one works from the link I posted above.