geoffdavis/esphome-mitsubishiheatpump #95: random disconnections

diybateriaslifepo4, 2023-03-05:
Hello, I suffer many disconnections, it happens to me with some esphome devices, but it happens much more often with the one I have for the mitshubishi air conditioner, this does not happen with tasmota (and I have many devices with that firmware), is there something that can do to fix it?

P.S. I have a high quality unifi network, with AP and unifi router, 100% signal, I have also tried to put none in energy saving...

[image]

rudizl, 2023-03-05:
Try to set manual ip, that solved my problem with disconnections. 
[image]

diybateriaslifepo4, 2023-03-05:
I have manual ip configured in my router unifi, Anyway, I'm going to try to put it manually, thanks

diybateriaslifepo4, 2023-03-06:
> Try to set manual ip, that solved my problem with disconnections. [image]

I've tried and unfortunately the same thing keeps happening... :)

DhrDemesmaeker, 2023-10-08:
I have the exact same problem (two heatpumps, same problem).
Did you ever manage to solve the problem?

diybateriaslifepo4, 2023-10-10:
Hello, my problem was in the unifi configuration, after some time working stably I can say that I have solved the problem, I am attaching screenshots of the changes I made in case they are useful to you.

[image]
[image]
[image]
[image]
[image]

jamiebegin, 2024-03-01:
> Hello, my problem was in the unifi configuration, after some time working stably I can say that I have solved the problem, I am attaching screenshots of the changes I made in case they are useful to you.
> 
Which options did you change that improved the stability?

jamescadd, 2024-03-01:
I am curious if the HA instance is also on LAN 1 or another network?

diybateriaslifepo4, 2024-03-04:
> > Hello, my problem was in the unifi configuration, after some time working stably I can say that I have solved the problem, I am attaching screenshots of the changes I made in case they are useful to you.
> 
> Which options did you change that improved the stability?

I have posted some screenshots with the configuration made, after a few months I can say that it still works perfectly.

diybateriaslifepo4, 2024-03-04:
> I am curious if the HA instance is also on LAN 1 or another network?

yes, HA server and all of my domotic things are on LAN1 too

closed 2025-03-31 by geoffdavis