geoffdavis/esphome-mitsubishiheatpump #96: Air to Water heat pumps

Ulrar, 2023-04-04:
Hi,

I just got an air to water ecodan heat pump installed, which has the CN105 header.

It's air to water so there's no fans (well, there's fans on the outside unit of course, but I'm assuming that's not what the fan modes are for), there's no cool and no swing mode that I'm aware of.
Should I just set the supported mode to HEAT only, and leave fan and swing to an empty list ? Or would this component be too geared towards forced air and simply not work ?

Just figured I'd ask before plugging in, I assume sending unsupported commands to the heat pump won't hurt but I don't know that for sure.

TimoFriedri, 2023-08-09:
I am having the same situation. Having a Ecodan Air-Water central heating system.

Anz update on the support of these?

Globemaster86, 2023-08-26:
Did either of you have any success with using these with an ecodan air to water system? Planning on doing the same with mine just seeing if there are any success stories first and if it was limited or full functionality from this method?

TimelessNL, 2023-09-06:
I have no experience with this custom_component (_yet_) and thus can't say if it supported or not, but please take a look at [this repository](https://github.com/clandmeter/esphome/tree/main/ecodan) for which you will need a [Procon MelcoBEMS MINI (A1M)](https://www.mitsubishitechinfo.ca/sites/default/files/IM_MelcoBEMS%20MINI%20%28A1M%29%20-%20User%20Manual%20V1.1.2_fw3.0.23.pdf). This will probably give you much more control over your Ecodan unit.

TimoFriedri, 2023-09-06:
I was not aware of this modbus box. I agree this looks like a solid solution but the best price for an A1M box I can find is 240EUR, which is quite a lot to get some sensor values.

roberttoja-priv, 2024-01-15:
Hi, you can find what you search for here -> https://github.com/tobias-93/esphome-ecodan-heatpump

closed 2025-03-31 by geoffdavis