geoffdavis/esphome-mitsubishiheatpump #99: esphome::climate::Climate::away' is deprecated: away is deprecated

mrkaqz, 2023-04-24:
While compiling for the code in ESPHOME. the log gave some warning below.

Warning below:

------------------------------------------------------------------------------------------------------------------

Compiling /data/1st-fl-room-ac/.pioenvs/1st-fl-room-ac/src/esphome/components/restart/switch/restart_switch.cpp.o
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/climate/climate.h: In constructor 'esphome::climate::Climate::Climate()':
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In constructor 'MitsubishiHeatPump::MitsubishiHeatPump(HardwareSerial*, uint32_t)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:35:25: note: synthesized method 'esphome::climate::Climate::Climate()' first required here
   35 |     hw_serial_{hw_serial}
      |                         ^

zepheris, 2023-04-25:
I just updated ESPHome to 2023.4.1 and received the same error as above:

`Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/restart/button/restart_button.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/sensor/automation.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/sensor/filter.cpp.o
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/climate/climate.h: In constructor 'esphome::climate::Climate::Climate()':
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In constructor 'MitsubishiHeatPump::MitsubishiHeatPump(HardwareSerial*, uint32_t)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:35:25: note: synthesized method 'esphome::climate::Climate::Climate()' first required here
   35 |     hw_serial_{hw_serial}
      |                         ^
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/socket/socket.cpp.o
Compiling /data/hvac-1/.pioenvs/hvac-1/src/esphome/components/status/status_binary_sensor.cpp.o`

Trying to figure out where the code is that specifies climate away. I would think its in the ESP Home code, but I'm not a coder and having trouble understanding where its listed.

phidauex, 2023-04-27:
I believe this is an ESPHome bug/quirk, but because it doesn't hurt anything and will eventually go away on its own when they remove the parameter entirely, it is listed as "won't fix". I believe it is safe to ignore as a warning only, not impacting functionality.

https://github.com/esphome/issues/issues/2668

closed 2025-03-31 by geoffdavis