geoffdavis/esphome-mitsubishiheatpump #100: Zone control?

bleomycin, 2023-04-27:
Your project looks wonderful! Forgive me is this is a dumb question. I have a new mitsubishi hvac system, I have a single outdoor unit that's able to control 4 indoor cassettes (zones). Is your project able to control these 4 zones independently via home assistant or will that require something additional? I wasn't able to figure it out from the readme. Thanks!

bleomycin, 2023-04-27:
> 

haha ok so I need a board for each indoor unit as well as potentially the outdoor unit? Thanks I'll give it a go!

lesleypersyn, 2023-04-27:
> > 
> 
> haha ok so I need a board for each indoor unit as well as potentially the outdoor unit? Thanks I'll give it a go!

Only indoor units.

closed 2025-03-31 by geoffdavis