geoffdavis/esphome-mitsubishiheatpump #107: Warning: Component esphome.coroutine took a long time for an operation (0.10 s).

kbarnes3, 2023-07-19:
I updated to ESPHome 2023.7.0 and a new warning is showing up every few seconds in the logs for all my ESP devices with this project running:

```
[11:08:35][W][component:204]: Component esphome.coroutine took a long time for an operation (0.10 s).
[11:08:35][W][component:205]: Components should block for at most 20-30ms.
```

This is new as of 2023.7.0 (I was previously running 2023.6.5) and my other ESP devices do not have similar warnings in their logs. All my heat pumps are using ESP8266 controllers running this project.

albertofralbe, 2023-07-19:
Same happening to me updating two ESP8266 to the last version. Setting a remote temperature stopped working after a few hours, while the ac control are still working fine

ErikT80, 2023-07-23:
Same issue here after updating to 2023.7.0

immortalnzl, 2023-07-25:
Is anyone else having them drop off the network? When I check the logs, the esphome.coroutine warning is gone. But I need to reinstall the firmware to get connection again.

kbarnes3, 2023-07-25:
I haven't had any reliability problems with my 3 controllers since this update, I've just had the warnings above logged frequently.

banemonster, 2023-08-02:
Seems to be related to this open issue with ESPhome.

[ESPHome Issue 4717](https://github.com/esphome/issues/issues/4717)

brillb, 2023-08-12:
Same for me since 2023.7.0:
`10:53:24 | [W] | [component:205] | Components should block for at most 20-30ms.
10:53:26 | [W] | [component:204] | Component esphome.coroutine took a long time for an operation (0.10 s).`

ziolelle, 2023-08-16:
Same also for me with Wemos D1 Mini:

geoffdavis, 2023-09-07:
All, this is unlikely to get _fixed_ by me. I'd suggest changing your logging level to filter the message out if it bugs you.

A blurb like this ought to do it:

```yaml
logging:
  logs:
    esphome.coroutine: ERROR
```

The listed log message isn't new, just it's priority in the ESPHome logging system. As such, it's been working for years, and It's not likely to cause any real-world performance problems, assuming that you're not running five different hardware sensors and expecting near-real-time feedback.

The issue, as far as I can divine it from the [linked ESPHOME ticket](https://github.com/esphome/issues/issues/4717) (thanks for the find @banemonster), is that the Serial UART reads and writes are blocking calls that cannot be interrupted. Without digging into the bowels of the Arduino serial library, I don't see a way for the underlying HeatPump component to "yield" the CPU until more data appears. These heatpumps typically communicate at 4800 baud, and that's not exactly lightning quick.

Someone with far deeper understanding of the Arduino Serial libraries may have a different answer. If so, I'm happy to accept a pull-request, but you'll likely need to work with @swicago on the underlying library.

closed 2023-09-07 by geoffdavis

geoffdavis, 2023-09-09:
You're right, I had applied the old version to my systems, and it didn't work, then I fixed it but forgot to change my comment before hitting send and running out the door. I edited my comment.

jfisbein, 2023-09-10:
Only one small typo, the root configuration element is `logger` not `logging`