geoffdavis/esphome-mitsubishiheatpump #110: hvac_action Update

javilopezalarcon, 2023-08-26:
Good morning!! I would like if it were possible, that the update of the havac_action was instantaneous, there are times that it is cooling and appears as idle, if at that time I give it to install in esphome to the wemos, when it finishes, give well the value ... It happens quite often that the state is not the real one... I would need it for the automations of the grids with servos, since I manage the position based on the temperature that remains to be reached in the room but as long as it is not in idle that there would be closed.

Thnks!

javilopezalarcon, 2023-08-26:
Whenever possible and not too much headache, for now the automation changes it to the consumption of the air instead of the state and filtering previously if it is cold or hot

SSMCD, 2023-09-09:
I'm seeing a similar issue with `hvac_action` displaying incorrectly.  You can see below that, in `Cool` mode, when the current temperature climbs above the setpoint there is a small blip where `hvac_action` changes to `Cooling`, then immediately switches to `Idle`.  It's not until the current temp reaches the setpoint that it switches back to `Cooling`, then eventually back to `Idle` after the current temp drops below the setpoint. 

[image]

I suspect it may be an issue with the underlying [SwiCago](https://github.com/SwiCago/HeatPump) library though.  See [this post](https://github.com/SwiCago/HeatPump/issues/39#issuecomment-284119724) and [this PR](https://github.com/SwiCago/HeatPump/pull/42#issue-211874258) which explain how the "operating" status is determined.  

My initial guess was that byte `0x06` is more than just "on" or "off" (`0x01` or `0x00`) for some units.  Maybe differentiating different cooling stages?  But anything other than `0x00` should still return `true`, i.e. `Cooling`.  So I still don't know.

I have three MSZ-FH units each with a Wemos D1 mini flashed with ESPHome.  Not sure if it's possible to log the raw packets with this setup, which would be needed to confirm.

SSMCD, 2023-09-10:
Upon further testing, it looks like the `hvac_action` received from the hp is being overwritten, without any additional updates from the unit. 

```
[02:27:36][D][climate:011]: 'HP-Bedroom' - Setting
[02:27:36][D][climate:040]:   Target Temperature: 23.06
[02:27:36][D][MitsubishiHeatPump:364]: control - Was HeatPump updated? YES
[02:27:36][D][climate:378]: 'HP-Bedroom' - Sending state:
[02:27:36][D][climate:381]:   Mode: COOL
[02:27:36][D][climate:383]:   Action: COOLING
[02:27:36][D][climate:386]:   Fan Mode: LOW
[02:27:36][D][climate:401]:   Current Temperature: 23.50°C
[02:27:36][D][climate:407]:   Target Temperature: 23.06°C
[02:27:39][I][MitsubishiHeatPump:434]: Climate mode is: 2
[02:27:39][I][MitsubishiHeatPump:454]: Fan mode is: 131841
[02:27:39][I][MitsubishiHeatPump:469]: Swing mode is: 2
[02:27:39][I][MitsubishiHeatPump:486]: Vertical vane mode is: SWING
[02:27:39][I][MitsubishiHeatPump:504]: Horizontal vane mode is: <<
[02:27:39][I][MitsubishiHeatPump:510]: Target temp is: 23.000000
[02:27:39][D][climate:378]: 'HP-Bedroom' - Sending state:
[02:27:39][D][climate:381]:   Mode: COOL
[02:27:39][D][climate:383]:   Action: IDLE
[02:27:39][D][climate:386]:   Fan Mode: LOW
[02:27:39][D][climate:401]:   Current Temperature: 23.50°C
[02:27:39][D][climate:407]:   Target Temperature: 23.00°C
```

I traced it back to the `hpSettingsChanged()` function:

```c++
        } else if (strcmp(currentSettings.mode, "COOL") == 0) {
            this->mode = climate::CLIMATE_MODE_COOL;
            if (cool_setpoint != currentSettings.temperature) {
                cool_setpoint = currentSettings.temperature;
                save(currentSettings.temperature, cool_storage);
            }
            this->action = climate::CLIMATE_ACTION_IDLE;
        } else if (strcmp(currentSettings.mode, "FAN") == 0) {
```

Commenting-out `this->action = climate::CLIMATE_ACTION_IDLE;` allows `hvac_action` to maintain `Cooling` state throughout the entire cooling cycle.

closed 2025-03-31 by geoffdavis