geoffdavis/esphome-mitsubishiheatpump #112: Target temperature revert to previous setting

dark-nico, 2023-09-02:
Hello, 

I've juste set up a D1 board with MSZ-KJ heat pump. 
Communications work great in both direction (setting from HA and update after using the IR Remote). 
I've got a weird behavior when change target temperature with the IR remote and change another parameter from HA, it revert back to the previous target temperature also the target temp updated fine in HA. 

Step to reproduce: 
Lets say I've got the HVAC running with the following settings set from HA : 
Mode: COOL
Fan Mode: LOW
Swing Mode: OFF
Target Temperature: 20.00°C

Then I update Target temp to 25C from the IR remote : 
[16:27:10][I][MitsubishiHeatPump:311]: Climate mode is: 2
[16:27:10][I][MitsubishiHeatPump:331]: Fan mode is: 769
[16:27:10][I][MitsubishiHeatPump:342]: Swing mode is: 0
[16:27:10][I][MitsubishiHeatPump:350]: Target temp is: 25.000000
[16:27:10][D][climate:378]: 'PAC Bureau' - Sending state:
[16:27:10][D][climate:381]:   Mode: COOL
[16:27:10][D][climate:383]:   Action: IDLE
[16:27:10][D][climate:386]:   Fan Mode: LOW
[16:27:10][D][climate:398]:   Swing Mode: OFF
[16:27:10][D][climate:401]:   Current Temperature: 23.00°C
[16:27:10][D][climate:407]:   Target Temperature: 25.00°C
[16:27:13][D][climate:378]: 'PAC Bureau' - Sending state:
[16:27:13][D][climate:381]:   Mode: COOL
[16:27:13][D][climate:383]:   Action: COOLING
[16:27:13][D][climate:386]:   Fan Mode: LOW
[16:27:13][D][climate:398]:   Swing Mode: OFF
[16:27:13][D][climate:401]:   Current Temperature: 22.50°C
[16:27:13][D][climate:407]:   Target Temperature: 25.00°C
Changes are seen both in ESPHOME logs and HA widget. 

Then I change the Fan speed to DIFFUSE from HA : 
[16:28:49][D][climate:011]: 'PAC Bureau' - Setting
[16:28:49][D][climate:024]:   Fan: DIFFUSE
[16:28:49][D][MitsubishiHeatPump:241]: control - Was HeatPump updated? YES
[16:28:49][D][climate:378]: 'PAC Bureau' - Sending state:
[16:28:49][D][climate:381]:   Mode: COOL
[16:28:49][D][climate:383]:   Action: IDLE
[16:28:49][D][climate:386]:   Fan Mode: DIFFUSE
[16:28:49][D][climate:398]:   Swing Mode: OFF
[16:28:49][D][climate:401]:   Current Temperature: 22.50°C
[16:28:49][D][climate:407]:   Target Temperature: 25.00°C
[16:28:53][I][MitsubishiHeatPump:311]: Climate mode is: 2
[16:28:53][I][MitsubishiHeatPump:331]: Fan mode is: 2049
[16:28:53][I][MitsubishiHeatPump:342]: Swing mode is: 0
[16:28:53][I][MitsubishiHeatPump:350]: Target temp is: 20.000000
[16:28:53][D][climate:378]: 'PAC Bureau' - Sending state:
[16:28:53][D][climate:381]:   Mode: COOL
[16:28:53][D][climate:383]:   Action: IDLE
[16:28:53][D][climate:386]:   Fan Mode: DIFFUSE
[16:28:53][D][climate:398]:   Swing Mode: OFF
[16:28:53][D][climate:401]:   Current Temperature: 22.50°C
[16:28:53][D][climate:407]:   Target Temperature: 20.00°C

For no reason, target temperature is now set back to 20°C.

I don't really know where to start looking, but I can do tests if you point me in the right direction.

nathanjw, 2023-09-05:
This behavior comes from the SwiCago/HeatPump library. It builds up a set of parameters it's supposed to set and then sets all of them at once (I think the underlying Mitsubishi serial protocol is kind of shaped like that) when software updates anything. It does have a mode (autoUpdate or externalUpdate, I forget which is which) that updates its state when the heat pump gets messages from elsewhere (IR, mostly), but it has some weird timeout-based behavior. See discussion for example at https://github.com/SwiCago/HeatPump/issues/192.

The takeaway seems to be that you're best off tossing the remote in a drawer somewhere and trying to do everything through software, if you can possibly arrange it.

geoffdavis, 2023-09-20:
Per @nathanjw the underlying library will read changes back from an external source. If it's conflicting, I'm not sure what to say. I'm going to close this out.

closed 2023-09-20 by geoffdavis

dark-nico, 2023-09-22:
Hi @geoffdavis , thanks for replying. 
I've spent some time diving on both you code and the library code and I've found what is causing this issue. 
Thanks to @nathanjw to pointing me to the external update option. 
This options (both external update and autoupdate are not enabled by default as it's not the common use case. 

I've forked your repo and made a new branch to activate the external update and bump the library to the last version. 
You can find the commit here : 
https://github.com/dark-nico/esphome-mitsubishiheatpump/commit/a1de208944d629103a6b738335d113fa38773689

I did not do a merge request as I don't think this is going to be a modification needed for the majority of people. 
I leave this here if this can help someone.