geoffdavis/esphome-mitsubishiheatpump #113: Support for CNS port boards

ColdTurkeycnc, 2023-09-14:
Thank you for all your work. Got this working easily with esphome only to find that not all my A/C's have the 105 port.

Any chance of integrating the CNS port models into the project? Understanding that wiring will be required with no 5V pin etc.

Thank you.

ag-wood, 2023-09-16:
@ColdTurkeycnc -  are you sure the CNS port models are Mitsubishi Electric?  I have Mitsubishi Heavy Industry units and they are a different thing altogether (different company despite the simalarity in name, I believe).  I used the code and instructions from https://github.com/absalom-muc/MHI-AC-Ctrl for my Mitsubishi Heavy Industry units and they work a treat.

ColdTurkeycnc, 2023-09-16:
Mate, you are dead right. They are MHI units. Was just hoping that they may be able to be included into this as well. Had great success using this method and esphome, really liked how it all went. Cheers for the link though.

ag-wood, 2023-09-16:
I also used ESPHome with the code at the link provided  - If you have home assistant, you just need to copy the src files from the MHI-AC-Ctrl into the root of the ESPHome directory and included them in the config.   Once loaded onto a Wemos D1 Mini I adapted the schematic at https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/images/MHI-AC-Ctrl_Schematic.png to my needs.

ag-wood, 2023-09-16:
Just to answer the thought about whether they could be included in this project - I would guess not as the MHI units also use a different protocol (SPI) where the Mitsubishi Electric are using basic Serial comms.  So it'd be a total rewrite of the code which is already available elsewhere.

ColdTurkeycnc, 2023-09-16:
Thanks heaps for the info mate, much appreciated

closed 2025-03-31 by geoffdavis