geoffdavis/esphome-mitsubishiheatpump #119: [Feature Request] Add additional sensors from the SwiCago lib

swannet, 2023-10-20:
Hi,

Is it possible to add sensors that are available in the SwiCago library but not yet in this esphome module? For example the "compressorFrequency" and the timer info of the unit?

-Sander

tiltec, 2023-11-15:
I tried it here: https://github.com/geoffdavis/esphome-mitsubishiheatpump/compare/master...tiltec:esphome-mitsubishiheatpump:master

It adds a new sensor that can read the compressor frequency. You can configure it like this:
```yaml
climate:
  - platform: mitsubishi_heatpump
    id: hp
    compressor_frequency:
      name: "compressor frequency"
      id: compressor_frequency
```

Unfortunately, it doesn't return anything on my device (MSZ-HR50VF). But in theory, it should work.

danolson1, 2023-12-11:
I have a similar question. Can I get the device to report the 'current_temperature' value as a sensor. Something like this:

```
sensor:
  - platform: mitsubishi_heatpump
    current_temperature
```

echavet, 2024-03-16:
You can create a card this way:
```yaml
type: entity
entity: climate.esp32_hp_clim_sejour
attribute: current_temperature
```

or use a Jinja2 template:
```jinja2
{% set climate_entity = states.climate.esp32_hp_clim_sejour %}
{% if climate_entity %}
  <h1>Informations Climatiseur</h1>
  <p><strong>État :</strong> {{ climate_entity.state }}</p>
    <p><strong>Température cible :</strong> {{ climate_entity.attributes.temperature }}°C</p>
    <p><strong>Température actuelle :</strong> {{ climate_entity.attributes.current_temperature }}°C</p>
    {% if climate_entity.attributes.fan_mode %}
    <p><strong>Mode ventilateur :</strong> {{ climate_entity.attributes.fan_mode }}</p>
  {% endif %}
{% else %}
  <p>Entité climatiseur non trouvée.</p>
{% endif %}
```

closed 2025-03-31 by geoffdavis