geoffdavis/esphome-mitsubishiheatpump #121: Success with the MSZ-GE60/71/80 series

wrouesnel, 2023-11-11:
A note that I've had success with this code on an MSZ-GE60/71/80 series using an M5 Atom Lite.

Crucially, I couldn't build the current develop branch to install this, but I was able to cherry-pick across the UART pin configuration commits and get it working (pushed on my fork: https://github.com/wrouesnel/esphome-mitsubishiheatpump) On this unit the baud rate was 2400, and with the M5 Atom I needed to use pin remapping for some of the broken out top pins. It was also necessary to add the 10kOhm pull up resistors from the TX/RX pins, but I _did not_ need any level shifters (of course this is literally a few hours old so don't trust me that the M5 won't die suddenly or something).

(note: this is ESPhome as I do it - i.e. time keeping etc. are all being configured in that home.yaml file)

```yaml
# Codes for a Mitsubishi Heatpump running on an M5Stack Atom Lite
# substitutions:
#   name: hptest
#   friendly_name: Test Heatpump

packages:
  home: !include home.yaml

external_components:
- source:
    type: local
    path: components

esphome:
  name: me-hpctrl
  name_add_mac_suffix: true
  # Boards tested: ESP-01S (ESP8266), Wemos D1 Mini (ESP8266); ESP32 Wifi-DevKit2
  project:
    name: esphome.mitsubishi-heatpump
    version: "1.0"
  platform: ESP32
  board: m5stack-core-esp32

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  level: VERBOSE
  #baud_rate: 0

web_server:
  port: 80

wifi:
  ap: {}

captive_portal:

ota:

# Enable Home Assistant API
api:

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

climate:
  - platform: mitsubishi_heatpump
    name: Air Conditioning

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART2
    tx_pin: 21
    rx_pin: 25 
    baud_rate: 2400
```

Bascht74, 2023-12-30:
Hi @wrouesnel I used a simial setup. 

Just used the Atom Lite and a GROVE Cable (connects to pin 25/32), no pullup resistors, no level shifters. 
I cut the plastic of on end of the GROVE Cable and insertet freeed pins into a JST PAP-05V-S housing. Click, Click, Click, Click --> ready. So I have a cable that I can connect directly to the Lite and on the other side into the CN105-Port... 

Using your version, only with `tx_pin: 32`, `rx_pin: 26`, `boud rate` is `2400` as well...

BenJamesAndo, 2024-02-16:
Thank you both for this information. I was able to get this running on a M5 Atom Matrix that I had lying around. I used a [Grove to 4 pin dupont cable](https://www.jaycar.com.au/glove-4p-breakout-lead-to-4-x-jumper-pins-packet-of-5/p/XC9192). I then plugged that into a [Dupont to CN105 cable](https://www.ebay.com.au/itm/285315164757). It meant no cable crimping but ideally I would like to improve it to just use one cable similar to Bascht74.

I was able to get it running using geoffdavis version with
```
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@develop
```
and
```
esphome:
  name: mitsubishi-ac
  project:
    name: esphome.mitsubishi-ac
    version: "MSZ-GA60VA"
  platform: ESP32
  board: m5stack-atom
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"
```

I used baud_rate of 9600. 2400 didn't work for me.
```
    hardware_uart: UART1
    tx_pin: 32
    rx_pin: 26
    baud_rate: 9600
```

It also worked with wrouesnel fork and [echavet's fork](https://github://echavet/MitsubishiCN105ESPHome)
echavet being my favourite at the moment as it has the most controls.

tmchow, 2024-07-20:
I’ve got a grove cable and a PAP-05V-S. What exactly is the wiring configuration into the PAP-05V-S?

closed 2025-03-31 by geoffdavis