geoffdavis/esphome-mitsubishiheatpump #123: Set point changes

oot64mm64, 2023-11-14:
I found similar issues but not exactly what I'm seeing so I'm opening a new issue.

I am using this project on D1 boards with two indoor units and Home Assistant.

(1) Mitsubishi, 20K BTU Hyper Series Outdoor Unit (MXZ-2C20NAHZ-U1)
(2) Mitsubishi, 09K BTU Indoor Wall Mounted Units (MSZ-FS09NA-U1))

I am using the set remote temp service as described in method 2 here: https://github.com/geoffdavis/esphome-mitsubishiheatpump#remote-temperature

In HA I wrote a script that converts a F temp reading into C and calls the service. Then I wrote an automation to call the script every minute. I monitor and control the HPs with the standard climate Lovelace card.

The problem I'm having is that in Heat mode I will set a set point (66 or 68 generally) with the Lovelace card and the HP will start heating to that point, then after some amount of time or some condition that I haven't figured out, the set point will change. Usually to 62.0 but sometimes 61.5. I discover later that it's low and the room is cold again 

I am not using the IR remotes at all. Could this be a problem with the ESPHome code on the D1? Or should I be looking in HA? I have not found anything obvious in logs but I'm not sure where to look.

phidauex, 2023-11-16:
Hmm, mine is working well in a similar configuration, but I convert from F to C inside the ESPHome service call rather than before, but that shouldn't make a difference.

Can you graph the climate entity in HA and see when the setpoint is changing? That might give a clue:
[image]

You could also take a look at the logbook and see if something is changing the state of the climate entity:
[image]

The actual ESPhome logs would be helpful too, but it is much harder to catch something in the act there.

oot64mm64, 2023-11-16:
Interesting about the conversion in the service. I do wonder if the conversion is related, maybe something goes wrong and HA or ESPHome or the Heat Pump revert to a "known" state. I have done what you suggested and found only one clue, in the HA Core logs about 1 minute before the set point changed was this error: https://pastebin.com/MTezSmg5
I agree it has been difficult to catch something in the act on the ESPhome logs. Any suggestions about this? Do I have to be connected to the D1 and watching when it happens?

phidauex, 2023-11-16:
Hmm, that error doesn't make a lot of sense to me... The thing that is strange to me is that if you send an invalid value to the set_remote_temperature function I don't see why that would change the setpoint. You reset to the internal temperature if you send it a 0, for instance.

Here is my ESPhome config entry that defines the service - I'm using a lambda to convert from F to C. Does yours look much different?

```
# Enable Home Assistant API
api:
  encryption:
    key: "xxxxx"
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
# Select between the C version and the F version
#        - lambda: 'id(hp).set_remote_temperature(temperature);'
        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'
```

oot64mm64, 2023-11-16:
Mine looks exactly the same except for the conversion line. I agree, this function shouldn't make a lick of difference for the set point of the HP. 
```
api:
  encryption:
    key: "xxxxxx"
    
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(hp).set_remote_temperature(temperature);'

    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'
```

Right now the logs are cluttered and I missed one of the random changes by a couple lines. Can you help me adjust the log section to skip some of these, or are they built-in to ESPHome?
My logs section of config looks like this:
```
# Enable logging
logger:
  baud_rate: 0
  logs:
    component: ERROR
```
And the logs are filled with wifi signal and uptime messages every minute:
```
18:40:48 | [D] | [sensor:093] | 'hp_downstairs WiFi Signal': Sending state -54.00000 dBm with 0 decimals of accuracy
18:40:50 | [D] | [sensor:093] | 'hp_downstairs Uptime': Sending state 47607.13281 s with 0 decimals of accuracy
```

oot64mm64, 2023-11-19:
This seems to have resolved itself as it hasn't happened in 3 days when it was happening every few hours before. There were several ESPHome updates, perhaps something in there affected this.

closed 2023-11-19 by oot64mm64