geoffdavis/esphome-mitsubishiheatpump #127: HR25VF only in web interface

xtypebox, 2023-12-09:
Hello. I'm currently using a D1 mini clone and I managed to control the AC in the web interface created by esphome. In home assistant all the entity's are unavailable. 
MY AC are HR25VF and HR42VF.
Any recommendations?

SGXander, 2023-12-13:
sounds more like your D1 has changed IP address or the HA API connection isn't right. remove and re-add the device in HA first

SGXander, 2023-12-15:
not seen that before. Have you removed from the devices and integrations page and re-added it?

xtypebox, 2023-12-15:
> not seen that before. Have you removed from the devices and integrations page and re-added it?

I removed the device and now I also tried the mithsubishi-mqtt method and same behavior. Just works using the control in the web browser. In the mqtt broker in home assistant the device keeps getting unavailable message.

My cables are 50cm can that be the issue? Or do i need to cut the RX and TX traces to usbchip? But why I can control the ac unit using the browser in that case?

Thanks

SGXander, 2023-12-15:
No, if you can control via the web interface and see the current temp on the web interface as well then it's nothing to do with hardware. it's only the home assistant API connection that is failing. So focus on network and connectivity to HA while troubleshooting.

do you have the below in your ESPHome <device>.yaml:
```
api:
  encryption:
    key: <key>
```
?

xtypebox, 2023-12-15:
> No, if you can control via the web interface and see the current temp on the web interface as well then it's nothing to do with hardware. it's only the home assistant API connection that is failing. So focus on network and connectivity to HA while troubleshooting.
> 
> do you have the below in your ESPHome .yaml:
> 
> ```
> api:
>   encryption:
>     key: <key>
> ```
> 
> ?

In the first try I just have the api without the "encryption: key: <key>" but then when I change and re-add the device te HA ask to put the key and it connects correctly. But when i put the cable back to ac unit same behavior.

Maybe the voltage drops when connected to ac unit?

xtypebox, 2023-12-15:
> it connects to the API when not connected to the A/C unit? 

**Yes, the api works fine when not connected to ac unit. Just the current temp is ND but it's normal**

you must remove power to the AC unit completely before connecting/disconnecting the CN105 port. It is not hot-pluggable.
> 

**I remove the power to connect and disconect** 

> Have you applied the settings as directed for the D1/8266 devices:
> 
> ```
> external_components:
>   - source: github://geoffdavis/esphome-mitsubishiheatpump
> 
> esp8266:
>   board: esp01_1m
> 
> logger:
>   logs:
>     MitsubishiHeatPump: DEBUG
>   baud_rate: 0
> 
> climate:
>   - platform: mitsubishi_heatpump
>     name: AC
>     id: ${id_prefix}_ac
>     icon: mdi:air-conditioner
>     hardware_uart: UART0
>     baud_rate: 2400
>     update_interval: 500ms
>     supports:
>       mode: ['HEAT_COOL', 'COOL', 'HEAT', 'DRY', 'FAN_ONLY']
>       fan_mode: ['AUTO', 'DIFFUSE', 'LOW', 'MEDIUM', 'MIDDLE', 'HIGH']
>       swing_mode: ['OFF', 'VERTICAL']
> 
> button:
>   - platform: restart
>     name: Restart
>     id: ${id_prefix}_restart
> ```
> 
> Also the restart button is key. if you look at my notes in [here](https://github.com/SwiCago/HeatPump/issues/13) you'll see the issue with the RX not working as the D1 boots quicker than the A/C unit but you're having a different problem I think.

**I will try with the restart button.**

**Shoul i change this:  esp8266:>   board: esp01_1m to D1 mini?**

SGXander, 2023-12-15:
As defined [here](https://esphome.io/components/esp8266.html?highlight=esp8266) you can change it to anything on the list but AzDelivery don't have an entry so I can't confirm if that will work or not. mine is set to esp01_1m

xtypebox, 2023-12-18:
> So focus on network and connectivity to HA while troubleshooting.

Well it seems to be network related because when the power went out and when everything started up I was able to control the AC through the esphome panel. But now it seems inaccessible again.

I have the D1 minis in a separate vlan. What can be the problem? Thanks

SGXander, 2023-12-18:
> Well it seems to be network related because when the power went out and when everything started up I was able to control the AC through the esphome panel. But now it seems inaccessible again.
>
> I [...]

For it to work cross-vlan you'll need to ensure that mDNS is working and has holes punched through your firewall for it. similarly you'll want to allow any-any between HA and those devices (sure there are ports listed for more specific rules somewhere but that's what I have set on mine)

xtypebox, 2023-12-20:
> For it to work cross-vlan you'll need to ensure that mDNS is working and has holes punched through your firewall for it. similarly you'll want to allow any-any between HA and those devices (sure [...]

Don't seems to be network related after all. Maybe the supply voltage of ac unit isn't enough and interfere with wifi chip?
I don't know what more can I do.

xtypebox, 2024-01-01:
It turned out to be a problem with one of the access points. Fortunately there was a firmware update and everything is working correctly.

Thanks for all the support.

closed 2024-01-01 by xtypebox