geoffdavis/esphome-mitsubishiheatpump #128: Storing working example files

danolson1, 2023-12-21:
I got things working on an ESP32C3. Is there a place I can post the Homeassistant yaml file and relevant construction notes?

tfasz, 2025-03-06:
I'd start with posting here. ;-) I'd be interested to see it as I'm getting compilation errors when trying to build it for my ESP32C3.

danolson1, 2025-03-08:
Note, this is only working with ESPHome versions up to 2023.12.5. At some point, there was an update that broke my workaround and I haven't had time to fix it. I think I'm going to go back to using ESP32 devices for my mitsubishi heat pumps.

```
# Notes: need to attach wifi antenna for good signal reception

substitutions:
  name: hp-4
  friendly_name: ADU West Bedroom Heat Pump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
    #board_build.mcu: esp32c3 # not needed, not sure what this does
    board_build.extra_flags:
      # Override, defaults to '-DARDUINO_USB_CDC_ON_BOOT=1'     
      # This is necessary to allow mitsubishi heatpump to work correctly
      # More info here https://wiki.seeedstudio.com/XIAO_ESP32C3_Pin_Multiplexing/
      - "-DARDUINO_USB_CDC_ON_BOOT=0"  

esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.4.0 # 6.4.0 and 5.4.0 worked

# Enable logging
logger:
 #hardware_uart: UART0
 baud_rate: 0 # disable the logger

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxx"
    password: "xxxx"

web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    
    # ESP32C3, must be UART0
    hardware_uart: UART0
```

closed 2025-03-31 by geoffdavis