geoffdavis/esphome-mitsubishiheatpump #129: Warning about coroutine taking a long time.

lloydc99, 2023-12-22:
I just did an ESPHome update and now I've got a new warning popping up every 2-3 seconds.
I'm running on a Wemos D1 Mini.

```
09:55:40 | [W] | [component:214] | Component esphome.coroutine took a long time for an operation (0.10 s).
09:55:40 | [W] | [component:215] | Components should block for at most 20-30ms.
09:55:42 | [W] | [component:214] | Component esphome.coroutine took a long time for an operation (0.10 s).
09:55:42 | [W] | [component:215] | Components should block for at most 20-30ms.
09:55:44 | [W] | [component:214] | Component esphome.coroutine took a long time for an operation (0.10 s).
09:55:44 | [W] | [component:215] | Components should block for at most 20-30ms.
```

It seems to come up just fine.

```
[09:52:59][I][app:102]: ESPHome version 2023.11.6 compiled on Dec 22 2023, 09:25:11
[09:52:59][C][wifi:559]: WiFi:
[09:52:59][C][wifi:391]:   Local MAC: D8:BF:C0:F2:E0:D4
[09:52:59][C][wifi:396]:   SSID: [redacted]
[09:52:59][C][wifi:397]:   IP Address: 192.168.1.11
[09:52:59][C][wifi:398]:   BSSID: [redacted]
[09:52:59][C][wifi:400]:   Hostname: 'esphome-web-f2e0d4'
[09:52:59][C][wifi:402]:   Signal strength: -78 dB ▂▄▆█
[09:52:59][C][wifi:406]:   Channel: 11
[09:52:59][C][wifi:407]:   Subnet: 255.255.255.0
[09:52:59][C][wifi:408]:   Gateway: 192.168.1.1
[09:52:59][C][wifi:409]:   DNS1: 192.168.1.1
[09:52:59][C][wifi:410]:   DNS2: 0.0.0.0
[09:52:59][C][logger:416]: Logger:
[09:52:59][C][logger:417]:   Level: DEBUG
[09:52:59][C][logger:418]:   Log Baud Rate: 0
[09:52:59][C][logger:420]:   Hardware UART: UART0
[09:52:59][C][uptime.sensor:031]: Uptime Sensor 'esphome-web-f2e0d4 Uptime'
[09:52:59][C][uptime.sensor:031]:   Device Class: 'duration'
[09:52:59][C][uptime.sensor:031]:   State Class: 'total_increasing'
[09:52:59][C][uptime.sensor:031]:   Unit of Measurement: 's'
[09:52:59][C][uptime.sensor:031]:   Accuracy Decimals: 0
[09:52:59][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[09:52:59][C][homeassistant.time:010]: Home Assistant Time:
[09:52:59][C][homeassistant.time:011]:   Timezone: 'UTC0'
[09:52:59][C][restart.button:017]: Restart Button 'Sunroom Split Restart'
[09:52:59][C][version.text_sensor:021]: Version Text Sensor 'esphome-web-f2e0d4 ESPHome Version'
[09:52:59][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[09:52:59][I][MitsubishiHeatPump:062]: ESPHome MitsubishiHeatPump version 2.4.1
[09:52:59][I][MitsubishiHeatPump:504]:   Supports HEAT: YES
[09:52:59][I][MitsubishiHeatPump:505]:   Supports COOL: YES
[09:52:59][I][MitsubishiHeatPump:506]:   Supports AWAY mode: NO
[09:52:59][I][MitsubishiHeatPump:507]:   Saved heat: 21.0
[09:52:59][I][MitsubishiHeatPump:508]:   Saved cool: 25.0
[09:52:59][I][MitsubishiHeatPump:509]:   Saved auto: -1.0
[09:52:59][C][captive_portal:088]: Captive Portal:
[09:52:59][C][web_server:168]: Web Server:
[09:52:59][C][web_server:169]:   Address: esphome-web-f2e0d4.local:80
[09:52:59][C][mdns:115]: mDNS:
[09:52:59][C][mdns:116]:   Hostname: esphome-web-f2e0d4
[09:52:59][C][ota:097]: Over-The-Air Updates:
[09:52:59][C][ota:098]:   Address: esphome-web-f2e0d4.local:8266
[09:52:59][C][api:139]: API Server:
[09:52:59][C][api:140]:   Address: esphome-web-f2e0d4.local:6053
[09:52:59][C][api:142]:   Using noise encryption: YES
[09:52:59][C][wifi_info:009]: WifiInfo IPAddress 'esphome-web-f2e0d4 IP'
[09:52:59][C][wifi_info:011]: WifiInfo SSID 'esphome-web-f2e0d4 SSID'
[09:52:59][C][wifi_info:012]: WifiInfo BSSID 'esphome-web-f2e0d4 BSSID'
[09:52:59][C][wifi_signal.sensor:009]: WiFi Signal 'esphome-web-f2e0d4 WiFi Signal'
[09:52:59][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[09:52:59][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[09:52:59][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[09:52:59][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
```

rs, 2023-12-22:
I got the same issue after the last upgrade of esphome. Let me know if you need more info.

rs, 2023-12-22:
Seems like it does also break the sending of commands to the unit.

rs, 2023-12-22:
Issue reproduced with both 2023.12.2 and 2023.12.3.

lloydc99, 2023-12-22:
My unit continues to respond to commands. I will try to rollback to an earlier commit.

lloydc99, 2023-12-22:
I rolled mitsubishiheatpump back to 2.4.0, which was my original working install, and I still had the problem.
That points to this coming from my recent upgrade of ESPhome to 2023.11.6.

```
external_components:
- source:
    type: git
    url: https://github.com/geoffdavis/esphome-mitsubishiheatpump.git
    ref: 2.4.0
```

lloydc99, 2023-12-22:
It looks like this was [reported](https://github.com/esphome/issues/issues/4717#issuecomment-1644975074) back in July in the esphome issues.

phidauex, 2023-12-27:
It was definitely caused by the ESPhome update, not the mitsubishiheatpump package. For what its worth, while it spams the logs, my units are still responding to temperature update commands just fine.

JanJoh, 2023-12-28:
> Seems like it does also break the sending of commands to the unit.

DO you mean your device shows as "Unavailable?

I'd advise you to check your entities. For unknown reason I all of a sudden had a new set of entities with the suffix _2 while the original ones had become unavailable. 

Deleting the unavailable entities, and renaming the new ones fixed that issue for me.

jesus51998, 2024-02-20:
logger:
#  Eliminates the 2 error lines that appear but does not solve the problem
  logs:             
    component: ERROR

Add to your yaml file.
The problem continues but, is not visible

rippe77, 2024-03-19:
I have this issue. 
[16:15:59][W][component:214]: Component esphome.coroutine took a long time for an operation (2.13 s).
[16:16:00][W][component:215]: Components should block for at most 20-30ms.

And in addition the command to the unit do not go through. For example setting the setpoint or turning off the unit. It seems it goes through, but the changed settings do not have any effect. 

[16:19:00][D][MitsubishiHeatPump:409]: Setting remote temp: 22.5
[16:19:04][D][climate:011]: 'oh_esp32_ilp' - Setting
[16:19:04][D][climate:040]:   Target Temperature: 20.00
[16:19:04][D][MitsubishiHeatPump:241]: control - Was HeatPump updated? YES
[16:19:04][D][climate:396]: 'oh_esp32_ilp' - Sending state:
[16:19:05][D][climate:399]:   Mode: HEAT
[16:19:05][D][climate:401]:   Action: IDLE
[16:19:05][D][climate:404]:   Fan Mode: MIDDLE
[16:19:05][D][climate:416]:   Swing Mode: OFF
[16:19:06][D][climate:419]:   Current Temperature: 22.50°C
[16:19:06][D][climate:425]:   Target Temperature: 20.00°C
[16:19:07][W][component:214]: Component api took a long time for an operation (3.27 s).
[16:19:07][W][component:215]: Components should block for at most 20-30ms.
[16:19:10][W][component:214]: Component esphome.coroutine took a long time for an operation (2.13 s).
[16:19:10][W][component:215]: Components should block for at most 20-30ms.

Reading the data from the unit seem to work just fine (current temperature, etc.)

closed 2025-03-31 by geoffdavis