geoffdavis/esphome-mitsubishiheatpump #132: Blue led flashing regularly on ESP-01

enomam, 2024-01-03:
Hey all,

I've got this working on all my Mitsubishi heatpumps (mainly msz-ap20vg units), and it's working well. I'm using ESP-01, with breakout boards.
I notice that about once per second, the blue led flashes. I suspect it's related to this issue ( https://github.com/esphome/issues/issues/4717 ), and the timing _seems_ to be in sync with each log entry.

I've stopped logging those issues to the console, but still get the flashing led. Is there a way to stop it from flashing?

I've asked here first, as not 100% sure if this is a esphome-mitsubishiheatpump thing, an esphome this, or a swicago thing.

I'm running EspHome 2023.10.3, and my config looks like this:

```
substitutions:
  name: esphome-web-a77526
  friendly_name: Lounge AC - ESP01

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

esp8266:
  board: esp01_1m

logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0
  logs:
    component: ERROR

mqtt:
  broker: 10.0.10.110

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-A77526"
    password: "****"

captive_portal:

climate:
  - platform: mitsubishi_heatpump
    name: ${friendly_name}
    hardware_uart: UART0
    #baud_rate: 2400
    #update_interval: 500ms
    supports:
      mode: [HEAT_COOL, COOL, HEAT, DRY, FAN_ONLY]
      fan_mode: [AUTO, DIFFUSE, LOW, MEDIUM, HIGH]
      swing_mode: ["OFF", VERTICAL]
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 0.5

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${friendly_name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${friendly_name} WiFi Signal
    update_interval: 60s
```

Protoncek, 2024-01-04:
In my case (on my esp01) blue led is connected to GPIO2, i use it for connection monitoring, so i guess that if you define an output on that pin and set it to 0 it should be ok.

enomam, 2024-02-11:
Sorry for the delay in responding, but I've been going through the esphome docs, and unclear how to create an output, and set it to 0.

Protoncek, 2024-02-12:
First define an output:
```
output:
  - id: unused_pin
    platform: gpio
    pin: GPIO2

```
then set it to one (it's actually 1 for turn the led off, not 0 since led is connected reversed - from +3.3V to gpio):
```
esphome:
  on_boot:
    then:
      output.turn_on:
        id: unused_pin
```

closed 2025-03-31 by geoffdavis