geoffdavis/esphome-mitsubishiheatpump #134: Impossible to connect to the heatPump

yann500, 2024-01-12:
On MQTT Explorer, my heat pump work perfectly:
192.168.0.34
heatpump = {"power":"ON","mode":"HEAT","temperature":22,"fan":"AUTO","vane":"SWING","wideVane":"|"}
status = {"roomTemperature":21,"operating":true}
timers = {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
debug = 

But after compiling of ESPhome config, I can't connect to ESP. 
======================== [SUCCESS] Took 1118.87 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of heatpump.local
ERROR Error resolving IP address of heatpump.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

Thanks in advance for your help.

Yann.

disruptivepatternmaterial, 2024-01-13:
This would appear to have nothing to do with this project/code, this is ESPHome not finding your ESP on the network: check out https://esphome.io/components/wifi.html and look at the `use_address` option.

yann500, 2024-01-14:
Thanks for your help, but I didn't see this option in your YAML example. Why my esphome has to specific configuration ?

closed 2025-03-31 by geoffdavis